emsesp/EMS-ESP32 #334: Would it be possible to add telegram info within entity info in api-response?

tp1de, 2022-01-27:
@proddy @MichaelDvP 
Would it be possible to add the the telegram info for the respective entity within api-response?
E.g. telegram-type (handler), offset, no of bytes - or bit, multiplicator etc. 

e.g. for boilermaxpower:

{
  "name": "burnmaxpower",
  "fullname": "burner max power",
  "value": 100,
  "type": "number",
  "min": 0,
  "max": 100,
  "uom": "%",
  "writeable": true,
  "visible": true,
  "handler": "E6",
  "offset": "04",
  "bytes": 1,
  "bit": na,
  "multi": 1
}

proddy, 2022-01-27:
Why?

tp1de, 2022-01-27:
This would enable to filter those bytes out of telegrams which are already recognized and processed and try to analyze further other parameters which are not covered yet. I would like to do this with a javascript program.
In some case ems-esp is using different telegram types/offset compared to the respective  km200 entity.
I could try to further analyse telegrams for improvement of ems-esp code.

proddy, 2022-01-27:
I don't see the value to others (meaning the 1500+ other users). Perhaps we should show you how you can make changes yourself to complete your KM200 project, or make a special branch for this effort. I'd rather not pollute EMS-ESP with changes that divert from its original intention or provide no real value to others.

tp1de, 2022-01-27:
Understood. It would be a help for me to get a small documentation how extract these information in a most easiest way out of the coding. I could then make an overview for everybody who might be interested which you could add to the ems-esp documentation. I would be happy to support if you like me to do so ...

MichaelDvP, 2022-01-27:
Take a look into the device-file and search the register-telegram-type you want to know. [For your example 0xE6 here](https://github.com/emsesp/EMS-ESP32/blob/3d1a050e22153a0fc947dac0d8dae01f5b4f663f/src/devices/boiler.cpp#L78). You see type, name, if it is fetched and the function. Then search the function [and find it here](https://github.com/emsesp/EMS-ESP32/blob/3d1a050e22153a0fc947dac0d8dae01f5b4f663f/src/devices/boiler.cpp#L655). There is an example telegram in the header comment and the values are read. You can see `burnMaxPower_` with offset 4. 
Now search for burnmaxpower, [the registration is here](https://github.com/emsesp/EMS-ESP32/blob/3d1a050e22153a0fc947dac0d8dae01f5b4f663f/src/devices/boiler.cpp#L146-L147) and shows you that its of type UINT (8bit, unsigned), has no scaling, unit percent and there is a function to set the value.
If you search again you also [find here](https://github.com/emsesp/EMS-ESP32/blob/3d1a050e22153a0fc947dac0d8dae01f5b4f663f/src/devices/boiler.cpp#L486) that this value is also read from 0x16 telegram on ems boilers.

proddy, 2022-05-05:
so no need to write extra code then. good!

closed 2022-05-05 by proddy