emsesp/EMS-ESP32 #336: Support for Junkers FB100 Thermostat - telegram handler for 0x171

akanto, 2022-01-28:
There is a [fair amount of Junker Thermostat ](https://github.com/emsesp/EMS-ESP32/blob/b65866217aab44f1501ef5fafefa53125cfba6f8/src/device_library.h#L95) supported by this project, and I would like to ask for supporting Junkers FB100 as well.

This thermostat has a very similar look&feel to already supported FW100 models. Based on the logs, a handler is required for the type 0x171 to get it supported.

```
2022-01-28 17:27:07.977 T 11454: [emsesp] Unknown(0x1A) <- Boiler(0x08), UBAParameters(0x16), data: 02
2022-01-28 17:27:07.992 D 11455: [telegram] [DEBUG] New Rx telegram, message length 2
...
2022-01-28 17:27:08.676 T 11462: [emsesp] Unknown(0x1A) -> All(0x00), JunkersMonitor(0x171), data: 03 02 00 DC 00 EC 00 00 00
2022-01-28 17:27:08.676 D 11463: [emsesp] No telegram type handler found for ID 0x171 (src 0x1A) 
```

Thanks.

[image]

akanto, 2022-01-28:
I have  3 thermostats:
FW200 -> master one and handling hc1 and hc2 with an FB10 thermostat assigned to hc1 as control device
FB10 -> as mentioned above
FB100 -> it controls the 3rd heating circuit, and it is not a master one

[emsesp_system_info.txt](https://github.com/emsesp/EMS-ESP32/files/7960822/emsesp_system_info.txt)

```
   {
      "type": "Unknown", -> this is the FB10
      "name": "unknown (DeviceID:0x18, ProductID:109, Version:16.05)"
    },
    {
      "type": "Unknown", -> this is the FB100 
      "name": "unknown (DeviceID:0x1A, ProductID:110, Version:14.05)"
    }
```

MichaelDvP, 2022-01-28:
Ok, the master FW200 at id 0x10 controllsall circuits, and the FB10 is remote for hc1 and FB100 is remote for hc3. 
We have to add to devices:
```
{109, DeviceType::THERMOSTAT, F("FW10"), DeviceFlags::EMS_DEVICE_FLAG_JUNKERS},
{110, DeviceType::THERMOSTAT, F("FB100"), DeviceFlags::EMS_DEVICE_FLAG_JUNKERS},
```
to recognize the devices.  It's a bit strange that a remote sends the telegram 0x171, that should be a master telegram.
This will need some more logs in next step.

Can you compile, or should i send a bin for testing.

akanto, 2022-01-28:
Unfortunately, I can't compile it. Please, be so kind as to send the binary and I am happy to test it out.

MichaelDvP, 2022-01-28:
Here it is, 
[EMS-ESP-3_4_0b2-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7961057/EMS-ESP-3_4_0b2-ESP32.zip)
This is v3.4, you have to reload page after update to show the values, Also if you're using HA, clear all mqtt entries starting with `homeassistant/` to make autodiscovery working.

This should register the thermostats, but values are only an master FW200. Please send `http://emsesp.local/api/thermostat`  and a minute of `loglevel all`. I expected that both remotes send 0x123 telegram to master, but your log shows 0x171.

akanto, 2022-01-28:
Thank you.

```
{
  "datetime": "20:21:07 28.01.2022",
  "hc1": {
    "seltemp": 21.5,
    "currtemp": 24.3,
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 21.5,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "fb10",
    "program": "prog_e"
  },
  "hc2": {
    "seltemp": 22.5,
    "currtemp": 24.1,
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 22.5,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "off",
    "program": "prog_e"
  }
}
```

I can confirm, that the names are correctly detected. Please find the logs attached.
Log: [log_3.4_patched.txt](https://github.com/emsesp/EMS-ESP32/files/7961286/log_3.4_patched.txt)
Info: [emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/7961337/emsesp_info.txt)

akanto, 2022-01-29:
It looks like the telegrams are purposefully registered only for the master thermostat and the data is fetched from the master one, which works correctly for hc1 and for FB10 on the hc2, since hc2 data is available through the master. 

But the hc3 data controlled by FB 100 (0x1A), is not available through the master, so the registration for FB 100 (that controls hc3) might be required.

https://github.com/emsesp/EMS-ESP32/blame/dev/src/devices/thermostat.cpp#L149

```
     // JUNKERS/HT3
    } else if (model == EMSdevice::EMS_DEVICE_FLAG_JUNKERS) {
        if (actual_master_thermostat == device_id) {
            monitor_typeids = {0x016F, 0x0170, 0x0171, 0x0172};
            for (uint8_t i = 0; i < monitor_typeids.size(); i++) {
                register_telegram_type(monitor_typeids[i], F("JunkersMonitor"), false, MAKE_PF_CB(process_JunkersMonitor));
            }

            if (has_flags(EMS_DEVICE_FLAG_JUNKERS_OLD)) {
                // FR120, FR100
                set_typeids = {0x0179, 0x017A, 0x017B, 0x017C};
                for (uint8_t i = 0; i < monitor_typeids.size(); i++) {
                    register_telegram_type(set_typeids[i], F("JunkersSet"), false, MAKE_PF_CB(process_JunkersSet2));
                }
            } else {
                set_typeids = {0x0165, 0x0166, 0x0167, 0x0168};
                for (uint8_t i = 0; i < monitor_typeids.size(); i++) {
                    register_telegram_type(set_typeids[i], F("JunkersSet"), false, MAKE_PF_CB(process_JunkersSet));
                }
            }
        }
        register_telegram_type(0x123, F("JunkersRemote"), false, MAKE_PF_CB(process_JunkersRemoteMonitor));
    }

    if (actual_master_thermostat != device_id) {
        return; // don't fetch data if more than 1 thermostat
    }
```

MichaelDvP, 2022-01-29:
Yes, normally the master 0x10 handles all heatingcircuits and the devices with id 0x18..0x1F are remotes sending only the actual room temperature, as your FB10. There is another way with no master, only with single hc thermostats, then these thermostats have only  ids 0x18..0x1F and 0x10 is missing. In this scenarion the 0x18 handles hot water and hc1, the others only the hc. 
But you have a mixed configuration i've never seen before. It's quite difficult to handle it. On registering the devices we dont know what telegrams it sends and the logic is: if 0x10 is present, it's a remote.
We have to register all telegrams to this device and look later what it really sends.

akanto, 2022-01-29:
Yesterday, I went through the FW200 and also FB100 documentation, but I have not seen any sign of misconfiguration, actually, the configs are extremely limited. The FW200 is my master and an IPM module handles the hc1, and hc2,  and the other IPM also the FB100 is set to manage the heating circuit 3, and I have double-checked all wirings and everything and they look OK.

My heating system "just worked" in the last 6 years, so it seems like Junkers/Bosch either designed in this way, or there is some glitch somewhere that I can't find, but I am happy to experiment.

MichaelDvP, 2022-01-29:
I don't think it is amisconfiguration, but emsesp actually can not handle this configuration. 
Emsesp can handle it, if FW200 also controls hc3 and FB100 is configured as remote for hc3.
It is new to us, that a mixed configuration is possible. It's not a simple change to handle it in emsesp.

akanto, 2022-01-29:
Thanks for the explanation. 

> "FW200 also controls hc3 and FB100 is configured as remote for hc3"
Yes, exactly this was the scenario that I have looked up in the doc, and to be honest, I thought that this is what I have :). 

Sadly,  beyond setting the heating circuit id to 3 respectively on IPM (with the physical rotator selector) and also setting the id to 3 in the menu of FB100, I have not seen any more settings or even physical jumpers that could have any impact.

MichaelDvP, 2022-01-29:
I've downloaded the FW200 manual to look how to configure hc3, simple answer: it does only support hc1 and hc2. So your configuration is right and the only way. Astonishing how bosch always finds incompatible new solutions for each brand based on the same bus.
I think we have to find a solution to map hc3/4 data to the master thermostat and send the commands to the FB100. This will take some days, sorry.

akanto, 2022-01-29:
Many thanks for your support and for digging into this problem.

MichaelDvP, 2022-01-29:
Maybe a simple addition works. Seems that all supported Junkers thermostats does not have more than 2 circuits, 
Try this file, this should map the hc3 data to the masterthermostat. 
[EMS-ESP-3_4_0b2-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7964056/EMS-ESP-3_4_0b2-ESP32.zip)

akanto, 2022-01-29:
Thanks, I will try it, however, I think I have hit this issue: https://github.com/emsesp/EMS-ESP32/issues/325, and I can't upgrade through UI anymore.

Let me try OTA.

norberts1, 2022-01-29:
@MichaelDvP yes, the FW200 can only handle two heating-circuits, but there is no fix assignment to hc1 and hc2. It is configurable also to other hc-numbers, but at least only two circuits.
@MichaelDvP :    "FW200 also controls hc3 ..."
No, I don't think so. 
That FW200 get informations from FB100 for required room-temperatures. Together with that outdoor-temperatur the FW200 controls the boiler to generate the maximum required flow-temperatur for all 3 heating-circuits. That IPM-moduls (IPM2 for hc1 and hc2 and IPM1 for hc3) handles that for there heating-circuits.

Telegram 0x171 and others are not exclusive assigned to 'master'-moduls, they can be send also from other moduls (device-IDs) on the bus. There are 8 heating-circuits possible and they can send this telegram (only the heating-circuit byte differs).

This telegram 0x171 is send by the FB100 configured for hc3 in that system.
To send commands to the FB100 it's required to send them to the device-address 0x1A (Remote-controller for hc3) and not to the master-address 0x10.
I'll think decoding of telegram 0x171 := hc3 is the same as for 0x16F := hc1 and 0x170 :=hc2 independent from the sender-device-id.

akanto, 2022-01-29:
Thanks for this 2nd patch. It started to register hc3 under the master, and climate hc3 started to pop up in homeassistant, and values looked accurate.

However, with this new patch, I have completely lost the ability to control all thermostats, I have expected that probably I will have problems with controlling hc3, but unfortunately, this 2nd patch has broken hc1 and hc2 writes as well. Reads seems to be OK and accurate and fast for hc1/hc2. Kindly note that with the first patch, controlling hc1/hc2 just worked flawlessly.

I can send logs of patch2, if you wish, but as I understood @norberts1 comment, registering it under master might not be the right direction.

MichaelDvP, 2022-01-29:
@norberts1 Thanks.
> It is configurable also to other hc-numbers, but at least only two circuits.

Sad, hopefully noone do this. 
Since emsesp only supports one thermostat my approach is to check FW200 for hc1/2  (0x16F/0x165 and 0x170/0x166)) and send/receive telegrams 0x171/0x167 to devices 0x1A, 0x172/0x168 to 0x1B, etc but map the data to hc3, h4 of the first thermostat. Then thermostat_data lists all heatingcircuits on a single thermostat, but the bus-communication is to different ids.
I change the sender id of incoming telegrams to 0x10 for telegram 0x171 and the destination id to 0x1A  when sending 0x167.

> I'll think decoding of telegram 0x171 := hc3 is the same as for 0x16F := hc1 and 0x170 :=hc2 independent from the sender-device-address.

Yes, but that does not fit to emsesp data structure, the devices are sorted by sender address.

proddy, 2022-01-29:
> Since emsesp only supports one thermostat

the factory design pattern support having multiple device classes so in theory, we can have multiple independent thermostats. It would require some thinking on how to represent the data in MQTT and API but technically it's not too hard. And it would get rid of the nasty 'master thermostat' setting

MichaelDvP, 2022-01-29:
@akanto Yes, please send a log and set selected temp for all hc while logging. The patch should not affect hc1nd hc2 in any way. 
@proddy : Yes, like multiple mixers or solars, but thermostats are overlapping type-ids and device-ids, making difficult to seperate.

akanto, 2022-01-30:
Hmm, this is odd, now I cannot reproduce the issue anymore and with the patch2, I can control hc1, hc2, and even hc3. I have tried to switch between eco and auto mode a couple of times and it worked correctly, every time, also from ems-usp web UI and from HA.

```
{
  "datetime": "10:30:10 30.01.2022",
  "hc1": {
    "seltemp": 21.5,
    "currtemp": 23.9,
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 21.5,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "fb10",
    "program": "prog_e"
  },
  "hc2": {
    "seltemp": 22.5,
    "currtemp": 23.6,
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 22.5,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "off",
    "program": "prog_e"
  },
  "hc3": {
    "seltemp": 22,
    "currtemp": 23.7,
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 22,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "off",
    "program": "prog_c"
  }
}
```

For your reference, I have attached the logs, but it looks like patch2 works, and I don't know what has caused the control glitch that I have mentioned in my former comment.
[log.txt](https://github.com/emsesp/EMS-ESP32/files/7965637/log.txt)

Many thanks for the excellent support.

MichaelDvP, 2022-01-30:
Thanks for the log, there is a tx-error reported wrongly for telegram 0x167, causing that some settings of hc3 are not refreshed freuquently. This is the fix:
[EMS-ESP-3_4_0b2-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7965850/EMS-ESP-3_4_0b2-ESP32.zip)
Please update and send a new log to check.

akanto, 2022-01-30:
Please find the logs attached: [patch_3_logs.txt](https://github.com/emsesp/EMS-ESP32/files/7966093/patch_3_logs.txt)

MichaelDvP, 2022-01-30:
Thanks, now the bus communication is ok and telegrams are fetched correctly. There is only a wrong log message. I'll fix it and make a PR. 

@proddy 
> we can have multiple independent thermostats.

Yes, but mqtt and commands need a lot of rework and testing. Milestone for v3.5?

MichaelDvP, 2022-03-01:
I've made some experimental changes to remove the master-thermostat setting and support multiple thermostats native. In Web the values are shown on the right device. For mqtt and api values there should be no change. The `thermostat_data` topic collects all thermostat data and publishes in a single topic.  (Little change: the `emsesp/api/boiler/values` listes the ww entities in front.) 
I can only test with RC35 master and simulated hc2-remote thermostat RC20. 
@akanto: Can you check if this works with your configuration? 
@proddy There was a user with no 0x10 master but multible RC200 thermostats for different hc's (0x18, 0x19, ..), do you remember who it was?

source is [here](https://github.com/MichaelDvP/EMS-ESP32/tree/dev_no_master_thermostat), bin here:
[EMS-ESP-3_4_0b8-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/8163260/EMS-ESP-3_4_0b8-ESP32.zip)

akanto, 2022-03-01:
Thanks, I will take a look at it, and I will let you know the test results. By the way, the original patch that you have sent earlier had worked perfectly in the past month.

MichaelDvP, 2022-03-01:
> Is https://github.com/emsesp/EMS-ESP/issues/238 the issue you were referring too?

No, the configuration of lmdecroos is the one i've tesed. But i've found #139. @mtvw could you please test this software with your system?

@proddy: Possible configurations of multiple thermostats are:
1. master @ 0x10 and remotes @ 0x18,... (RC35+RC20 or RC3xx andRC200), remotes sending only a single telegram with measured roomtemp, but should not named roomtemp to avoid collisions.
2. master @0x10 with hc1-4 and additional thermostats on hc5... (0x1C..) The thermostat controlls a ne hc, there should be not conflict.
3. or junkers where master supports only hc1 and hc2 an extra thermostat on hc3 (@akanto config)
4. a combination of RC200  with 0x18 hc1 and ww, 0x19 hc2, ...

The remotes now sends `remotetemp` and the remote controlled hc should list in mqtt the roomTemp from master and the identical remotetemp from remote-thermostat. 

I've resorted the tags: TAG_BOILER_WW for the extra `boiler_data_ww` topic. TAG_DEVICE_WW to prefix the values with `dhw` published in `device_data` topic . TAG_DEVICE_DATA for all values that are not nested and published in devcies_data topic without prefix. TAG_NONE only for the nested collection of all values by device, not set  for the values itself.
`boiler_data` will collect no boiler-ww, but all nests with hs1..hs16  in systems with cascade module.

proddy, 2022-03-06:
> @proddy: Possible configurations of multiple thermostats are:

Any easy way to test is to compile with EMSESP_DEBUG flag, upload to a bare ESP32 and then invoke the test commands from the Console to simulate telegrams. e.g. `test 2thermostats` will add an RC310 and CR100 and throw in some data. See line 56 in test.cpp

> TAG_DEVICE_WW to prefix the values with dhw published in device_data topic

I don't mind renaming `ww` to `dwh` everywhere if it makes sense. It's a breaking change but since v3.4 is still in dev this is a good time to do it.

akanto, 2022-03-06:
I have built it from the source and uploaded it, and for the first look, it seems OK.

Logs: [log_patch-b8.txt](https://github.com/emsesp/EMS-ESP32/files/8193088/log_patch-b8.txt)

`
{
  "datetime": "18:18:00 06.03.2022",
  "hc1": {
    "seltemp": 22,
    "currtemp": 24.4,
    "climate": "roomTemp",
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 22,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "fb10",
    "program": "prog_e",
    "remotetemp": 24.4
  },
  "hc2": {
    "seltemp": 22.5,
    "currtemp": 24,
    "climate": "roomTemp",
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 22.5,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "off",
    "program": "prog_e"
  },
  "hc3": {
    "seltemp": 22,
    "currtemp": 23.8,
    "climate": "roomTemp",
    "mode": "auto",
    "modetype": "heat",
    "heattemp": 22,
    "ecotemp": 19,
    "nofrosttemp": 5,
    "control": "off",
    "program": "prog_c"
  }
}
`

MichaelDvP, 2022-03-07:
Thanks, yes the combined thermostat output looks good. In Web-UI and telnet show-command you should see now the values sorted to the different thermostats::
0x10 with hc1 and hc2, in hc1 there is no remote-temperature. 
0x18 with hc1 and only remote temperature
0x1A with hc3
This is how the thermostats manage the values.

commit 35013a71 (2022-02-05): Merge pull request #346 from MichaelDvP/dev_336

fix #336, map Junkers hc3/4 to masterthermostat
commit eaf651a4 (2022-02-05): fix #336, map Junkers hc3/4 to masterthermostat
closed 2022-05-23 by proddy