emsesp/EMS-ESP32 #340: Refresh dashboard values without reloading the page itself.

mvjt, 2022-01-29:
**Is your feature request related to a problem? Please describe.**
When I'm anazing all the values available at the dashboard, after some time, page refreshes and puts me at the top of the page again. This is slightly annoying as I have to scroll down trying to find where I was.

**Describe the solution you'd like**
Refresh values in the background.

MichaelDvP, 2022-01-31:
The devices list and entities are coupled to page load, but these are changing only in the first 2 minutes while scanning the bus. 
This approach is:
- if no device opened: refresh page
- if device/sensor is opend: refresh only this device
- refesh all on toggle
- refresh all on device change
- refresh all on refresh button

Please check: [EMS-ESP-3_4_0b2-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7969683/EMS-ESP-3_4_0b2-ESP32.zip)

mvjt, 2022-01-31:
Sounds good thank you

proddy, 2022-02-05:
thanks michael

commit 43d0197a (2022-02-05): Merge pull request #347 from MichaelDvP/dev_340

fix #340, refresh only device/sensor-data if open
commit 77df8cc6 (2022-02-05): fix #340, refresh only device/sensor-data if open
closed 2022-02-05 by proddy