emsesp/EMS-ESP32 #343: IVT Geo 508C, too many "No telegram header type..."

mvjt, 2022-01-30:
**Bug description**
EMS log is full with No telegram header type found. Some examples below:
2022-01-30 14:04:11.509 D 235: [emsesp] No telegram type handler found for ID 0x26D (src 0x10)
2022-01-30 14:04:11.911 D 236: [emsesp] No telegram type handler found for ID 0x26D (src 0x10)
2022-01-30 14:04:14.454 D 237: [emsesp] No telegram type handler found for ID 0x06 (src 0x50)
2022-01-30 14:04:15.614 D 238: [emsesp] No telegram type handler found for ID 0x4A0 (src 0x08)
2022-01-30 14:04:16.822 D 239: [emsesp] No telegram type handler found for ID 0x269 (src 0x10)
2022-01-30 14:04:16.859 D 240: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 14:04:17.271 D 241: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 14:04:17.309 D 242: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 14:04:17.709 D 243: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 14:04:17.914 D 244: [emsesp] No telegram type handler found for ID 0xF7 (src 0x08)
2022-01-30 14:04:18.271 D 245: [emsesp] No telegram type handler found for ID 0x26C (src 0x10)
2022-01-30 14:04:18.309 D 246: [emsesp] No telegram type handler found for ID 0x26C (src 0x10)
2022-01-30 14:04:18.346 D 247: [emsesp] No telegram type handler found for ID 0x26D (src 0x10)
2022-01-30 14:04:18.759 D 248: [emsesp] No telegram type handler found for ID 0x26D (src 0x10)
2022-01-30 14:04:19.171 D 249: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 14:04:19.571 D 250: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 14:04:19.609 D 251: [emsesp] No telegram type handler found for ID 0x26C (src 0x10)
2022-01-30 14:04:20.021 D 252: [emsesp] No telegram type handler found for ID 0x26D (src 0x10)
2022-01-30 14:04:20.099 D 253: [emsesp] No telegram type handler found for ID 0xF7 (src 0x08)
2022-01-30 14:04:21.024 D 254: [emsesp] No telegram type handler found for ID 0xF7 (src 0x08)
2022-01-30 15:47:25.724 D 328: [emsesp] No telegram type handler found for ID 0x4A0 (src 0x08)
2022-01-30 15:47:29.031 D 329: [emsesp] No telegram type handler found for ID 0x293 (src 0x10)
2022-01-30 15:47:29.431 D 330: [emsesp] No telegram type handler found for ID 0x294 (src 0x10)
2022-01-30 15:47:30.231 D 331: [emsesp] No telegram type handler found for ID 0x293 (src 0x10)
2022-01-30 15:47:30.269 D 332: [emsesp] No telegram type handler found for ID 0x294 (src 0x10)
2022-01-30 15:47:30.352 D 333: [emsesp] No telegram type handler found for ID 0x06 (src 0x50)
2022-01-30 15:47:35.811 D 339: [emsesp] No telegram type handler found for ID 0x4A0 (src 0x08)
2022-01-30 15:47:36.656 D 340: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 15:47:37.059 D 341: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 15:47:37.631 D 342: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 15:47:37.669 D 343: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 15:47:38.081 D 344: [emsesp] No telegram type handler found for ID 0x26A (src 0x10)
2022-01-30 15:47:38.481 D 345: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 15:47:38.881 D 346: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 15:47:38.919 D 347: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 15:47:39.344 D 348: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 15:47:39.744 D 349: [emsesp] No telegram type handler found for ID 0x26B (src 0x10)
2022-01-30 15:47:39.781 D 350: [emsesp] No telegram type handler found for ID 0x26C (src 0x10)
2022-01-30 15:47:40.194 D 351: [emsesp] No telegram type handler found for ID 0x26C (src 0x10)
2022-01-30 15:47:40.606 D 352: [emsesp] No telegram type handler found for ID 0x26C (src 0x10)
2022-01-30 15:47:41.006 D 353: [emsesp] No telegram type handler found for ID 0x26C (src 0x10)

**Steps to reproduce**
Connect IVT Geo 508C (OEM Bosch 7800) 

**Expected behavior**
I expect EMS-ESP32 to translate unknown telegrams to entities/values.

**Screenshots**
*If applicable, add screenshots to help explain your problem.*
[image]

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api?device=system&cmd=info*
{
  "System": {
    "version": "3.4.0b2",
    "uptime": "000+05:05:59.041",
    "uptime (seconds)": 18359,
    "freemem": 108,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "8C:CE:4E:95:50:97",
    "IPv4 address": "10.10.10.121/255.255.255.0",
    "IPv4 gateway": "10.10.10.1",
    "IPv4 nameserver": "1.1.1.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 84511,
    "bus reads (tx)": 2395,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 5963,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 1,
    "API fails": 1
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "IVT Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx (DeviceID:0x08, ProductID:172, Version:02.03)",
      "entities": 59,
      "handlers received": "0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "Rego 2000/3000 (DeviceID:0x10, ProductID:172, Version:02.03)",
      "entities": 21,
      "handlers received": "0x06 0x02A5 0x02A6",
      "handlers fetched": "0x02B9 0x0471 0x02BA 0x02B0 0x029C 0x0472 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x02AF 0x029B 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)",
      "entities": 0
    },
    {
      "type": "Controller",
      "name": "Rego 3000 (DeviceID:0x09, ProductID:240, Version:38.03)",
      "entities": 0
    },
    {
      "type": "Connect",
      "name": "Wireless sensor base (DeviceID:0x50, ProductID:236, Version:03.03)",
      "entities": 0
    },
    {
      "type": "Unknown",
      "name": "unknown (DeviceID:0x40)",
      "entities": 0,
      "handlers received": "0x0435"
    }
  ]
}

**Additional context**

MichaelDvP, 2022-01-31:
Hmm, it's not so much unknown. 0xF7and 0xF9 are configurations, not data. There are much 0x26A..F, empty telegrams from thermostat to km200. these are junkers thermostat settings and your thermostat does not support these telegrams, but seems km200 queries them. 
If i remove all this trash we have: 
- `Boiler(0x08) -> All(0x00), ?(0x04A0), data: 00 00 00 00` many times with always the same zero values
- `Boiler(0x08) -> All(0x00), ?(0x04A2), data: 90 00 00 00 00 01 00 00` also unchanged,
- `Boiler(0x08) -> All(0x00), ?(0x0489), data: 80 00 00 00 00 00 00 00 00 3A (offset 4)` starts with 80 00,  probably a unconnected temperature sensor, the last byte changes after a minute from 3A to 2D,  was there a temperature changing from 5.8 to 4.5 °C? Something in the outdoor unit?
- `Boiler(0x08) -> All(0x00), ?(0x0497), data: 01 68 01 F4 01 F0 01 F9 02 44`, offset 5 changes after 77 sec from F0 to F1, all other bytes unchanged. looks like status bits published on change and not frequently.

Unknown device id 0x40 is a "rf room temperature sensor" and should be named and listed as generic. The handler contains only the room temperature and is already lsited in thermostat.

>  I expect EMS-ESP32 to translate unknown telegrams to entities/values.

Difficult to tranlate something unknown.

mvjt, 2022-01-31:
Hi,

"Boiler(0x08) -> All(0x00), ?(0x0489), data: 80 00 00 00 00 00 00 00 00 3A (offset 4) starts with 80 00, probably a unconnected temperature sensor, the last byte changes after a minute from 3A to 2D, was there a temperature changing from 5.8 to 4.5 °C? Something in the outdoor unit?"

Yes outdoor temperature was in that range so it is probably the outdoor sensor. 

Thank for clarifying the rest. Appreciated.

mvjt, 2022-02-06:
One entity I miss is "hc1 room influence" which is set to 3 in the HP...Could that be one of them? Below are unknown and during that sequence I changed from 3->4->3

```
Watching unknown telegrams
000+01:36:51.240 N 2: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Summer(0x02B0), data: <empty>
000+01:36:51.717 N 3: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:36:53.030 N 4: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:36:55.827 N 5: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Set(0x02BB), data: <empty>
000+01:36:55.940 N 6: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Monitor(0x02A7), data: <empty>
000+01:36:56.332 N 7: [emsesp] Boiler(0x08) -> All(0x00), ?(0x04A0), data: 00 00 00 00
000+01:36:56.511 N 8: [emsesp] Boiler(0x08) -> All(0x00), ?(0x04A2), data: 90 00 00 00 00 01 00 00
000+01:36:57.630 N 9: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:36:58.677 N 10: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Curves(0x029D), data: <empty>
000+01:36:58.998 N 11: [emsesp] Thermostat(0x10) -> Controller(0x09), ?(0xF9), data: FF 01 AF 01 07 00 00 00 00 00 00 00 28 00 00 00 64 00 00 00 28
000+01:36:59.177 N 12: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Summer(0x02B1), data: <empty>
000+01:37:00.440 N 13: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Summer2(0x0473), data: <empty>
000+01:37:00.618 N 14: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:01.643 N 15: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:02.165 N 16: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Set(0x02BC), data: <empty>
000+01:37:02.805 N 17: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:03.265 N 18: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Monitor(0x02A8), data: <empty>
000+01:37:03.609 N 19: [emsesp] Boiler(0x08) -> All(0x00), ?(0x04A0), data: 00 00 00 00
000+01:37:04.668 N 20: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:06.261 N 21: [emsesp] Connect(0x50) -> All(0x00), RCTime(0x06), data: 16 02 15 06 0D 38 06 00 50 FF 00
000+01:37:07.228 N 22: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Curves(0x029E), data: <empty>
000+01:37:07.677 N 23: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Summer(0x02B2), data: <empty>
000+01:37:07.792 N 24: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:08.170 N 25: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: FF 03 88 4F 00 00
000+01:37:08.408 N 26: [emsesp] Boiler(0x08) -> All(0x00), ?(0x0488), data: 80 00 00 61 00 00 01 00 00 00 00 5A 00 14 00 00 64
000+01:37:09.290 N 27: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300Summer2(0x0474), data: <empty>
000+01:37:09.607 N 28: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:09.985 N 29: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0xF9), data: EA 00 00 12 07 00 00 00 28 00 00 00 32 00 00 00 32 00 00 00 32
000+01:37:10.423 N 30: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0xF9), data: EA 00 00 06 07 00 00 00 28 00 00 00 34 00 00 00 34 00 00 00 34
000+01:37:11.618 N 31: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:13.357 N 32: [emsesp] Boiler(0x08) -> All(0x00), ?(0x04A0), data: 00 00 00 00
000+01:37:13.765 N 33: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300WWtemp(0x031B), data: <empty>
000+01:37:14.680 N 34: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:15.607 N 35: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:16.630 N 36: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:17.717 N 37: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:18.630 N 38: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:18.928 N 39: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 2)
000+01:37:21.190 N 40: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 10)
000+01:37:21.590 N 41: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 5)
000+01:37:21.990 N 42: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 7)
000+01:37:22.390 N 43: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 6)
000+01:37:22.855 N 44: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:23.340 N 45: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 3)
000+01:37:23.682 N 46: [emsesp] Boiler(0x08) -> All(0x00), ?(0x04A0), data: 00 00 00 00
000+01:37:24.643 N 47: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:25.607 N 48: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:26.607 N 49: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:28.618 N 50: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
000+01:37:29.630 N 51: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
```

MichaelDvP, 2022-02-07:
Sorry, the roominfluence of hc1 should be in telegram 0xAF. What does it show on `call thermostat info`?  
Better for determine the value is to set the watch to the thermostat `watch 10`, then change the value and look for telegrams that publishes a single offset, e.g. `Thermostat(0x10)->All(0x00) ... (offset xy)`

mvjt, 2022-02-07:
Does not seem the thermostat is sending that.

```
ems-esp:# call thermostat info
{
  "damped outdoor temperature": 2.5,
  "building type": "medium",
  "minimal external temperature": -18,
  "dhw mode": "low",
  "dhw circulation pump mode": "off",
  "dhw charge duration": 60,
  "dhw charge": "off",
  "dhw disinfecting": "off",
  "dhw disinfection day": "so",
  "dhw disinfection time": 120,
  "hc1 selected room temperature": 22,
  "hc1 current room temperature": 21.1,
  "hc1 mode": "manual",
  "hc1 mode type": "eco",
  "hc1 eco temperature": 15,
  "hc1 manual temperature": 22,
  "hc1 comfort temperature": 21,
  "hc1 summer temperature": 18,
  "hc1 target flow temperature": 0,
  "hc1 set summer mode": "auto",
  "hc1 summer mode": "off"
}
```

During the following sequence I changed room influence from 3->5->3
```
ems-esp:# watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x10
001+01:05:35.269 N 1: [emsesp] Controller(0x09) <- Thermostat(0x10), ?(0xF9), data: 11 FF 01 AF 01
001+01:05:35.307 N 2: [emsesp] Thermostat(0x10) -> Controller(0x09), ?(0xF9), data: FF 01 AF 01 07 00 00 00 00 00 00 00 28 00 00 00 64 00 00 00 1E
001+01:05:36.478 N 3: [emsesp] Controller(0x09) -> Thermostat(0x10), RC300Summer(0x02AF), data: 32 (offset 1)
001+01:05:37.482 N 4: [emsesp] Controller(0x09) <- Thermostat(0x10), ?(0xF9), data: 11 FF 01 AF 01
001+01:05:37.520 N 5: [emsesp] Thermostat(0x10) -> Controller(0x09), ?(0xF9), data: FF 01 AF 01 07 00 00 00 00 00 00 00 28 00 00 00 64 00 00 00 32
001+01:05:39.245 N 6: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x02F6), data: 01 (offset 2)
001+01:05:39.262 N 7: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 2)
001+01:05:39.904 N 8: [emsesp] Controller(0x09) -> Thermostat(0x10), RC300Summer(0x02AF), data: 1E (offset 1)
001+01:05:41.093 N 9: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x02F6), data: 01 (offset 11)
001+01:05:41.113 N 10: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 11)
001+01:05:41.501 N 11: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x02F6), data: 01 (offset 10)
001+01:05:41.512 N 12: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 10)
001+01:05:41.533 N 13: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x02F6), data: 01 (offset 5)
001+01:05:41.549 N 14: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 5)
001+01:05:42.093 N 15: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x02F6), data: 01 (offset 7)
001+01:05:42.113 N 16: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 7)
001+01:05:42.501 N 17: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x02F6), data: 01 (offset 6)
001+01:05:42.512 N 18: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 6)
001+01:05:43.053 N 19: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x02F6), data: 01 (offset 3)
001+01:05:43.062 N 20: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x02F6), data: <empty> (offset 3)
001+01:05:51.645 N 21: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x0293), data: 01
001+01:05:51.664 N 22: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x0293), data: <empty>
001+01:05:51.685 N 23: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x0294), data: 01
001+01:05:51.699 N 24: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x0294), data: <empty>
```

mvjt, 2022-02-07:
Seems it is not transmitted on the bus.

MichaelDvP, 2022-02-08:
I think you are too fast, there is a message, but from controller to thermostat. That controller is interface for thermostat is uncommon, but possible.
```
001+01:05:36.478 N 3: [emsesp] Controller(0x09) -> Thermostat(0x10), RC300Summer(0x02AF), data: 32 (offset 1)
001+01:05:39.904 N 8: [emsesp] Controller(0x09) -> Thermostat(0x10), RC300Summer(0x02AF), data: 1E (offset 1)
```
data 0x32 = 50, data 0x1E = 30, s we have a factor 10 in that value.
In [NorbertS' list](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf) there is the roominfluence in offset 0, where we read it, and offset 1 is named as `Control factor of roominfluence`. Can you post the complete telegram, `read 10 2AF` and describe what exactly you set on your contoller: name and unit Is it roominfluence in degrees C, or different? Are there other parameters for roominfluence?

mvjt, 2022-02-19:
This now works! Control factor is shows properly!

closed 2023-02-19 by proddy