emsesp/EMS-ESP32 #345: Sometimes firmware updates reset the board profile on the E32

bbqkees, 2022-01-30:
I get regular emails that when someone updates the firmware on an E32, that the board profile resets to S32.
The problem here is that you loose the Ethernet connection and you have to go to the boiler, connect to the AP and set it to E32 again.

This happens I think when updating from 3.x to 3.4 and of course if you flash it via USB.

Would it be an idea to add a second bin file on the releases page with E32 as the default board profile?
So in case something goes wrong, the default profile is E32 and you can still connect remotely via Ethernet.

I was thinking to hardwire a product ID on two or three GPIO in future revisions, in this case EMS-ESP can check on boot which board it is running on.

proddy, 2022-01-30:
Interesting, I haven't seen this behavior. The board profile logic changed in v3.4 to support extra parameters but didn't expect it to not recognize previous settings. Have you been able to reproduce this?

MichaelDvP, 2022-01-31:
@proddy  You have renamed the json from ["code"](https://github.com/emsesp/EMS-ESP32/blob/29110e96e53ff8ebb006b44a74aaa19d2bb2dd04/src/web/WebSettingsService.cpp#L274) to[ "board_profile"](https://github.com/emsesp/EMS-ESP32/blob/3d1a050e22153a0fc947dac0d8dae01f5b4f663f/src/web/WebSettingsService.cpp#L287) in the settings file.

bbqkees, 2022-01-31:
Here I did not encounter these problems myself.
It's always difficult to get to the exact reason why someones' Gateway had this reset. It looks like there are several unrelated causes.
Most often they have no idea how they ended up in the reset.

bbqkees, 2022-01-31:
I just got an email from a user who had this issue after updating from 3.2.x to 3.3.1.

pypb, 2022-01-31:
I encountered the same issue going from 3.3.0 to 3.3.1, profile was reset to S32.

mvjt, 2022-01-31:
Same here when going from 3.3.x to 3.4.

proddy, 2022-02-05:
ok, so I need to test this and load up an 3.3 onto an ESP32 and set the boardtype to E32, then upgrade to 3.4 and see if and why it switches back to S32.

proddy, 2022-02-06:
Can't reproduce this, but added some improved logging to catch it if it happens again. Would be good to know from the users experiencing this how they are upgrading, via ota, via browser (which) and which OS

mvjt, 2022-02-06:
> Can't reproduce this, but added some improved logging to catch it if it happens again. Would be good to know from the users experiencing this how they are upgrading, via ota, via browser (which) and which OS

In my case I did the following
1. Connected my NEW out of the box E32 to the network (cable)
2. Saw it had FW 3.3.x, downloaded the 3.4 beta and uploaded it (using MacOS, Safari which we know has issues). 
3. Device rebooted and was lost on the network (cable), did not think about checking if AP was broadcasting as I was not using it
4. Firmware upgraded using serial cable, still no Ethernet
5. Saw that SSID was broadcasting, changed device profile and everything started to work again.

proddy, 2022-02-06:
I suspect its related to that previous OSX firmware uploaded bug where it would try and upload a broken .bin file and it would remove part of the flash memory which has the board profile setting. If it happens again let me know!

MichaelDvP, 2022-02-07:
I had this one time, but not sure after what update, because only the project settings were on defaults, wifi, mqtt, ntp, etc. were unchanged.  It was surely a update v3.4 to v3.4 with little changes. There was a upload error http:500 in one update, second upload worked and i suspect this 500 was the cause for deleting the project file. 
A test with 3.1, 3.2, 3.3, 3.4 updating works without changing data. (i've loaded factory settings and configure manual in 3.1 to have a clear start).

bbqkees, 2022-02-07:
I don't think we can really prevent it from ever happening.
However, when it happens on a wired E32 it's very inconvenient or sometimes maybe even problematic.

Is there any failsafe we can think of?

MichaelDvP, 2022-02-07:
Don't know if it is competly failsafe and if it works. Just an idea:
There is a nv storage in esp32, [see here](https://github.com/espressif/esp-idf/blob/master/examples/storage/nvs_rw_value/main/nvs_value_example_main.c). I think nvs survives the OTA/webupdate. On first emsesp-config store the ethernet module to nvs (only if the value is  not existing). On start load the default from nvs and if settingsfile is missing/broke/reseted this default eth-module is set. If the nvs-value exists, it is never written again. (except reformatting/repartitioning/flash-erase).

proddy, 2022-02-07:
I really would like to know the root cause. Now after a recent change if the board profile is reset by chance during a f/w upload it'll be shown in the log, so we can trace it back. And instead of resetting to the S32 it will go to CUSTOM. I think we can close this and see what happens.

closed 2022-02-12 by proddy