emsesp/EMS-ESP32 #350: re previous bug

florin-laurentiu, 2022-02-05:
hello,
just install last developing firmware; remain all problems, except that HC 1 thermostat can be controlled by HA.
please, see attachements:
[image]
[image]
[image]
[image]

proddy, 2022-02-05:
what is the bug exactly?

florin-laurentiu, 2022-02-05:
see in attached pictures:
1.- some "unknown datafield" that appear 4 times in SOLAR in EMS as also on
HA.
2.- after new firmware , even i redefine DALLAS sensors - put again names -
, they appears in HA but no indication/values; all times VALUES for them is
ZERO.

proddy, 2022-02-06:
1 - is by design. if you don't like it open a new enhancement request and Michael will remove it
2 - create a new GH issue and describe step by step how to reproduce this, because it works for everyone else

closed 2022-02-06 by proddy