emsesp/EMS-ESP32 #353: Show warning on NTP page when server is not reachable

mvjt, 2022-02-06:
**Is your feature request related to a problem? Please describe.**
When the NTP server is not reachable, it is not visible on the NTP page. It makes it hard to know when it does not work.

**Describe the solution you'd like**
Warning message on the NTP page when sync fails. Or even better a "Test" button on the Settings page.

**Describe alternatives you've considered**
Using Wireshark, but it is much more complicated.

proddy, 2022-02-06:
I looked into this and there's no easy way to figure out if the NTP time was sync'd or not. I'll add a note to the documentation stating if you see the year 1970 after hitting the refresh button in the WebUI then the server can't be reached.

mvjt, 2022-02-07:
I’m OK with any decision but what If you just compare current UTC time from the device with UTC.Now when the page is rendered?Sent from ProtonMail for iOS På sön,

proddy, 2022-02-07:
the ESPs don't work that way. The time is only set once so now() will return the UTC time. You could compare dates < 2022 and report and error, but this is not very useful since it's only calculated when the user is actually on the page during rendering. And showing a warning won't add much (since it'll show the wrong date/year anyway). It's not worth the time to make it better. But feel free to experiment and do a PR though!

closed 2022-02-07 by proddy