emsesp/EMS-ESP32 #356: Entity (hc1 current room temperature) does not update unless triggered in EMS UI

mvjt, 2022-02-13:
**Bug description**
hc1 current room temperature (maybe more) are not updating unless I go into the EMS UI and refresh/read from the bus. I'm plotting the entity through HASS and as the screenshot below shows, it only updates when I manually trigger it in the EMS UI (which I've done two days in a row morning time). The hc1 current room temperature value should come from the "Rf room temperature sensor". The RF Room sensor entity is listed under "Unknown" but has the correct value. When I login to EMS UI, I can see that RF Room Sensor temp is correct, but hc1 current room temperature is incorrect/stuck. When I've refreshed, the hc1 current room temperature is updated to the correct value both in EMS and in HASS.

**Steps to reproduce**
Not sure but it has been like this since I installed the GW a couple a week ago.

**Expected behavior**
Updates provided with requiring me to manually refresh the entity in EMS UI.

**Screenshots**
[image]

**Device information**
```
{
  "System": {
    "version": "3.4.0b4",
    "uptime": "001+00:36:47.504",
    "uptime (seconds)": 88607,
    "freemem": 115,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "8C:CE:4E:95:50:97",
    "IPv4 address": "10.10.10.121/255.255.255.0",
    "IPv4 gateway": "10.10.10.1",
    "IPv4 nameserver": "10.10.10.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 415755,
    "bus reads (tx)": 23688,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 1,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 28584,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 0,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "IVT Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx (DeviceID:0x08, ProductID:172, Version:02.03)",
      "entities": 94,
      "handlers received": "0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "Rego 2000/3000 (DeviceID:0x10, ProductID:172, Version:02.03)",
      "entities": 48,
      "handlers received": "0x06 0x02A5 0x02A6",
      "handlers fetched": "0x02B9 0x02AF 0x029B 0x0471 0x02BA 0x02B0 0x029C 0x0472 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.08)",
      "entities": 0
    },
    {
      "type": "Controller",
      "name": "Rego 3000 (DeviceID:0x09, ProductID:240, Version:38.03)",
      "entities": 0
    },
    {
      "type": "Connect",
      "name": "Wireless sensor base (DeviceID:0x50, ProductID:236, Version:03.03)",
      "entities": 0
    },
    {
      "type": "Unknown",
      "name": "rf room temperature sensor (DeviceID:0x40)",
      "entities": 1,
      "handlers received": "0x0435"
    }
  ]
}
```

MichaelDvP, 2022-02-13:
Download actual dev, proddy has fixed this a few hours ago.

proddy, 2022-02-13:
@MichaelDvP shall we pick a better name for the RF Temp Sensor instead of "Unknown"? Also pick a nice icon for the webUI from https://react-icons.github.io/react-icons

mvjt, 2022-02-13:
I downloaded and installed b5 and the issue is the same

mvjt, 2022-02-13:
I have recorded it on video, but it is 29 MB and I cannot attach more than 10 MB. Let me know if needed and how you want me to share it.

closed 2022-02-13 by mvjt

proddy, 2022-02-13:
> The hc1 current room temperature value should come from the "Rf room temperature sensor". 

EMS-ESP doesn't support this in v3.4. I take it http://ems-esp.local/api/thermostat doesn't show a `currtemp` ?

proddy, 2022-02-13:
there might be a discrepancy between the temps due to a rounding error, so I wouldn't worry too much about that now. What you could try is using MQTTExplorer, going to the thermostat_data topic and under Value click the ~ next to currtemp and see how often it gets updated which should be every 10 seconds. Also note a refresh from the WebUI only refreshes the UI, it doesn't fetch any data from the EMS bus.

mvjt, 2022-02-13:
New observations:
* I changed the `seltemp` on the HP. When I looked at the EMS UI `currtemp` had updated without GW restart, but `seltemp` was incorrect. Restarting the GW and `seltemp` also updated.

Maybe `currtemp` was updated when I change the `seltemp` on the HP.

proddy, 2022-02-13:
the currtemp (also known as roomTemp) should come from the Telegram 0x2A5 which is broadcasted every 10 seconds I think. Try going to the Telnet console and typing `watch 2a5` and wait until you have captured a few telegrams and post them back here. The temperature should be in the first few bytes of the data block.

mvjt, 2022-02-13:
OK so this is really weird. It seems the HP stops transmitting the temperature value after the RF Sensor has reported an updated temperature. Nothing came in an hour after the RF Temp changed.

It seems like a bug with the HP, but why does it work after GW restart? Is it because "you" manually ask the HP for the value?

MichaelDvP, 2022-02-13:
>shall we pick a better name for the RF Temp Sensor instead of "Unknown"?

That's a fixed bug, it is `generic` and waits for telegram 0x435. (see #103, [here](https://github.com/emsesp/EMS-ESP32/blob/17bc9c231a063e140ce93f11c1fa2f219072c0c2/src/emsesp.cpp#L1090)) and [here](https://github.com/emsesp/EMS-ESP32/blob/17bc9c231a063e140ce93f11c1fa2f219072c0c2/src/devices/generic.cpp#L30).

The roomtemperature is republished by the thermostat. I expect the rf-sensor sending very rarly, most battery operated rf-sensors sending only on change with hysteresis. 
To check  set a watch to 0x435 and log the incoming data. After checking normal refreshrate heat up thesensor by hands or breath and see if react faster.
The refresh button in web-UI only updates the web from emsesp, it does not trigger a read on the bus. Also the rf-sensor is not readable, we have to wait for the bus-publish.

MichaelDvP, 2022-02-13:
> It seems like a bug with the HP, but why does it work after GW restart? Is it because "you" manually ask the HP for the value?

Yes, after restart all hcs are queried by emsesp to read the values, after this read the monitor-telegram in not fetched, only the settings because monitor is broadcasted. If your IVT does not broadcast the 0x2A5, 0x2A6, .. we need to add a special thermostat mode with fetched monitor-telegrams (updated every minute), or add the fetch to all thermostats. 
Add [here](https://github.com/emsesp/EMS-ESP32/blob/17bc9c231a063e140ce93f11c1fa2f219072c0c2/src/devices/thermostat.cpp#L346) a line `toggle_fetch(monitor_typeids[hc_num - 1], toggle_);` and check if it works.

mvjt, 2022-02-13:
Hi,

It does send it, but VERY sporadically. The RF Sensor updates much quicker. It sent once, then after 4 minutes again, and then after a while twice in 5 seconds...And suddenly it stops. Seems like a bug in HP FW to me.

I´ll test more in a bit and come back. Thanks guys

mvjt, 2022-02-13:
I noticed it started to talk more since I connected the Ethernet cable to the HP...Once every 5 minutes.

mvjt, 2022-02-13:
And not it has stopped publishing again...Really weird...

mvjt, 2022-02-13:
Now I know what causes the thermostat to start publishing again. Its when I wake it up on the touch screen. I guess when it has been asleep for a while it stops sending unless being asked?

proddy, 2022-02-13:
wow, ok, kinda makes sense

mvjt, 2022-02-14:
Confirmed this morning again. See the end of the graph where I just pressed the touch display.
[image]

mvjt, 2022-02-18:
> > It seems like a bug with the HP, but why does it work after GW restart? Is it because "you" manually ask the HP for the value?
> 
> Yes, after restart all hcs are queried by emsesp to read the values, after this read the monitor-telegram in not fetched, only the settings because monitor is broadcasted. If your IVT does not broadcast the 0x2A5, 0x2A6, .. we need to add a special thermostat mode with fetched monitor-telegrams (updated every minute), or add the fetch to all thermostats. Add [here](https://github.com/emsesp/EMS-ESP32/blob/17bc9c231a063e140ce93f11c1fa2f219072c0c2/src/devices/thermostat.cpp#L346) a line `toggle_fetch(monitor_typeids[hc_num - 1], toggle_);` and check if it works.
> 
> @proddy the monitor-fetch was in v3.3, and is missing in v3.4, maybe better to add it back for all themrostats. If bosch makes one thermostat not broadcasting it frequently, there will be others.

Understand you guys are busy but let me know if something is needed from my side in regards to this "monitor-fetch"...