emsesp/EMS-ESP32 #357: Excluding entities from the Customization page does not remove the MQTT Discovery entry

proddy, 2022-02-13:
As reported by @mvjt in [Discord](https://discord.com/channels/816637840644505620/816638587508490250/941978699102838795) :
> 2. Disabling an entity on the Customize page in EMS will start generating issues (not part of dictionary". (Re-enabling the entity will require EMS-GW restart) 

Confirmed that this is a bug. For example disabling gas, gas stage 2 and flame current will fill the HA log with:

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1822
First occurred: 16:03:59 (72 occurrences)
Last logged: 16:07:49

Template variable warning: 'dict object' has no attribute 'burngas' when rendering '{{value_json.burngas}}'
Template variable warning: 'dict object' has no attribute 'burngas2' when rendering '{{value_json.burngas2}}'
Template variable warning: 'dict object' has no attribute 'flamecurr' when rendering '{{value_json.flamecurr}}'
```

proddy, 2022-02-13:
found the issue and will try and do a fix tonight, otherwise next weekend as I'm out on the road again

proddy, 2022-02-13:
Think this is fixed now. Will leave open and do some more testing. in b5

mvjt, 2022-02-13:
Re-downloaded b5 (downloaded it earlier today). I now get the error message ONCE in HASS, but it does not continue every 10s. Good enough for me :-) Thanks a lot!

proddy, 2022-02-13:
yes, I can see why that would happen. When a device entity is excluded (e.g. gas) it first gets removed from the boiler_data MQTT topic and then a fraction of a second later the old HA MQTT config topic is removed. So you'll see the warning in HA only once when an entity is selected for exclusion in the WebUI. I should tidy the code up a little to prevent that at some point. Let me know if you find anything else odd.

proddy, 2022-02-17:
closing this bug as the problem is fixed and will create another one for improving the logic

closed 2022-02-17 by proddy