emsesp/EMS-ESP32 #368: Excluding thermostat entities should also remove the climate discovery topic

proddy, 2022-02-19:
As reported by eltoro on Discord, the MQTT Discovery climate component needs seltemp (selected temperature), room temperature and mode to render correctly. If any of these entities are disabled/excluded in EMS-ESP via the WebUI customization page, HA will report warnings constantly.

The fix is to remove the MQTT /climate/config topic for a heating circuit if the required entities are not active (i.e have no value) or marked as excluded.

MichaelDvP, 2022-02-23:
I'm testing to move the HA climate from thermostat to mqtt. Depending on roomTemp active the climate is created with or without currtemp and disable roomtemp in customizations also disables climate. 
It's a suggestion,i'll make a PR, check first if you find it usefull.

proddy, 2022-02-23:
thanks, it would be easier when moving to the mqtt class as it exposes the DeviceValue class (dv) and you can check the flags like ACTIVE and VISIBLE. I'm thinking of extending the flags to support that other enhancement so you can choose per device entitity whether to exclude it from API, MQTT & Web/Console

proddy, 2022-02-26:
@MichaelDvP implemented this, I've tested and it works flawlessly., Nice work Michael! I think we can close this, bump to the version to b8

closed 2022-03-05 by proddy