emsesp/EMS-ESP32 #371: Climate.hc1 thermostat goes missing after ~24 hrs in HomeAssistant

stefboerrigter, 2022-02-21:
**Bug description**
After roughly 24 hours HomeAssistant reports that my thermostat is missing.
**EMS-ESP32 version**:v3.4.0b6

**Steps to reproduce**
Install mentioned ESP32 version and wait ~24 hours for the node to go missing in HomeAssistant

**Expected behavior**
The climate.hc1 thermostat to be alive continuously so i have an thermostat in HomeAssistant.

**Screenshots**
HomeAssistant affected node: 
[image]
Thermostat still alive on EMS bus
[image]
MQTTExplorer BEFORE issue is observed :
[image]
MQTTExplorer AFTER issue is observed:
[image]
The MQTTExplorer images show some topics to be missing after a while (no updates after xx Minutes / hours?)

**Device information**
`{
  "System": {
    "version": "3.4.0b6",
    "uptime": "003+06:51:31.772",
    "uptime (seconds)": 283891,
    "freemem": 164,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp32",
    "SSID": "FamBoerrigter",
    "BSSID": "98:DA:C4:3F:4B:1E",
    "RSSI": -45,
    "MAC": "C8:C9:A3:C9:6E:5C",
    "IPv4 address": "192.168.188.25/255.255.255.0",
    "IPv4 gateway": "192.168.188.1",
    "IPv4 nameserver": "192.168.188.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 208816,
    "bus reads (tx)": 52073,
    "bus writes (tx)": 7,
    "bus incomplete telegrams": 5,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 92264,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 0,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Ecomline Excellent",
      "device id": "0x08",
      "product id": 206,
      "version": "01.08",
      "entities": 61,
      "handlers received": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xC2 0x15 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC20/Moduline 300",
      "device id": "0x17",
      "product id": 77,
      "version": "03.03",
      "entities": 10,
      "handlers received": "0xA3 0x06 0xA2",
      "handlers fetched": "0x91 0xA8 0x90 0x8F",
      "handlers pending": "0x12 0xAF"
    },
    {
      "type": "Controller",
      "name": "Ecomline",
      "device id": "0x09",
      "product id": 206,
      "version": "01.08",
      "entities": 0
    }
  ]
}`

**Additional context**
I don't think it is relevant because in my opinion the issue is only MQTT related, but i do have an iRT <> EMS adapter in between.

proddy, 2022-02-21:
this may be related to a recent change. There's a PR in discussion at the moment about this. If either the seltemp or currtemp become invalid it will remove the HA climate component.

stefboerrigter, 2022-02-21:
I have noticed this already in previous versions, hence my (the latest) comments in #196
This is in the main branch and also in latest EMS-ESP (non ESP32 version) already similar behavior in my opinion (But not guaranteed).
Should the topic be updated every N iterations?

proddy, 2022-02-21:
It would be interesting to see when the thermostat_data can't read seltemp or currtemp (both needed for HA to work), you're the only one to report it so it may be iRT related?

stefboerrigter, 2022-02-21:
all other temperatures are perfectly readable over this temperature:
[image]
See also seltemp and curretemp in MQTTExplorer screenshot prior to the issue (seltemp and curretemp visible)

proddy, 2022-02-21:
no and no, it's only needed to be created once.

From your screenshot above, around 8:15pm the thermostat setpoint temperature (doeltemperatuur) gets a funny value. EMS-ESP sees this as a lost entity and promptly removes the HA climate control. Try and see why hc1.seltemp is doing

stefboerrigter, 2022-02-21:
You mean the setpoint to 21.5 in the HA climate value? This is me fiddling with the setpoint to see if it's still there in the HomeAssistant UI. I've restarted the node to have clear logging from today and i'll see in about a day if it has the same behavior. 
So your guess; 
EMS-ESP thinks it loses the entity and removes the climate topic? If so, why would it then still report the sel_temp and cur_temp to the ems-esp topic as seen in my initial screenshots? Or would these topics continue to be reported nevertheless?

proddy, 2022-02-21:
ah ok. Best thing is to try the next dev version. I'll add some additional logging so you can when and why it's removing the climate control.

MichaelDvP, 2022-02-22:
The actual dev and master contains no code to remove the climate config, the removal is part of the PR#369.

It looks to me as the mqtt broker is restarted and the retained messages are not stored (persistence set to false). The other HA topics are recreated on the reconnect, but climate as special thermostat value does not check for mqtt-connect.

@stefboerrigter Please check if and why the broker is restarted and set persistence to true.

sq4bja, 2022-02-22:
I have another error in Domoticz with Climate device. 
`EMS-ESP: Climate device no idea how to interpretate state values (thermostat_hc1`
RC310 with GB072 and newest EMS-ESP Beta 7.

stefboerrigter, 2022-02-22:
@MichaelDvP; i've enabled retain in mqtt settings. 
Indeed my server (on which my broker runs) resets nightly due to other (unresolved) issue. I'll check tomorrow if it has resolved anything.

stefboerrigter, 2022-02-22:
Should I add latest fix you merged in #369 to test together?

stefboerrigter, 2022-02-23:
@MichaelDvP and @proddy 
The test with using retained for MQTT and having persistence true in my broker seems to have solved the issue. 
Thanks for your cooperation!

closed 2022-02-23 by stefboerrigter

proddy, 2022-02-23:
@stefboerrigter https://emsesp.github.io/docs/#/Troubleshooting?id=messages-are-not-always-coming-in-via-mqtt

stefboerrigter, 2022-02-23:
Thanks @proddy, after reading this its a classic case of RTFM, but also knowing what to search for.
I was too much focused on only the climate note (which is used for Auto Discovery in HA) that was missing and therefore not stumbled upon this Troubleshooting page.