emsesp/EMS-ESP32 #374: ems-esp and Buderus MM100 without boiler

0x0d, 2022-02-23:
Hi, I`ve ordered a BBQKees EMS Gateway S32 and wanted to connect it to my Buderus MM100 mixer modules.
My boiler(Vaillant) is using a ebus protocol for communication and cannot be connected to mixer modules directly, so I just wanted to control those mixer modules from my home automation system via MQTT.

Gateway is directly connected to an MM100 module via EBUS, without any other heating equipment.
Mixer module has two EMS bus ports: EMS IN and EMS OUT. I tried to connect EMS gateway to both of them without any success.
"show ems" command just gives me a: "EMS disconnected" message every time.

What i`ve tried:
 - Different combinations of Tx protocols, with no success
 - Swapping +- of EMS cable
 - Different MM100 modules - same behaviour on all of them
 - No useful messages appeared in the console log, or web-interface log which could help me to debug the problem.
 - The console "watch" command also gives no any information.

I am using a 3.3.1 ems-esp32 firmware version.

How can I debug the problem?
Can it be because of "master"(0x08 ID) absence on ESP bus?

MichaelDvP, 2022-02-23:
You need a master on the bus. And you can not use emsesp as master, because the master have a different hardware interface. Clients sending in current modulation to the master and receiving as voltage modulation from master. The master receives in current modulation and sends in voltage modulation, that ensures that all clients only communicating with the master.
The cascade module MC400 contains a master and maybe it works without a boiler.

0x0d, 2022-02-23:
Thanks for explanation. 
I am sure that this information must be specified somewhere in documentation :)

I'll try to look at MC400. But can RC310 or RC200 thermostats do the trick? They are much more cheaper. 
In documentation mentioned that they can control MM100 modules.

MichaelDvP, 2022-02-23:
I think the RCxxx are only clients and needs a master for communication. 
There is a SC300 for controlling solar modules independent, but i dont know if it works with mixer modules.
Without control unit like RC300 it's difficult to configure the MM100 correctly. Theoretical this can be done with emsesp, but you also need a bosch insider to tell you the right ems commands, we do not know them.
I think the chance that it works without boiler is very little, make sure you can send back expensive modules.

closed 2023-02-19 by proddy