emsesp/EMS-ESP32 #376: Setting "heating temperature" and "selected flow temperature" values get overwritten by the boiler after 60s

Scoty, 2022-02-24:
**Why the value of `selected flow temperature` gets changed back to the `heating temperature` value after about 60s ?**

I want to be able to set my heating temperature based on readings from my room temperature sensors and local weather info provider. I have Buderus Logamax U052-24K  without any external controlling thermostat installed and BBQKees Gateway S32 on v3.4.0b7 with HT3 set as mode, because EMS gives me TX errors and the service manual of Logamax U052-24K says that it uses UBA-H3 board, which is short of Heatronics3 I guess.

So before "coding it all" I tried using the GUI Dashboard to set the `heating temperature`. I see in the logs event that the write telegram was successfully sent (), but the `heating temperature` remains set to what is set via the hardware dial. That is my first issue :).

```
2022-02-24 13:27:22.900 D 563: [emsesp] Last Tx read successful
2022-02-24 13:27:42.332 D 564: [command] Calling command 'boiler/heatingtemp' (heating temperature) with value 50
2022-02-24 13:27:42.332 D 565: [emsesp] Write command successful
2022-02-24 13:27:42.435 D 566: [telegram] Sending write Tx [#159], telegram: 0B 08 16 01 32
2022-02-24 13:27:42.450 D 567: [emsesp] Process setting 0x16 for device 0x08
2022-02-24 13:27:42.466 D 568: [emsesp] Last Tx write successful
```

In the discussion of #359 I read that I should set the `heating temperature` to the max value  of the range that I plan to use and change the `selected flow temperature` instead. So I have tried that and it **somewhat worked** - after changing the `selected flow temperature` to a different (always lower than `heating temperature`) value it changes and a new read-only value (`set flow temperature`) gets added to Dashboard, which is the value which I set. 

If I keep the GUI open and refresh once per minute - the value of `selected flow temperature` remains the same as the value of `set flow temperature`, however if I close the webpage or do not refresh it - the value of `set flow temperature` gets changed to the value of the "heating temperature" by the boiler I guess, as I haven't sent any write commands and do not see any in the logs.

So my question is: how to control the `heating temperature` in a way that it remains set to whatever I set it via the EMS Gateway and not get overwritten.

// offtopic
The value of `burner selected max power` is wrong it should be 50% (that is what I set via the service menu - I even checked it again after installing the Gateway), but is reported as 27%. I guess that should be reported as another issue.

**Screenshots**
[image][image]

[image]

**Device information**
```
{
  "System": {
    "version": "3.4.0b7",
    "uptime": "000+09:01:30.668",
    "uptime (seconds)": 32490,
    "freemem": 163,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "SSID": "Scoty",
    "BSSID": "24:4B:FE:D4:F1:20",
    "RSSI": -58,
    "MAC": "7C:87:CE:49:B1:D0",
    "IPv4 address": "192.168.50.129/255.255.255.0",
    "IPv4 gateway": "192.168.50.1",
    "IPv4 nameserver": "192.168.50.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 12561,
    "bus reads (tx)": 109,
    "bus writes (tx)": 3,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 1,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "13.04",
      "entities": 59,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xC2 0x35 0x26 0x2A"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "13.04",
      "entities": 0
    }
  ]
}
```

**Additional context**
Full logs of the interaction:

```
2022-02-24 13:15:23.174 DEBUG 422: [emsesp] Last Tx read successful
2022-02-24 13:15:52.668 DEBUG 423: [command] Calling command 'system/info' (show system status) with value 
2022-02-24 13:16:22.556 DEBUG 424: [emsesp] Fetching values for deviceID 0x08
2022-02-24 13:16:22.556 DEBUG 425: [telegram] Tx read request to device 0x08 for type ID 0x14
2022-02-24 13:16:22.556 DEBUG 426: [telegram] Tx read request to device 0x08 for type ID 0x16
2022-02-24 13:16:22.556 DEBUG 427: [telegram] Tx read request to device 0x08 for type ID 0x33
2022-02-24 13:16:22.640 DEBUG 428: [telegram] Sending read Tx [#121], telegram: 0B 88 14 00 1B
2022-02-24 13:16:22.679 DEBUG 429: [emsesp] Last Tx read successful
2022-02-24 13:16:22.764 DEBUG 430: [telegram] Sending read Tx [#122], telegram: 0B 88 16 00 1B
2022-02-24 13:16:22.819 DEBUG 431: [emsesp] Last Tx read successful
2022-02-24 13:16:22.889 DEBUG 432: [telegram] Sending read Tx [#123], telegram: 0B 88 33 00 1B
2022-02-24 13:16:22.889 DEBUG 433: [emsesp] Fetching values for deviceID 0x09
2022-02-24 13:16:22.939 DEBUG 434: [emsesp] Last Tx read successful
2022-02-24 13:17:22.557 DEBUG 435: [emsesp] Fetching values for deviceID 0x08
2022-02-24 13:17:22.557 DEBUG 436: [telegram] Tx read request to device 0x08 for type ID 0x14
2022-02-24 13:17:22.557 DEBUG 437: [telegram] Tx read request to device 0x08 for type ID 0x16
2022-02-24 13:17:22.557 DEBUG 438: [telegram] Tx read request to device 0x08 for type ID 0x33
2022-02-24 13:17:22.559 DEBUG 439: [telegram] Sending read Tx [#124], telegram: 0B 88 14 00 1B
2022-02-24 13:17:22.599 DEBUG 440: [emsesp] Last Tx read successful
2022-02-24 13:17:22.716 DEBUG 441: [telegram] Sending read Tx [#125], telegram: 0B 88 16 00 1B
2022-02-24 13:17:22.765 DEBUG 442: [emsesp] Last Tx read successful
2022-02-24 13:17:22.840 DEBUG 443: [telegram] Sending read Tx [#126], telegram: 0B 88 33 00 1B
2022-02-24 13:17:22.840 DEBUG 444: [emsesp] Fetching values for deviceID 0x09
2022-02-24 13:17:22.890 DEBUG 445: [emsesp] Last Tx read successful
2022-02-24 13:18:03.221 DEBUG 446: [command] Calling command 'boiler/selflowtemp' (selected flow temperature) with value 50
2022-02-24 13:18:03.221 DEBUG 447: [emsesp] Write command successful
2022-02-24 13:18:03.311 DEBUG 448: [telegram] Sending write Tx [#127], telegram: 0B 08 1A 00 32
2022-02-24 13:18:03.326 DEBUG 449: [emsesp] Process setting 0x1A for device 0x08
2022-02-24 13:18:03.337 DEBUG 450: [emsesp] Last Tx write successful
2022-02-24 13:18:03.337 DEBUG 451: [telegram] Sending post validate read, type ID 0x18 to dest 0x08
2022-02-24 13:18:05.364 DEBUG 452: [telegram] Sending read Tx [#128], telegram: 0B 88 18 00 1B
2022-02-24 13:18:05.428 DEBUG 453: [emsesp] Last Tx read successful
2022-02-24 13:18:22.558 DEBUG 454: [emsesp] Fetching values for deviceID 0x08
2022-02-24 13:18:22.558 DEBUG 455: [telegram] Tx read request to device 0x08 for type ID 0x14
2022-02-24 13:18:22.558 DEBUG 456: [telegram] Tx read request to device 0x08 for type ID 0x16
2022-02-24 13:18:22.558 DEBUG 457: [telegram] Tx read request to device 0x08 for type ID 0x33
2022-02-24 13:18:22.604 DEBUG 458: [telegram] Sending read Tx [#129], telegram: 0B 88 14 00 1B
2022-02-24 13:18:22.643 DEBUG 459: [emsesp] Last Tx read successful
2022-02-24 13:18:22.728 DEBUG 460: [telegram] Sending read Tx [#130], telegram: 0B 88 16 00 1B
2022-02-24 13:18:22.784 DEBUG 461: [emsesp] Last Tx read successful
2022-02-24 13:18:22.822 DEBUG 462: [telegram] Sending read Tx [#131], telegram: 0B 88 33 00 1B
2022-02-24 13:18:22.822 DEBUG 463: [emsesp] Fetching values for deviceID 0x09
2022-02-24 13:18:22.872 DEBUG 464: [emsesp] Last Tx read successful

```

MichaelDvP, 2022-02-24:
The left dial on your boiler sets the heatingTemperature, it's not an electronical setting, so the value is always reseted to the dial-position if you try to change via emsesp.
If you don't use a thermostat, this is also the flowtemp published as selflowtemp (telegram 0x18, offset 0). A thermostat can set the flowtemp to a lower value and repeats this every minute. The setting is made to the setflowtemp (telegram 1A, offset 0) value and reported only if there is a write to it. If there is no write to the setflowtemp the boiler goes back to the heatingtemp and publish it as selflowtemp. Try in terminal a `read 08 1A` to see if setflowtemp also is changed back but in not published (so emsesp shows the old value).

To change the flowtemp you have to send a command frequently (every minute), like a thermostat do.

For the burner power check (read) the telegrams 0x18, offset3 for max selected power and telegram 0x16 offset 2 for max  power.

Scoty, 2022-02-24:
Thanks for the detailed answer - I confirm that sending 0 as value to the `setflowtemp` every 30 seconds prevented the heating from being activated for the last 5 minutes. I stopped the loop a minute ago and as you said since the boiler was no longer receiving `setflowtemp` it switched back the value of `heating temperature` and the burner/heating started, due to the lower value of the `current flow temperature`.

Btw by "Try in terminal a `read 08 1A` ... " do you mean that there is some kind of terminal in the GUI, where I can send telegrams to the Boiler ?

MichaelDvP, 2022-02-24:
> o you mean that there is some kind of terminal in the GUI,

No, there is a telnet server [(see in doc)](https://emsesp.github.io/docs/#/Console). In v3.4 you have to enable it in settings.

MichaelDvP, 2022-02-24:
> The read 08 1A shows no data -

Yes that's the same with my boiler (GB125/MC10), i get <empty> back from a read. But i can see the messages from thermostat:
```
ems-esp:$ read 8 1A
001+06:25:54.666 I 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBASetPoints(0x1A), data: <empty>
eems-esp:$ watch 1A
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x1A
001+06:22:35.594 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2E 64 00 00
001+06:23:36.344 N 2: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2E 64 00 00
```

What about the burner power? Have you checked `read 8 18` and `read 8 16` for the values. 
Most burners have more power on dhw as on heating, because water loading have higher flow. Bosch sometimes shows this with burner power > 100% for dhw, sometimes als 100% for dhw and lower power for heating. Maybe this is the scaling that you can set 50% of max heating, which is 27% of absolut max burner (dhw) power. But this is a guess, i don't know and always get confused if bosch changes it's own rules on every new boiler model.

Scoty, 2022-12-12:
> @Scoty I know it's almost a year ago, so maybe you've already changed that setting but I see you've got the hot tap water temperature at 45 degrees. To prevent getting Legionella growth I recommend to set the temperature at 60 degrees.

Hey @royka1, thanks for the info, but I think that the boiler runs the thermal disinfection program (~ 70 degrees for 5-10 minutes) at least once per week. Also, it doesn't have a body to store the hot water - it is just flow heating, so I think it is harder to get Legionella there.

royka1, 2022-12-12:
> 

Hey @Scoty I understand your point of view. It's not only in the boiler itself, it could also grow in the pipes and appendages. At 45 degrees it even still grows, at 50 degrees it takes 2hrs to kill it (I don't shower that long), you will also loose some heat on it's way to the shower. I didn't read the whole page yet, will do later. It's actually a part of my job at a water company. 
A laboratory takes "randomly" samples for us at different addresses, last week I had to visit an address where Legionella was detected. The moment they detected it, they took samples at a few neighbors but didn't found anything there, it turned out that they were the only one that had lowered the temperature of the dhw. I've to admit that there was only the non-pneumophila (less dangerous), but it's a indicator for us. Ones it's in your system, it is hard to kill it, then you should replace a lot of appendages and if thermally disinfection doesn't work eventually chemically disinfection.
It could come with the water from the water company and normally flushed out or get killed when the installation is right, but when the temperature is too low (or cold water too high) there could grow big colonies and that could be dangerous.
So I'm a bit surprised that this company advices to lower the temperature to 45 degrees.

closed 2023-02-19 by proddy