emsesp/EMS-ESP32 #377: Missing ww entities with RC35/BK13 on 3.4.0b7

g6094199, 2022-02-25:
every entity binary_sensor.boiler_ww* is missing on HA and the EMS GUI

which is everything regarding warm water beside boiler_ww_3_way_valve_on_ww

on the latest 3.3.x stable WW regarding entities are there.

Also the BC10/RFM20 which is shown in 3.3 is gone in 3.4

MichaelDvP, 2022-02-25:
When upgrading from v3.3 to v3.4 do the following steps:
1.  reload page in browser or close and reopen browser to update all scripts.
2.  navigate to mqtt settings and disable mqtt-discovery, Click `SAVE`, 
3.  Wait 5 minutes (MINUTES not secondes),
4. Enable mqtt-discovery and click `SAVE`

The controller is not shown on webpage, it never has any value. 
ww-entities are renamed to `dhw`, the above procedure ensures that the ww's are removed and dhw's are created.

proddy, 2022-02-25:
this is good, I think we need to add these steps to the releasenotes and doc so everyone is clear of the breaking changes and steps to go from 3.3 and 3.4

and we should release 3.4 soon