emsesp/EMS-ESP32 #383: Unrecognized EMS device (Device ID:0x17 Product ID:0 Version 00.00) - Thermostat Buderus, RC 10

chriscaloudis, 2022-03-07:
Hello all, I hope I posted in the right section...
I managed to set up my S32 gateway (via connection to the thermostat wires).

The problem that I am facing is that my thermostat (RC10) is not recognized, so I get no current or setpoint temperature readings.
The reading next to THERMOSTAT is Device ID:0x17 Product ID:0 Version 00.00 .

This is my dashboard
[image]

The connection looks fine as per the following picture
[image]

I am running firmware version 3.3.1
[image]

These are the ems-esp settings
[image]

Similar(?) issue I found on Github
https://github.com/emsesp/EMS-ESP32/issues/160

Please let me know if any additional information is required!
Any help will be greatly appreciated!

MichaelDvP, 2022-03-07:
The RC10 as generic thermostat is a fallback for Tado thermostats not sending any version info, introduced in #174, but this is not compatible with HT3 boilers. 
It's strange that your thermostat does not reply to version request. The RC10 in #160 sends a product-id, your's not.
Please try to identify your thermostat, send a photo of front, and/or type-plate, check in thermostat menu if you find any version info. 
Please also make a a log with loglevel all of more than a minute, download it and attach here.
I'm not sure about all the changes from v3.3.1 to current 3.4-dev, I think it's worth trying first to update and make the checks/logs with the dev. (reload web-page in browser after update).

chriscaloudis, 2022-03-07:
I updated firmware to v3.4.0b7. Behavior remains the same.

[image]

Therefore, as per your kind request, this is the front and back of the thermostat.

[image]

[image]

I also uploaded the log with loglevel all (with Tx mode set to EMS).
[log.txt](https://github.com/emsesp/EMS-ESP32/files/8197037/log.txt)

I'm not sure about one thing, please let me know what you think.
I should stick to Tx mode set to EMS, is that right? 
Picture below is EMS Bus and Activity Status with Tx mode set to EMS

[image]

MichaelDvP, 2022-03-07:
The settings are looking ok to me. I have not known that a Buderus thermostats works on HT3 systems, always learning...
The thermostat sends 0xAE telegrams, seems to be the same as in #160, but we do not get a version info to identify it.
Can you go to Help and download system info (and post it) to make sure it is ht3 protocoll. 
The tx-errors are zero! If there is no reply to version (like tado) this should give a single tx-error, so i expect that there is a real answer to version request, but the thermostat sends product-id zero. Can you please check with telnet the reply to commands : `read 17 2` and `read 8 7` 

@proddy : For now i see two possible solutions.
- If there is a real thermostat with prod-id 0, we should change the generic catch-all-id in `device-library.h` to 255 ([here](https://github.com/emsesp/EMS-ESP32/blob/e247b9e5f9cfb5f21f64d0495fdb443bae66e1aa/src/device_library.h#L145) and [here](https://github.com/emsesp/EMS-ESP32/blob/e247b9e5f9cfb5f21f64d0495fdb443bae66e1aa/src/emsesp.cpp#L842)) and hope there is no device with this product ID. 
- If this a mix ht3/buderus we can add [here](https://github.com/emsesp/EMS-ESP32/blob/e247b9e5f9cfb5f21f64d0495fdb443bae66e1aa/src/emsesp.cpp#L1100) a `if (rxservice_.is_ht3()) { flags = DeviceFlags::EMS_DEVICE_FLAG_RC20_N;`... 

Any other ideas?

proddy, 2022-03-07:
both do the trick, but first like to see what `read 17 2` comes back with

MichaelDvP, 2022-03-07:
In v3.4 there is a checkbox in settings to enable telnet. 
Also you have to use connection type telnet on port 23 (not SSH).
If mDNS (ems-esp.local) fails, try also the ip address, there is another issue in discussion, that mDNS fails after network connection is reconnected.

chriscaloudis, 2022-03-07:
Yeap, had to enable telnet console in the settings.

So, this is the output to the commands 'read 17 2' and 'read 8 7'

[image]

MichaelDvP, 2022-03-07:
So we have the tx-fail on requesting the version, the thermosttat is not answering, but the boiler has recognized the thermostat. My first solution will not work.
Can you now send the the `help/system info` to check the bus protocol.

chriscaloudis, 2022-03-07:
I edited my last post to include the system info...I am re-posting it here.
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/8199907/emsesp_info.txt)

chriscaloudis, 2022-03-07:
ho ho ho...some sort of miracle happened..
I switched to Tx mode HT3 and restarted the EMS-ESP...this is what popped-up!

[image]

MichaelDvP, 2022-03-07:
Oh, wow, so now the thermostat have a version info. That's good, the info with connection type Buderus killed my second solution.
Looks like the issue is solved, but the zero tx-fails before are strange. Also the ht3-boiler with buderus thermostat is something that makesme wonder. Please post the system-info and system-settings again with the right tx-mode.

chriscaloudis, 2022-03-07:
As per your request, I am uploading system-info and system-settings with Tx mode set to HT3.
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/8199995/emsesp_info.txt)
[emsesp_settings.txt](https://github.com/emsesp/EMS-ESP32/files/8199996/emsesp_settings.txt)

I would also like to report that I am getting EMS Writes(Tx) for the first time.
[image]

chriscaloudis, 2022-03-08:
So, given the fact that the module works fine, I guess it is time to close the issue.
I'd like to thank everyone that took interest in the case and contributed in any way!

closed 2022-03-08 by chriscaloudis