emsesp/EMS-ESP32 #386: NTP synchronization to thermostat panel

sq4bja, 2022-03-08:
Is it possible for ems-esp to automatically update the time in a thermostat, eg RC310? Manual update via SSH or JSON works fine, but can it be turned on automatically?

MichaelDvP, 2022-03-08:
That's `/api/thermostat/datetime` with payload `{"value":"ntp"}`. Working for RC35 and RC300.

proddy, 2022-03-08:
bingo. then @sq4bja does that work? If you're using HA you can create a trigger or a scheduled job.

MichaelDvP, 2022-03-09:
I've tested a automatic correction if time is more than 1 minute apart. In `process_RCTime` added 
```
    // check clock
    time_t now = time(nullptr);
    tm *   tm_ = localtime(&now);
    if (tm_->tm_year < 110) { // no NTP time
        return;
    }
    if ((tm_->tm_year - 100) != (telegram->message_data[0] & 0x7F) || tm_->tm_mon + 1 != telegram->message_data[1] || tm_->tm_mday != telegram->message_data[3]
        || tm_->tm_hour != telegram->message_data[2] || (tm_->tm_min + 1) < telegram->message_data[4] || tm_->tm_min > (telegram->message_data[4] + 1)) {
        set_datetime("ntp", -1); // set from NTP
        LOG_INFO(F("time correction from ntp"));
    }
```

I manually set time wrong and get the time correction message in log. Heating is a slow process, a minute does not really matter and my clock is only a few seconds a year apart. I'm now waiting for the dst-switching and see what happens.

sq4bja, 2022-03-09:
Yes, that's exactly what I meant. Time is not so important to heating, but the thermostat is often used as a wall clock. In my case of the RC310, the time difference is one minute per 2 months, so such a code modification would help a lot.
I know that I can trigger the time correction as an external event, eg dzVents in Domoticz, but I think that it should be done independently of external software.
The idea that the update should be 1 minute difference between the thermostat time and the NTP time is very good, however I think it would be more correct to update the time every 24 hours of uptime.
Regardless of your decision, any solution will be of great help.

MichaelDvP, 2022-03-11:
I think for scheduled update of the clock it is better to do this from external homeautomation (homeassistant/domotics/ioBroker/...). My first approach for automatic correction need some improvments. The clock publishing on the bus can be delayed 1 or 2 seconds, giving me a lot of corrections  in clock publishing is near full minute. Also we have a lot of thermostats with not writable clock, this will flood the log with correction messages.
Next approach with correction on 15 sec difference:
```
    // check clock
    time_t now = time(nullptr);
    tm *   tm_ = localtime(&now);
    if (tm_->tm_year < 110 || !has_command(&dateTime_)) { // no NTP time or no command
        return;
    }
    tm_->tm_year      = telegram->message_data[0] + 100;
    tm_->tm_mon       = telegram->message_data[1] - 1;
    tm_->tm_mday      = telegram->message_data[3];
    tm_->tm_hour      = telegram->message_data[2];
    tm_->tm_min       = telegram->message_data[4];
    tm_->tm_sec       = telegram->message_data[5];
    tm_->tm_isdst     = telegram->message_data[7] & 0x01;
    double difference = difftime(now, mktime(tm_));
    if (difference > 15 || difference < -15) {
        set_datetime("ntp", -1); // set from NTP
        LOG_INFO(F("time correction from ntp"));
    }
```
(with new function `EMSdevice::has_command(const void * p_value)`)

closed 2022-03-30 by sq4bja