emsesp/EMS-ESP32 #390: High number of Reads TX fails

kjw1985, 2022-03-10:
**Bug description**
After running 3.3.0 for quite some time without any issues I upgraded to 3.3.1, which lost connection after a day. So today I tried 3.4.0b8 which shows a high number of read TX fails compared to earlier. 

See output from Status page

  | SUCCESS | FAIL | QUALITY
-- | -- | -- | --
EMS Telegrams Received (Rx) | 1,570 | 1 | 100%
EMS Reads (Tx) | 166 | 123 | 58%
EMS Writes (Tx) |   |   |  
Temperature Sensor Reads |   |   |  
Analog Sensor Reads |   |   |  
MQTT Publishes | 521 | 0 | 100%
API Calls |   |   |  

As far as I've experienced at the moment everything else seems to be working fine.

**Steps to reproduce**
Upgraded from 3.3.1 to 3.4.0b8

**Expected behavior**
Less Reads TX errors.

**Screenshots**
See the difference between 3.3.1 and the 3.4.0b8 version installed today. Normally the number of errors seemed to be around 12 or something.
[image]

**Device information**
```json
{

  "System": {

    "version": "3.4.0b8",

    "uptime": "000+00:16:01.761",

    "uptime (seconds)": 961,

    "freemem": 152,

    "reset reason": "Software reset CPU / Software reset CPU",

    "temperature sensors": 0

  },

  "Network": {

    "connection": "WiFi",

    "hostname": "ems-esp",

    "SSID": "data24",

    "BSSID": "60:E3:27:21:46:B2",

    "RSSI": -60,

    "MAC": "94:B9:7E:FA:5B:68",

    "IPv4 address": "192.168.1.43/255.255.255.0",

    "IPv4 gateway": "192.168.1.1",

    "IPv4 nameserver": "192.168.1.17"

  },

  "Status": {

    "bus status": "connected, tx issues - try a different tx-mode",

    "bus protocol": "Buderus",

    "bus telegrams received (rx)": 1608,

    "bus reads (tx)": 170,

    "bus writes (tx)": 0,

    "bus incomplete telegrams": 1,

    "bus reads failed": 126,

    "bus writes failed": 0,

    "bus rx line quality": 100,

    "bus tx line quality": 58,

    "MQTT status": "connected",

    "MQTT publishes": 537,

    "MQTT publish fails": 0,

    "temperature sensors": 0,

    "temperature sensor reads": 0,

    "temperature sensor fails": 0,

    "analog sensors": 0,

    "API calls": 0,

    "API fails": 0

  },

  "Devices": [

    {

      "type": "Boiler",

      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",

      "device id": "0x08",

      "product id": 123,

      "version": "07.00",

      "entities": 68,

      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x34 0x2A",

      "handlers fetched": "0x14 0x16 0x33 0x26",

      "handlers pending": "0xC2 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"

    },

    {

      "type": "Thermostat",

      "name": "RC100/Moduline 1000/1010",

      "device id": "0x18",

      "product id": 165,

      "version": "40.04",

      "entities": 13,

      "handlers received": "0xA2",

      "handlers fetched": "0x02A5 0x02B9 0x029B",

      "handlers pending": "0xA3 0x06 0x12 0x02AF 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x02A7 0x02BB 0x02B1 0x029D 0x02A8 0x02BC 0x02B2 0x029E 0x02A9 0x02BD 0x02B3 0x029F 0x02AA 0x02BE 0x02B4 0x02A0 0x02AB 0x02BF 0x02B5 0x02A1 0x"

    },

    {

      "type": "Controller",

      "name": "BC10",

      "device id": "0x09",

      "product id": 190,

      "version": "01.04",

      "entities": 0

    }

  ]

}
```
**Extra information**
Part from the system log, but those are all like this (as far as I can see)
```
2022-03-10 20:53:14.526 E 163: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 99 FF 00 19 03 72
2022-03-10 20:53:16.564 E 164: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9A FF 00 19 03 73
2022-03-10 20:53:23.331 E 165: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9B FF 00 19 03 74
2022-03-10 20:53:29.363 E 166: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9C FF 00 19 03 75
2022-03-10 20:53:31.293 E 167: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9D FF 00 19 03 76
2022-03-10 20:53:33.426 E 168: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9E FF 00 19 03 77
2022-03-10 20:53:35.731 E 169: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9F FF 00 19 03 78
2022-03-10 20:54:14.992 E 170: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 99 FF 00 19 03 72
2022-03-10 20:54:16.992 E 171: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9A FF 00 19 03 73
2022-03-10 20:54:19.092 E 172: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9B FF 00 19 03 74
2022-03-10 20:54:21.125 E 173: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9C FF 00 19 03 75
2022-03-10 20:54:22.930 E 174: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9D FF 00 19 03 76
2022-03-10 20:54:25.192 E 175: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9E FF 00 19 03 77
2022-03-10 20:54:29.058 E 176: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 9F FF 00 19 03 78
```

MichaelDvP, 2022-03-11:
Thanks for the good documentation, I've fixed in a PR. Check with new dev after the pr is merged.

kjw1985, 2022-03-11:
Thanks for the fast fix. Will wait for the new dev and give it a try and report back.

closed 2022-03-11 by kjw1985

MichaelDvP, 2022-03-11:
The 14 errors are due to the logic with master-thermostat, ems-esp searches for thermostats for hc2.. at id 19, 1A, .. for an answer. 
I think these errors will not come up with the testsoftware which supports multiple thermostats.
Check this: [https://github.com/MichaelDvP/EMS-ESP32/releases/tag/dev_no_master](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/dev_no_master)

kjw1985, 2022-03-11:
Flashed the linked version after 6 minutes of uptime this is the status:

  | SUCCESS | FAIL | QUALITY
-- | -- | -- | --
EMS Telegrams Received (Rx) | 376 | 1 | 100%
EMS Reads (Tx) | 99 | 0 | 100%
EMS Writes (Tx) |   |   |  
Temperature Sensor Reads |   |   |  
Analog Sensor Reads |   |   |  
MQTT Publishes | 356 | 0 | 100%
API Calls | 1 | 0 | 100%

So it removes the previous shown "errors".

Before flashing the `dev_no_master` version I saw that the uptime was not what I expect so I took a look into my data.

On the 16 of December I flashed 3.3.0 on the 5th of March I flashed 3.3.1 and on the 10th of March I flashed the 3.4.0b8 version from yesterday. 

[image]

And here the uptime date since the 5th of March, as you can see I've had some trouble with dropping connections/crashing EMS-ESP as there is no data (I could only restart the device by unplugging and replugging it):
[image]

Might be that this something for another bug report?

kjw1985, 2022-03-11:
I will check it out. Thanks for the help

commit 52a9e500 (2022-03-11): fix #390, fetching summer2_ids
commit f539a236 (2022-03-11): Merge pull request #391 from MichaelDvP/dev

fix #390, fetching summer2_ids