emsesp/EMS-ESP32 #395: RC30_N sets the wrong temperature with home assistant

jerome992, 2022-03-12:
The RC30_N does not have an method the set the actual temperature. It is only possible to the the night or the day time temperature.

With HA the following is happening:

1.) the RC30_N is in night mode, with the following temperature (night:17, day:19)
2.) HA sends the following command: thermostat_hc1 day followed by a thermostat_hc1 20°C
3.) the EMS-ESP sends the head mode to the RC30_N, but the change is not applied immediately in the RC30_N.
4.) the EMS-ESP sends the temperature 20°C to night temperature of the the RC30_N, because it has still not received the ack from the RC30_N, that the mode has switched: (night:20, day:19)
5.) the RC30_N switch the mode to day. (night:20, day:19)

finally the RC30_N is in day mode with the wrong temperature of 19, because the temperature has been applied to the wrong mode.

A possible solution is provided as attachment: The new mode is saved in a temporary variable until the RC30_N has acknowledged the mode change.

Sincerely,
Jerome

[patch_rc30_n.txt](https://github.com/emsesp/EMS-ESP32/files/8237537/patch_rc30_n.txt)

proddy, 2022-03-12:
nice @jerome992 that you identified an issue and have a fix for it! Could you do a Pull Request for this? It's easier to test and merge. thanks!

proddy, 2022-03-12:
@MichaelDvP I think this is the last thing to go into v3.4 before we make it the next stable release

MichaelDvP, 2022-03-12:
Hmm, i think it's not RC30 specific. 
Sending multiple commands in short time will always be critical. Better wait for each command for readback /validate.
Write commands to the bus should be executed fast, so they are put to top of queue. But sending multiple commands will execute the latest first if waiting for the bus poll. 
Nevertheless the solution will work, The wantedmode is set before the seltemp command is executed.

I don't know what HA do, but isn't it better to send `daytemp` command? Or is the climate sending both commands?
The `thermostat_hc1` subscription is not in v3.4, climate goes direct to seltemp/mode-command, but with the same result.

jerome992, 2022-03-13:
It is specific to every thermostat that does not allow to set actual temperature, but that only allows to set a night or day temperature.

I have made the following tests with version 3.3.1:
1.) I changed the mode in the climate GUI and HA has emited ems-esp/thermostat_hc1=off
2.) I changed the temperature in the climate GUI and HA has emited ems-esp/thermostat_hc1=16.5
3.) I called the service climate.set_temperature and the HA has emited ems-esp/thermostat_hc1=heat0 followed immediately by a ems-esp/thermostat_hc1=15.0:
```
service: climate.set_temperature
data:
  temperature: 15
  hvac_mode: heat
target:
  entity_id: climate.thermostat_hc1
```
4.) I have applied a scene in HA and the HA has also emited ems-esp/thermostat_hc1=heat0 followed immediately by a ems-esp/thermostat_hc1=15.0.

I dont know how to wait for a feedback in HA, except to use a script to in HA like:
sequence:
```
  - service: climate.set_hvac_mode
    data:
      hvac_mode: 'heat'
    target:
      entity_id: climate.thermostat_hc1
  - delay:
      seconds: 30
  - service: climate.set_temperature
    data:
      temperature: 20
    target:
      entity_id: climate.thermostat_hc1
```

MichaelDvP, 2022-03-13:
Thanks for the detailed explanation. So this is only this service in step 3 that sends both commands, not the climate gui. Is this a HA standard function?
I think all thermostats are affected if they are in a different mode when these sequence is received. 

Sorry, i can't help with HA specific functions, but proddy uses HA.

jerome992, 2022-03-13:
The GUI only sends the parameter that the user has changed. If the user modifies both parameters (mode, temp) very shortly one after the other, both commands are send one after the other. If the user is too fast, the modification of the temperature goes wrong.

Probably all thermostats are affected because:
1.) HA uses set_temperature(f, HeatingCircuit::Mode::AUTO, hc_num);
2.) set_temperature uses get_mode
3.) get_mode gets the value from an advertisement on the EMS bus.

MichaelDvP, 2022-03-14:
Yes, i'm pretty sure it affects all thermostats. Then we can set [here in set_mode_n()](https://github.com/emsesp/EMS-ESP32/blob/a695f85359a3aa617a4d4e046e62380c86eb247e/src/devices/thermostat.cpp#L2003) 
```
    // set hc->mode temporary until validate is received
    if (model() == EMSdevice::EMS_DEVICE_FLAG_RC10) {
        hc->mode = set_mode_value >> 1;
    } else {
        hc->mode = set_mode_value;
    }
```
If the set_temperature command is coming fast `get_mode()` returns the temporary new mode, otherwise in command-validation the `hc->mode` is read from bus and all is in sync again. 
Only if the new mode is AUTO and it is a thermostats that checks `get_mode_type()` to set the temperature it may be wrong, but we can not know what modetype AUTO will set, we don't know the switch-program. This can only be solved it we delay the seltemp command until tx-queue is empty and validate-id is cleared.

proddy, 2022-03-20:
what shall we do here? put in a delay seems a bit overkill.

@jerome992 you probably know this but you can also call EMS-ESP commands from HA using the [Rest service](https://emsesp.github.io/docs/#/Home-Assistant?id=example-changing-a-value-via-the-api), for example to set the thermostat setpoint temperature. It will keep the current mode. It may be easier than getting HA's climate component to do it.

MichaelDvP, 2022-03-20:
The suggsested `set_mode_n()` modification do no harm, the post-validation sets thermostat mode a second later and this will help for the RC30 and some other thermostats. 

A delay is difficult, when switching mode from night to auto, mode is set instantly, but it takes up to minute to get the right modetype (i've logged this in RC35). We can not delay in set_temperature-function, this is in main thread, we'll  need a command queue and have to analyse the commands before executing and only delay the hcx/seltemp after hcx/mode. Since the delay has to be very long there will be a lot of complains that set_temp does not work. I don't think a delay is usefull.

MichaelDvP, 2022-03-30:
The RC30 should work now, @jerome992 can you confirm (and close issue if ok)? 
For thermostats with modetype settings i have not idea how to manage, the thermostat is too slow to get the values in acceptable time. If this is a problem we can open a new issue.

MichaelDvP, 2022-03-31:
Oh, sorry, i havn't realized that this is RC30_N, not RC30. 

I've checked the different thermostat types and working is now RC10, RC20, RC_20_N, RC25, RC30, RC35, RC100, RC300,
Not working, because Modetype is used: RC30_N and Junkers.
RC30_N is mostly identical with RC35, but [this doku](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#hk1betriebsart) says the parameter at start 42 (offset 37) is RC35 only, which is used for setting in auto-mode. @jerome992 can you check the telegram 3D by logging (published every minute) or in terminal `read 10 3D`. Also send a log (all) of a sequence where you change the temperature in auto-mode on the thermostat by turning the wheel. Maybe there is another field to set the temperature temporary, than we dont need the modetype.

The Junkers is discussed in #400, also the field for setting the temporary auto temperature is unknown.

proddy, 2022-04-09:
@MichaelDvP close this and re-open if it's still an issue?

MichaelDvP, 2022-04-09:
It is still an issue if mode is switching to auto and seltemp is set within a few seconds. Without knowing how to set a temporary-auto temperature is not possible to solve. For the RC30_N i would like to see `read 10 3D`, but did not get it. For switching mode to other states is not an issue anymore.
You can close it.

closed 2022-04-09 by proddy