emsesp/EMS-ESP32 #396: Issues v3.4.0b8

marc1de, 2022-03-12:
- "**boiler_dhw_3way_valve_active**" no longer available
- "**boiler_dhw_activated**" no longer available 
 _(so I'm unable to turn on/off hot water. "**dhw turn on/off**" is no alternative cause this entity contains no values)_
- "**thermostat_hc1_manual_temperature**" no longer available
- "**mdns**-scripts" no longer working"

MichaelDvP, 2022-03-12:
Check Settings->Customization.

marc1de, 2022-03-12:
@MichaelDvP 
ok, you were right: This entities could be enabled in "Settings->Customisation" again.
Do you have an explanation, why they were suddenly disabled after the upgrade?

BTW: "mdns" (ems-esp.local) still does not work. Even "**restart** or **Power on/off**" of ems-ems does not enable this function (in comparison to v3.4.0b7 where this always helped)

MichaelDvP, 2022-03-12:
The entities are numbered for customization exclusion. In b8 we added the burner stage 2 worktime, so all higher numbers are increased. and exclusion is shifted. Proddy plans a change to exclude web/mqtt/api idependent, so there will be a change in v3.4.1. For now it is as it is, and you have to check your customization after every update.

For me mdns is working, also after network disconnect, which brakes mdns before.  What exactly does not work?

proddy, 2022-03-12:
that's correct, the entity id's are sequenced and when we add a new entity it will alter the customizations which is unfortunate unless we add it to the end of the list.

marc1de, 2022-03-13:
@MichaelDvP @proddy 
Thanks for the detailed explanation. That clears my confusion and I'm prepard for "future updates" now.   :-)

**Regarding the mdns:**
As described in #355 I created a testscript in HA to set gpio's
[image]
This script usualy worked after a restart or PowerOn of ems-esp but stopped working after a while (usualy that was the next day). In v3.4.0b7 it was working again after a restart of ems-esp but this does not help in v3.4.0b8. In fact: this script **never** worked in v3.4.0b8

The same script with IP-Adress instead of mdns always worked.
_[so this is no big problem to me - I just wanted to let you know since I think this might be interesting for you)_

MichaelDvP, 2022-03-13:
Please check the log if mDNS is started, there should be a message like:
`2022-03-13 09:03:14.205 I 9: [emsesp] mDNS responder service started`
also check the output if you type on your computer: 
`ping ems-esp.local`
`nslookup ems-esp.local`
`nslookup ems-esp`

Could you test with [this build](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest), i've made a little change that mDNS should behave on coldstart like b7, but restarts mDNS after network disconnects as b8.

marc1de, 2022-03-13:
Ok, I uploaded your new build:

Log said, that mdns responder service was started:
[image]
But still my HA-script did'nt work:

So I opened a cmd-screen on my Windows10-PC. But now the things came up very weird to me:
[image]

As you can see, ping worked but nslookup showed no results. 

_Unfortunaly I cannot check that on my raspberry because my HA is running in a preconfigured container with a preloaded image. I have only access to the HA-container but not to the kernel-shell._ 

After this checks I decided to go back to version 3.4.0b7 to do the same tests but to my big surprise the HA-scripts didn't work there either (Tried several combinations of Restarts/PowerON's on HA as well on ems-esp. I haven't changed my scripts (I have one script with "ems-esp.local" and an identical script with IP-Address)  and I'm absolutely shure, that this script worked all the days before, when I restarted ems-esp. Next Step is that I'm going to restart my Fritzbox and see if that makes a difference

MichaelDvP, 2022-03-13:
Looks ok to me. Ping resolves the address, but nslookup not. So the ping is resolved by mDNS and not by DNS. All as it should be.
BTW: you're using a fritzbox as nameserver, this normaly resolves local adresses by hostname. `nslookup ems-esp` will show an entry as `ems-esp.fritz.box`, that you can also use.

marc1de, 2022-03-13:
Ok, here is the update after the restart of my Fritzbox

Again mdns-responder-service was started:
[image]

but now my ping to ems-esp.local didn't work no more:
[image]

coldstart or restart of ems-esp didn't help. (version was your special build of v3.4.0b8 again)
[same result ten minutes later: ping to ems-esp.local failed)

BTW: changing the script from "ems-esp.local" to "ems-esp.fritz.box" worked but still the situation is strange. But now I have a vague suspicion: my fritzbox is serving as local DNS-Server but this fritzbox uses another raspberry with pihole upstream dns-server. Usually I thought that this is only related to external dns-queries but now I'm not shure anymore if this may have an unexpected sideeffect.

marc1de, 2022-03-17:
@MichaelDvP 
well, this mdns-thing is very strange: I was wondering why I was sometimes able to ping to `ems-ems.local` and sometimes not.

In the following screenshot from my Windows-PC you can see this strange behavior:
I opened this cmd-window some days before and as you can see, the ping did not work yesterday... but in this same cmd-window the ping works today (no restarts or changes on `ems-esp` were made). 
[image]

As far as I can see, ems-esp publishes the mdns-info correctly after a restart but windows seems unable to use this information (no ping after restarts). Dont' know, what caused it to work correctly now.
Maybe it is related to my wifi connection because I found some hints in the net, that reported mdns-problems with wifi-connections. Unfortunaly I have no chance to connect my PC via cable right now.... 
and btw. that would not explain, why my HA-Raspberry is also affected because the RPI is connected via Ethernet

harvey637, 2022-03-26:
Hi,
there seems to be some limitation on the items to exclude, maybe buffer length or something?
As I have no WarmWater (DHW), nor pool, only heating,  there is a LOT to exclude :-)
I get error 400 then ...
Thx
Harvey

proddy, 2022-03-26:
I played a little with the mDNS and couldn't reproduce what you're seeing, from Windows and Linux, although my setup uses a Unifi network. I suspect the dns code on the espressif core isn't rock-solid and there are many new improvements coming in 4.2 so suggest we wait. What you could do is close this issue and create a new one specifically for the DNS issue so we can track in future versions.

proddy, 2022-03-27:
suggest we close this and create a separate issue for mDNS and ask others if they experience similar behavior

closed 2022-03-27 by proddy