emsesp/EMS-ESP32 #398: Temperature threshold for outdoor reduction (RC35)

brezel1965, 2022-03-13:
Can you please make an entry for the threshold for canceling the reduce at the outdoor temperature. I call it hc1_noreduceoutdoor. 

thx

[image]

https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#hk1betriebsart

MichaelDvP, 2022-03-13:
> Can you please make an entry for the threshold for canceling the reduce at the outdoor temperature.

Canceling the the reduce mode is one parameter above and already implemented as `noreducetemp` (defaults to -20°C). 
The marked parameter is for switching to reduce (below this temp,  default 5°C) when reducemode is set to outdoorhold.  Above this outdoor temperature the heating switches off in night. 
I can add the last three parameters with settings, Suggestion for names (internal value /  mqtt-name / display name):
```
MAKE_PSTR_LIST(reducetemp, F("reducetemp"), F("off/reduce switch temperature"))
MAKE_PSTR_LIST(vacreducetemp, F("vacreducetemp"), F("vacations off/reduce switch temperature"))
MAKE_PSTR_LIST(vacreducemode, F("vacreducemode"), F("vacations reduce mode"))
```
Is this ok?

brezel1965, 2022-03-13:
I had already found the noreducetemp.  It is as you described with the marked parameter, I use exactly this parameter in connection with a heat pump.  It's a good idea to add all three parameters, the other two I don't currently use.

Your suggestions would be good for me. ;-)

closed 2022-04-30 by proddy