emsesp/EMS-ESP32 #399: Calculate Costs for heating and water

zibous, 2022-03-14:
**Question**
I would like the cost of water heating and determine the heating costs.

[image]

I record the electricity consumption, gas consumption and also the hot water consumption via sensors and the data is available.

As I understand it, the heating circuit is disconnecte when the Boiler is heated, so it should be possible to split determining hot water and heating circuit preparation.

I want to calculate

- Percentage of energy demand for heating
- Percentage of energy demand hot water
- Proportion of gas volume heating
- Percentage of gas volume for hot water

Unfortunately, I can't find a way to do that would be grateful if someone could help me.

**Screenshots**

**Energy Consumption**
[image]

**Gas per hour**
[image]

**Water per hour**
[image]

**Device information**
```
{
  "System": {
    "version": "3.4.0b8",
    "uptime": "002+08:13:45.736",
    "uptime (seconds)": 202425,
    "freemem": 149,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0,
    "analog sensors": 0
  },
  "Network": {
    "connection": "WiFi",
	 ....
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 204131,
    "bus reads (tx)": 46963,
    "bus writes (tx)": 5,
    "bus incomplete telegrams": 11,
    "bus reads failed": 291,
    "bus writes failed": 1,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 64502,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0,
    "API calls": 0,
    "API fails": 0,
    "syslog started": true,
    "syslog level": "debug",
    "syslog ip": "10.1.1.96",
    "syslog queue": 1
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 66,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "74.02",
      "entities": 40,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 "
    },
    {
      "type": "Controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}
```

SpaceTeddy, 2022-03-15:
Hi zibous,
I do this in the following way with fhem and perl script.
- I count the gas value of the gas meter
- checking the ems_esp/boiler_data_wwcharging data every 10 seconds. If ems_esp/boiler_data_wwcharging eq "on" , I save the current gas value in a temp variable
- if ww is finished ems_esp/boiler_data_wwcharging eq "off" I save the actual gas value and substract it with the temp gas value. Then you got a gas value for each ww heating. 
- At the end of the day you can substract the gas value for the day with the ww gas consumption.

but, it get more complex if you have also other gas equipment in house,

greets 
Chris

zibous, 2022-03-17:
> Hi zibous, I do this in the following way with fhem and perl script. 
>     * I count the gas value of the gas meter

Hi Chris,
Thanks for the tip. I did the following for testing:

```yaml
- platform: template
  sensors:
    ## see: https://github.com/emsesp/EMS-ESP32/issues/152
    heating_power_heizkreis_watt:
      friendly_name: "Aktuelle Leistung Raumheizung"
      unit_of_measurement: "W"
      value_template: >-
        {%- if is_state("binary_sensor.boiler_heating_active", "on") -%}
          {{ states('sensor.boiler_burner_current_power')|int(0) }}
        {%- else -%}
          0
        {%- endif -%}
      entity_id:
        - sensor.boiler_burner_current_power
        - binary_sensor.boiler_heating_active

    heating_power_warmwasser_watt:
      friendly_name: "Aktuelle Leistung Warmwasseraufbereitung"
      unit_of_measurement: "W"
      value_template: >-
        {%- if is_state("binary_sensor.boiler_dhw_charging", "on") -%}
          {{ states('sensor.boiler_burner_current_power')|int(0) }}
        {%- else -%}
          0
        {%- endif -%}
      entity_id:
        - sensor.boiler_burner_current_power
        - binary_sensor.boiler_dhw_charging
```

But if I evaluate the data, then I do not understand this, because the heater consumes energy and does not use gas? 

[image]

```sql

SELECT sum("value") FROM "kWh" WHERE ("entity_id" = 'wug_kilowattstunde_gas_stunde') AND $timeFilter GROUP BY time(1h) fill(none)
SELECT sum("value") FROM "autogen"."W" WHERE ("entity_id" = 'heating_power_heizkreis_watt') AND $timeFilter GROUP BY time(1h) fill(null)
SELECT sum("value") FROM "autogen"."W" WHERE ("entity_id" = 'heating_power_warmwasser_watt') AND $timeFilter GROUP BY time(1m) fill(null)

```

Either I make a consideration error or I can not get the right data from the EMS ESP.

marc1de, 2022-03-18:
Maybe you can use the following entities to detect when Gas is used for warmwater or heating:
```
- binary_sensor.boiler_heating_active
- binary_sensor.boiler_gas
- binary_sensor.boiler_dhw_3way_valve_active
```
[image]

zibous, 2022-03-19:
@marc1de , Thanks for this information.

Another attempt via the `boiler_service_code_number`:

```yaml
## ---------------------------------------------------------------------------------
## 14.3 Betriebs- und Störungsanzeigen, die im Display angezeigt werden
## see: https://www.meinhausshop.de/media/pdf/meinhausshop-buderus-GB172-inst.pdf
## ---------------------------------------------------------------------------------
- platform: template

  sensors:

    ## ------------------------------------------------------------------
    ## Sensor for warmwater and heating_circuit
    ## ------------------------------------------------------------------
    heating_warmwater_on:
      friendly_name: "Heizung Warmwasserbetrieb"
      attribute_templates:
         service-code: "{{states('sensor.boiler_service_code_number')|int(0)}}"
         timestamp: "{{now().strftime('%Y-%m-%d %H:%M:%S')}}"
         date: >
             {{ now().timestamp() | timestamp_custom("%Y-%m-%d", true) }}
         time: >
            {{ now().timestamp() | timestamp_custom("%H:%M:%S", true) }}
      value_template: >-
        {%- if states('sensor.boiler_service_code_number')=='201' -%}
         1
        {%- else -%}
         0
        {%- endif -%}

    heating_circuit_on:
      friendly_name: "Heizung Heizkreisbetrieb"
      attribute_templates:
         service-code: "{{states('sensor.boiler_service_code_number')|int(0)}}"
         timestamp: "{{now().strftime('%Y-%m-%d %H:%M:%S')}}"
         date: >
             {{ now().timestamp() | timestamp_custom("%Y-%m-%d", true) }}
         time: >
            {{ now().timestamp() | timestamp_custom("%H:%M:%S", true) }}
      value_template: >-
        {%- if states('sensor.boiler_service_code_number')=='200' -%}
         1
        {%- else -%}
         0
        {%- endif -%}

    ## ------------------------------------------------------------------
    ## service code number text
    ## ------------------------------------------------------------------
    heating_service_code_message:
      friendly_name: "Heizung Meldung"
      attribute_templates:
         service-code: "{{states('sensor.boiler_service_code_number')|int(0)}}"
         timestamp: "{{now().strftime('%Y-%m-%d %H:%M:%S')}}"
         date: >
             {{ now().timestamp() | timestamp_custom("%Y-%m-%d", true) }}
         time: >
            {{ now().timestamp() | timestamp_custom("%H:%M:%S", true) }}
      value_template: >-
          {% set mapper = {
              '208':'Das Gerät befindet sich im Schornsteinfegerbetrieb. Nach 30 Minuten wird der Schornsteinfegerbetrieb automatisch deaktiviert',
              '200':'Das Gerät befindet sich im Heizbetrieb',
              '201':'Gerät im Warmwasserbetrieb',
              '202':'Taktsperre aktiv: Das Zeitintervall für das Wiedereinschalten des Brenners ist noch nicht erreicht',
              '305':'Dauer Warmhaltung bei GB172-24K: Das Zeitintervall für die Wasser-Warmhaltung ist noch nicht erreicht',
              '283':'Der Brenner wird gestartet',
              '265':'Der Wärmebedarf ist geringer als die minimale Wärmeleistung des Geräts. Das Gerät arbeitet im Ein-/Aus-Betrieb',
              '203':'Das Gerät befindet sich in Betriebsbereitschaft, kein Wärmebedarf vorhanden',
[... 26 lines cut ...]
              '215':'Gebläse zu schnell',
              '224':'Wärmeblock-Temperaturbegrenzer oder Abgastemperaturbegrenzer hat ausgelöst',
              '222':'Vorlauftemperaturfühler defekt (Kurzschluss)',
              '223':'Vorlauftemperaturfühler defekt (Unterbrechung)',
              '227':'Flamme wird nicht erkannt',
              '228':'Obwohl Brenner abgeschaltet ist, wird Flamme erkannt',
              '306':'Nach Gasabschaltung: Flamme wird erkannt',
              '231':'Netzspannung war unterbrochen.',
              '261':'Zeitfehler bei erster Sicherheitszeit',
              '280':'Zeitfehler bei Wiederanlaufversuch',
              '235':'KIM nicht kompatibel mit Basiscontroller BC25',
              '360':'KIM nicht erkannt.',
              '230':'Gasarmatur defekt.',
              '234':'Anschlusskabel der Gasarmatur oder Gasarmatur oder Basiscontroller BC25 defekt',
              '238':'Anschlusskabel der Gasarmatur oder Gasarmatur oder Basiscontroller BC25 defekt',
              '233':'KIM defekt',
              '259':'KIM oder Logamatic BC25 defekt'
            }
          %}
          {% set state = states('sensor.boiler_service_code_number') %}
          {% set message = mapper[state] if state in mapper %}
          {{ message }}
      entity_id:
        - sensor.boiler_service_code_number
```

marc1de, 2022-03-19:
yes, I think the solution with the servicecodenumber will also work.....maybe it's just easier to use a variable that already exists (so you can skip the translation form servicecode to a binary status). I have the same hardware than you have (GB172 / 150L Boiler and RC300) and `binary_sensor.boiler_dhw_3way_valve_active` worked perfectly well for me.

_(you could also use the `binary_sensor.boiler_tapwater_active`-value instead, but this is a "generated value" from ems-esp while `dhw_3way_valve_active` is directly related to the boiler)_

 
The only "mistake", that you made in your first attempt, was that you tried to use the `binary_sensor.boiler_dhw_charging` which  does not indicate the regular warmwater heating ( I think this value is related to the one-time heat-up procedure at startup)

zibous, 2022-03-19:
@marc1de 

> I have the same hardware than you have (GB172 / 150L Boiler and RC300) and `binary_sensor.boiler_dhw_3way_valve_active` worked perfectly well for me.

Thanks. 👍  
Yes the `binary_sensor.boiler_dhw_3way_valve_active` is the best way to get the state for the warmwater heating.

You might help me with my second problem. 
I have been trying to find out for several months why the pumps modulation is so high at night. 
I can not explain that and find no settings as I can prevent that.

[image]

It's only funny that it only occurs at night, even if the conditions are almost the same as the day.

**My setting Thermostat:**
```json
{
    "datetime": "12:43:03 19.03.2022",
    "floordry": "off",
    "dampedoutdoortemp": 10.1,
    "floordrytemp": 0,
    "building": "light",
    "minexttemp": -10,
    "wwsettemp": 60,
    "wwmode": "own_prog",
    "wwsettemplow": 45,
    "wwcircmode": "own_prog",
    "wwchargeduration": 60,
    "wwcharge": "off",
    "wwextra1": 0,
    "wwdisinfecting": "on",
    "wwdisinfectday": "we",
    "wwdisinfecttime": 180,
    "hc1": {
        "seltemp": 24,
        "climate": "selTemp",
        "mode": "auto",
        "modetype": "comfort",
        "ecotemp": 20,
        "manualtemp": 24,
        "comforttemp": 24,
        "summertemp": 15,
        "designtemp": 45,
        "offsettemp": 0,
        "minflowtemp": 25,
        "maxflowtemp": 60,
        "roominfluence": 0,
        "roominflfactor": 4,
        "curroominfl": 0,
        "nofrosttemp": 0,
        "targetflowtemp": 36,
        "heatingtype": "radiator",
        "summersetmode": "auto",
        "summermode": "off",
        "controlmode": "optimized",
        "program": "prog_1",
        "tempautotemp": 24,
        "fastheatup": 0
    }
}
```
**Boiler settings:**

```json
{
    "heatingactive": "on",
    "tapwateractive": "off",
    "selflowtemp": 36,
    "selburnpow": 100,
    "heatingpumpmod": 41,
    "outdoortemp": 11.4,
    "curflowtemp": 30.5,
    "burngas": "on",
    "burngas2": "off",
    "flamecurr": 17.8,
    "heatingpump": "on",
    "fanwork": "on",
    "ignwork": "off",
    "oilpreheat": "off",
    "heatingactivated": "on",
    "heatingtemp": 60,
    "pumpmodmax": 100,
    "pumpmodmin": 10,
    "pumpdelay": 5,
    "burnminperiod": 10,
    "burnminpower": 0,
    "burnmaxpower": 100,
    "boilhyston": -6,
    "boilhystoff": 6,
    "setflowtemp": 36,
    "setburnpow": 100,
    "curburnpow": 59,
    "burnstarts": 5303,
    "burnworkmin": 110619,
    "heatworkmin": 100823,
    "ubauptime": 406455,
    "lastcode": "0Y(276) 24.11.2021 03:51 (0 min)",
    "servicecode": "-H",
    "servicecodenumber": 200,
    "maintenancemessage": "H00",
    "maintenance": "date",
    "maintenancetime": 6000,
    "maintenancedate": "07.07.2023",
    "wwsettemp": 57,
    "wwseltemp": 57,
    "wwtype": "buffer",
    "wwcomfort": "eco",
    "wwflowtempoffset": 40,
    "wwmaxpower": 100,
    "wwcircpump": "off",
    "wwchargetype": "3-way valve",
    "wwhyston": -5,
    "wwhystoff": 0,
    "wwdisinfectiontemp": 70,
    "wwcircmode": "2x3min",
    "wwcirc": "off",
    "wwcurtemp": 58,
    "wwcurtemp2": 58,
    "wwcurflow": 0,
    "wwstoragetemp2": 58,
    "wwactivated": "on",
    "wwonetime": "off",
    "wwdisinfecting": "off",
    "wwcharging": "off",
    "wwrecharging": "off",
    "wwtempok": "on",
    "wwactive": "off",
    "ww3wayvalve": "off",
    "wwsetpumppower": 0,
    "wwstarts": 1088,
    "wwworkm": 9796
  }
```

marc1de, 2022-03-19:
huh,...I'm not an expert for this but as far as I know, the heatingpump-modulation depends on the parameters 2.1c to 2.1J in your servicemenue2.
[image]

In my case I use:
2.1C: 0
2.1E: not available
2.1H: 10%
2.1J: 100%

so my heatingpump-modulation should be somehow proportional to the burner power. But in fact, I never monitored this value. I now started a monitoring diagramm in HA, but as far as I can see until now, it raises the modulation to 100% for warmwater and to different heights when the ignition is started for heating (but only very short peaks and sometimes they are so short, that I dont see them in HA. Maybe the different values are also because of my reporting intervall)   

[image]
Sorry, but I have no idea, why your modulation is that high a nighttime. Maybe it's related to different temperatures for day/night, if you use the automated mode. Other cause might be, if you use separate thermostats on your radiators (that might change the relation between "Vor und Rücklauf" and that might cause the heating pump to react somehow). I have none because, I have floor heating and my heating circuits are separated with a "Wärmetauscher".

zibous, 2022-03-20:
@marc1de 

Thank you for your information. 💯 

It could be due to the **1C settings**, so I have to take a look at the service menu. I have installed pressure (DFRobot SEN0257 Analog Water Pressure Sensor) and temperature sensors (DS18B20 Digital temperature sensor), but the pressure sensors have not been working for a month and therefore I only get information about the heating circuit temperature. 

[image]

Unfortunately, I don't get any information from the EMS-ESP (except for the error messages `sensor.boiler_service_code_number`)  about the line pressure, so I can no longer monitor it.

marc1de, 2022-03-20:
Hmmm,....as far as i could see, the DFRobot SEN0257Analog Water Pressure Sensor needs an input voltage of 5V (and offers an output voltage up to 4,5V).  I can't see in your wiring diagramm, which power you use for your sensors....but if you connected them to the 3,3V, the voltage might be too low for them...

and if you connected the sensors to 5V, their output voltage exeeds the allowed range of the esp-inputs (max 3.3V), so that they might be destroyed.
 
**Be careful:** Even if you connected your esp to 5V - the esp32Node MCU uses an voltage regulator and works internaly with 3,3V, so this is the max Input voltage for the gpio's. As far is i know, the dallas-sensors work fine with 3,3V but if you plan to use your waters-sensors with 5V, you need levelshifters or at least 2 resistors as voltage-divider to protect the esp-inputs. 

But btw: very interesting to see, that you have a TFT-Screen connected. How do you control it?

zibous, 2022-03-20:
@marc1de 

> Hmmm,....as far as i could see, the DFRobot SEN0257Analog Water Pressure Sensor needs an input voltage of 5V (and offers an output voltage up to 4,5V). I can't see in your wiring diagramm, which [...]

> and if you connected the sensors to 5V, their output voltage exeeds the allowed range of the esp-inputs (max 3.3V), so that they might be destroyed.

ohhhh, will be the **ESP32** or the **DFRobot SEN0257** destroyed ?
I use the 5V from the ESP, but i have checked the output - no more than 3.00 Volt, because the Pressure is low.

> But btw: very interesting to see, that you have a TFT-Screen connected. How do you control it?

i use this one:
https://www.az-delivery.de/products/az-touch-wandgehauseset-mit-touchscreen-fur-esp8266-und-esp32
and ESP32 with ESPHome App - it works perfect. 
code: [ili9341_heizung.yaml.zip](https://github.com/emsesp/EMS-ESP32/files/8311324/ili9341_heizung.yaml.zip)

Details see: https://peter.siebler.at/posts/warmwasser-erfassen/

marc1de, 2022-03-20:
Thanks for your info about your touchscreen projekt. I'll have a look for it.

Regarding the esp-input-voltages:
> ohhhh, will be the ESP32 or the DFRobot SEN0257 destroyed ?

 Your sensors (SEN0257, Dallas18b20 ...) won't get destroyed because they are build and certified for 5V.
... and I believe, that your esp-inputs most likely will have survived too (if the max. output from the SEN0257 was at 3V, the max input rating of 3,3V at the esp-inputs was never exeeded.....and the dallas18b20 pullup resistor had 4k7 Ohm, so I think, the resulting **voltage** is in fact out of the allowed operational parameters of the esp but the resulting **current** through the 4k7-resistor was too low to destroy your inputs. But correct and better would be, to connect your dallas18b20 to 3,3V). 

The other sensors are unknown to me but you should keep in mind: 
- altough you can connect the esp32 to a 5V power source, this voltage is reduced by a voltage regulator
[image]
so the esp runs completely on 3,3V **and NO INPUT (!!!) may exeed this voltage**. If they do so, you have to reduce this voltage with a levelshifter or with a voltage divider (build with resistors). 

But you can easily check, if your inputs and your programm still works:
- remove one input line from your SEN0257 (eg. GPIO35)
- connect the GPIO to GND or 3,3V or to a voltage divider and you should see the changing values
[image]
If you use a voltage-divider like this (e.g. R1 and R2 as 4k7 Ohm) the output voltage will be 3,3V/2 = 1,65V
(you can use any Resistors between 1K and 10 for this tests)

closed 2023-02-19 by proddy