emsesp/EMS-ESP32 #400: Changing selected room temperature does not work (?)

KeesBleijenberg, 2022-03-14:
I bought a EMS-E32 and upgraded the firmware to version v3.4.0b8 (the latest)
Started with SRT(hc1 selected room temperature) = 18 and HT (hc1 heat temperature) = 18.
Then I change the value of SRT (with the mouse only) to 17.5 and click on send. The UI changes the HT (not the SRT) to 17.5. The SRT stays 18. After a refresh of the UI the values for SRT and HT are 18 again.

Is this a bug or is my understanding of SRT and HT wrong?
The thermostat mode 'heat' or 'auto' doesn't matter.

Is the SRT the value I get after setting a new temperature by hand on the thermostat?
Is the HT the heating temperature in program_a in the thermostat (used in auto mode)?

Where can I find documentation for the values I can change in the UI?

The heating installation is a Bosch 24 HRC Compact 3 and a Bosch FR100 thermostat. 

Kees

MichaelDvP, 2022-03-14:
Please select in the ems-esp gui `help` and download `system info` and `settings`, attach both files here.

The SRT is the reported actual setpoint and is not writable, HT is the setting for heat mode, also used in auto mode in modetype heat. When changing the SRT ems-esp checks the mode/modetype and sets the right writable value, in this case HT is written. After a while the thermostat should report the change it in SRT. If this does not work go to system->system log and set loglevel to all. Open another browserwindow, go there to dashboard and change the SRT, wait some seconds, select the first window and download the log, attach it here.
> Where can I find documentation for the values I can change in the UI?

Ems-esp only shows the values from boiler/thermostat/etc and give access to api/web/mqtt. The documentation for these values are in the handbook of your thermostat provided by Bosch.

KeesBleijenberg, 2022-03-15:
I think SRT should not be equal to heat temperature in mode auto. SRT should be the wanted temperature until setting another temperature.
The current program in the thermostat is 8:00 -> 18 degrees (=heat temperature in a program) and 23:00 -> 12 degrees. I can set the temperature by hand at 19 degrees at say 11:00 and up to 23:00 the temperature is 19 degrees. But tomorrow the temperature at 8:00 will be 18 degrees and not 19 degrees.
If I set the heat temperature at 11:00 to 19 degrees with the gateway, it works today as setting it by hand. But tomorrow the temperature at 8:00 will be 19 degrees. The gateway sets the temperature in the program. 
This has a side effect. The temperature in the program can only be set to integers (not 18.5 i.e). if I set the heat temperature to 18.5 the gateway tries to set the heat temperature in the program. The thermostat rejects therefore (?) the half degree in the program. The result is that the gateway cannot set the temperature to 18.5, 19.5 ... while I can set the temperature by hand to 18.5, 19.5 ...

SRT is  not writable. Why is it editable in the GUI?

I've include the settings and info file and the log while setting the heat temperature to 18.5 degrees. The UI immediately sets the heat temperature to 19 degrees.

[log (2).txt](https://github.com/emsesp/EMS-ESP32/files/8251768/log.2.txt)
[emsesp_settings.txt](https://github.com/emsesp/EMS-ESP32/files/8251771/emsesp_settings.txt)
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/8251772/emsesp_info.txt)

MichaelDvP, 2022-03-15:
Thanks for the data. Let's see the log:

`003+15:31:06.820 TRACE 297211: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 BE 00 B2`
seltemp is reported as 0x00BE => 19.0°C
`003+15:31:07.089 TRACE 297215: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0179), data: 01 08 10 3F 04 0A 18 26 01 FF 01`
eco-temp is 0x18 => 12°C, heat-temp is 0x26 => 19°C

`003+15:33:22.413 TRACE 297352: [emsesp] Me(0x0B) -> Thermostat(0x10), JunkersSet(0x0179), data: 24 (offset 7)`
command seltemp or heattemp sets heat-temp to 0x24 => 18°C
`003+15:33:22.520 TRACE 297356: [emsesp] Thermostat(0x10) -> All(0x00), JunkersSet(0x0179), data: 01 08 10 3F 04 0A 18 24 01 FF 01`
heattemp is set in settings
`003+15:33:24.786 TRACE 297366: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19`
ems-esp reads back the monitor
`003+15:33:24.840 TRACE 297368: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 B4 00 B2`
monitor reports actual seltemp is 0x00B4 => 18°C

`003+15:34:19.393 DEBUG 297456: [command] Calling command 'thermostat/heattemp' (heat temperature) with value 18.5`
`003+15:34:19.625 TRACE 297459: [emsesp] Me(0x0B) -> Thermostat(0x10), JunkersSet(0x0179), data: 25 (offset 7)`
heattemp is set to 0x25 => 18.5°C
`003+15:34:19.753 TRACE 297463: [emsesp] Thermostat(0x10) -> All(0x00), JunkersSet(0x0179), data: 01 08 10 3F 04 0A 18 26 01 FF 01`
But setting reports 0x26 => 19°C
`003+15:34:22.720 TRACE 297479: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 BE 00 B2`
Also seltemp is reported as 0x00BE => 19.0°C

Looks like your thermostat is rounding the temperature to full degrees. 

> The current program in the thermostat is 8:00 -> 18 degrees (=heat temperature in a program) and 23:00 -> 12 degrees. I can set the temperature by hand at 19 degrees at say 11:00 and up to 23:00 the [...]

Seems your thermostat allows a temporary setting, we do not know where it is coded in the telegrams. 
Please make a log (all) and change the the temperatue by hand, we should see in the log which data is changed and can add thsi to ems-esp.

KeesBleijenberg, 2022-03-15:
............
> Looks like your thermostat is rounding the temperature to full degrees.
>
>
> Seems your thermostat allows a temporary setting, we do not know where it is coded in the telegrams. Please make a log (all) [...]

I thought I had already sent the log with logging All.  I've logged it again.
[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/8252537/log.3.txt) 
I set the heat temperature to 18.5 with only the mouse and the UI changes it to 19 degrees.

I'am not sure, but my point is that it makes sense that the gateway cannot update the heat temperature to 18.5 degrees. You can't do it by hand either. But it should be possible to update 'selected room temperature' to 18.5 degrees, without changing the heat temperature. I realize that it's easier said then done. But it would be nice to have this added functionality in the gateway.

MichaelDvP, 2022-03-15:
> I set the heat temperature to 18.5 with only the mouse and the UI changes it to 19 degrees.

Sorry, misunderstood. You've said that you can change the temperature "by hand" and next day the program goes back to normal temperature, A temporary setting only valid until next switchpint. I thought this "by hand" means direct on the thermostat, NOT via gateway. And i want you to make this change on the thermostat while logging. Ems-esp can only set the stored temperatures for nofrost/eco/heat modes, but we do not know how to change the temporary value. With this log i hope to identify this value. 

In the first log there is a unkown telegram `Thermostat(0x10) -> All(0x00), ?(0x01D3), data: 01 00 00`, this could be a candidad for the temporary temperature. 

BTW:
There are also unkown telegrams 0x0230, 0x0231, 0x0232. Looks to me like switch programs, 7 (daily)  sets of 12 bytes, 3 cycles a 4 bytes with meaning temperature-time-temperature-time, The times are in steps of 15 min,
`Thermostat(0x10) -> All(0x00), ?(0x0230), data: 12 18 05 58 FF FF FF FF FF FF FF FF ` 
program A starts Monday heat 18°C (0x12) at 06:00 (0x18) and nofrost 5°C 0x05 at 22:00 (0x58), cycle 2 and 3 not set
`Thermostat(0x10) -> All(0x00), ?(0x0231), data: 12 18 0C 20 12 44 05 58 FF FF FF FF `
program B starts Monday heat 18°C 0x12 at 06:00 (0x18), eco 12°C ( 0x0C) at 08:00 (0x20), heat at 17:00 (0x44) and nofrost 0x05 at 22:00 (0x58)
After changing heattemp to 18->19°C all switchprograms changes 0x12 to 0x13. I think that's the reason why they are broadcasted. 
But i thought Junkers have 6 switchprograms(A-F), these are only 3?

KeesBleijenberg, 2022-03-15:
Your are right. The Bosch Fr100 thermostat has room for 6 programs. Maybe I have only used 3 of them. I'll try to find out.

MichaelDvP, 2022-03-15:
There is nothing published when changing the temperature, the thermostat monitors 18°C (00 B4) and next 18.5°C (00 B9)
```
003+18:49:55.129 TRACE 311952: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 B4 00 BF
003+18:50:37.853 TRACE 312005: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 B9 00 BF
 ```
There is only one little chance that we can write to monitor-telegram, i've seen this in #336 between thermostats. 
Please use telnet (enable in settings, use putty or another telnet, port 23), activate superuser by `su` command (password as set in web security, default `ems-esp-neo` and then type this command:
`call system send "0B 10 FF 02 00 6F 00 C8"`
this will change seltemp to 20°, or do nothing.

KeesBleijenberg, 2022-03-16:
I changed the temperature with the gateway in telnet I did: read 10 183 but got 'empty data'. 

The last comments we (?) thought that the problem is that the fr100 has a special temperature value, used if you change the temperature in auto mode. 
Let's make things simpeler. I set the thermostat mode from auto to heat. In my understanding the thermostat in heat mode acts as a normal thermostat. No program runs and the wanted temperature is the temperature you set by hand. There is only one temperature: the heat temperature. 
I reboot the gateway -> in the web interface hc1 mode is heat and hc1 mode type is also heat. The hc1 heat temperature is 18.
I set in the web interface the heat temperature to 18.5. The web interface loads again and the hc1 heat temperature is 19 degrees.
This is the log: [log (5).txt](https://github.com/emsesp/EMS-ESP32/files/8260278/log.5.txt)

Then I change the selected room temperature:
I reset the heat temperature to 18 degrees and set the selected room temperature to 18.5. The heat temperature is again 19 degrees.
This is the log: [log (6).txt](https://github.com/emsesp/EMS-ESP32/files/8260281/log.6.txt)

Setting the temperature to 18.5 degrees has the same problems in auto and heat mode. Maybe the problem for the gateway is not the extra value (that is another problem), but just setting the temperature to 18.5 degrees. For the thermostat this is no problem (I can set it by hand) but the gateway fails to achieve the same. 

Additional questions:
When I turned the knob on the thermostat to set it to heat mode the thermostat asked (in Dutch) Do you want to stop netcom Yes/No? I cannot find what it is and what I should answer. Any ideas?

Isn't it strange that you can change the thermostat mode in the web interface? On the thermostat the mode is selected by a knob!

MichaelDvP, 2022-03-16:
>  I did: read 10 183 but got 'empty data'.

And what about the other telegrams? I'm sure the telegram 1D3 isn't empty.
> Let's make things simpeler. 

Sorry, imho there is no simpler way, To change a value from ems-esp we need to know the telegram and position of this value. If the thermostat does not broadcast the value on change the only chance is to read some possible telegrams and search.

@norberts1 knows nearly all about ht3 systems, maybe he have a idea if, or how, the temporary temperature is changeable on FR100. Short description for @norberts1: changing the temperature via thermostat in auto mode only sends the new temperature in ID 367, pos 8/9 (your numbering) , there is no field in ID 377 to set it, writing to ID 367 is not possible. Is there a way to change the temporary set temperature without affecting stored temperatures in ID 377 11/12/13?

For the additional questions: Sorry, i have also no idea.

MichaelDvP, 2022-03-16:
BTW: i've read in the FR100 manual to understand the programs: Programs A, B, C uses the stored temperatures heat/eco/nofrost, thats why they are broadcasted if one of these temperatures change, programs D, E, F use individual temperatures, changing heat-temp does not affect them, so they are not send out. I'm pretty sure that a `read 10 233` will show up program D. `read 10 234` program E, etc. 
For hc2 i guess that it starts with `read 10 236` for prog A. Can you check this?

KeesBleijenberg, 2022-03-16:
I'am a bit lost. This is what I've done:
Set the temperature on the thermostat in auto mode to 19 degrees. Normally it is 18 degrees.
Then I started a telnet session in putty. 
┌──────────────────────────────────────┐
│ EMS-ESP version 3.4.0b8              │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└──────────────────────────────────────┘

ems-esp:$ su
Password:
000+00:03:14.058 N 0: [shell] su session opened on console
ems-esp:# read 10 183
000+00:05:50.288 I 1: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x0183), data: <empty>
ems-esp:# read 10 18d
000+00:06:18.655 I 2: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x018D), data: <empty>
ems-esp:# read 10 197
000+00:06:29.498 I 3: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x0197), data: <empty>
ems-esp:# read 10 1d3
000+00:06:37.190 I 4: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x01D3), data: 01 00 00
ems-esp:# read 10 233
000+00:08:07.487 I 5: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x0233), data: <empty>
ems-esp:# read 10 234
000+00:08:14.590 I 6: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x0234), data: <empty>
ems-esp:# read 10 235
000+00:08:18.602 I 7: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x0235), data: <empty>
ems-esp:#

MichaelDvP, 2022-03-16:
Thanks, now we know where the temporary temperature and the programs D, E, F are NOT stored and that my theories were wrong.
Sorry, but i'm out of ideas where to search.

KeesBleijenberg, 2022-03-17:
Sorry to hear you are running out of ideas.

I'am a bit lost. Could you comment:
When I set the thermostat by hand to 'heat' mode then no program runs on the thermostat. Correct?

When I set the temperature on the thermostat in heat mode by hand from 18 to 18.5 degrees some message must be sent to the boiler, because the boiler reacts to this higher temperature. After that the thermostat should broadcast I'am set to 18.5 degrees but it doen't. Correct?

There is a EMS command to change the temperature of the thermostat. If you send to the the thermostat the command: set yourself to 18 degrees everything goes fine, but when you send the command set yourself to 18.5 degrees the thermostat sets itself to 19 degrees and broadcasts: I am set to 19 degrees. Correct?

You mentioned that @norberts1 is an expert. Did you sent a question to him or do I have to sent him a question?

MichaelDvP, 2022-03-18:
A good start to understand the ems-bus is [https://emsesp.github.io/docs/#/How-the-EMS-bus-works](https://emsesp.github.io/docs/#/How-the-EMS-bus-works), and, if you can read german, [https://emswiki.thefischer.net/doku.php?id=start](https://emswiki.thefischer.net/doku.php?id=start). 
All calculation regarding room temperature are done by the thermostat. It calculates the the requiredflow temperature and sends it to the boiler. For your boiler this is in the log: `Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 37` The boiler then controlls the burner to reach this temperature.
How the thermostat calculates the flowtemperature depends on different parameters, stored in the thermostat. 
The thermostat send out periodically his status with some values like mesured roomtemp, setpoint, mode, etc. In your log: `Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 BE 00 B2`
The values you can read in Norberts doku on page 28: [https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf)
He uses a different counting methode and counts also the header, which is `90 00 FF  00 00 6F` in this case (means: Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F)).
If we want to change a value there are setting telegrams, which can be written. In your thermostat this is: `Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0179), data: 01 08 10 3F 04 0A 18 24 01 FF 01` , you can find it on page 29 of Norberts list.
These settings are normaly not send out, only if something changes. The methode to find a value is to change the value "by hand" on the thermostat and log the messages. But your thermostat only reports the changed setpoint in the Monitor-telegram and publishes no setting for this. If you change heat-temp by hand  to 19°C the thermostat will publish `Thermostat(0x10) -> All(0x00), JunkersSet(0x0179), data: 26 (offset 7)`  and on next read the data of 0x179 shows as `01 08 10 3F 04 0A 18 26 01 FF 01` (data are hex notation and setpoint is in 0,5°C steps, 26hex = 38dez = 19°C)
Other thermostats like RC300 have a field in settings that is called temporary auto temperature (see page 34, byte 14  in Norberts list) , but there is nothing like that in 0x179. And a second settings telegram for this thermostat is not known (also there is nothing like that on norberts list).

If someone is mentioned on github with @ sign, he will get a copy of the message, so norbert has get the question.

KeesBleijenberg, 2022-03-18:
Thank you for all your efforts and information.  I have some reading to do.

norberts1, 2022-03-19:
@MichaelDvP yes, I saw your messages around that temporary temperature settings on Fxyz-thermostats. 
4 years ago I have checked this on my FW100 thermostat and had compare able problems to setup this temporary request.
As you mentioned above with thermostats like RC300 and Bosch Cxyz it's possible with it's own command.
Now I own one CW400 in my system and can't check it anymore. All I have are old binary files created with FW100.
This files I can check mid next week.
Result: It seems **for me** the temporary temperature  setup (controlled by any modem) on Fxyz-thermostats is not implemented. Only the manual changes on the thermostats results in commands to the boiler (as you mentioned above).
Bud this is what I have found, it must not be true.
@MichaelDvP is there any option in the EMS-ESP to stream binary heater-busdata for logging to one interface?
Then I can support you a bit more using that HT3_Analyser.py from my project.

MichaelDvP, 2022-03-20:
@norberts1 Thanks for the feedback. 
> is there any option in the EMS-ESP to stream binary heater-busdata for logging to one interface?

Not as uart output from esp, but someone can connect a raspberry direct to gateway replacing the esp or in parallel to esp. 
But i think your analyser is mainly decoding the know telegrams from your list and only logging the unknown. telegrams. 

I have a buderus system with RC35 and can not test myself. Someone with this FR100 and a lot of enthusiasm and spare time has to test for unpublished telegrams. Maybe a telegram scanner will be helpfull, but reading all telegrams from 0x0001 to 0xFFFF will take a while to record and to analyse.

Also the fact that the thermostat does not publish a setting when changing the temperature, makes it plausible that there is no  way to set via bus.

norberts1, 2022-03-26:
@MichaelDvP yes, the analyser is mainly for known telegrams and logging for unknowns.
But it can load also binary logfiles from any other heater-system and try to decode them or showing still unknown telegrams.
This is then an iterative process to optimize the decoding also for heater-system with other configurations then I own.

In parallel I used that binary-files to load them into Libre-Office and used that integrated data-filter for extracting the wanted informations. This helped me to decode more telegrams then I can create myself.
See attached file as example for my older FW100.

[FW100_sparenheizentagfrostauto_comments.zip](https://github.com/emsesp/EMS-ESP32/files/8355959/FW100_sparenheizentagfrostauto_comments.zip)

closed 2023-02-19 by proddy