emsesp/EMS-ESP32 #402: RC35 - summertemp -  Permanently off option

majdzik84, 2022-03-15:
Hello.
On my RC35 I have the option to set the summer temperature. The minimum value is 10C and then an option is shown to me
"Permanently off." So it immediately turns on the summer mode. I would like to be able to enable this option via ems. Is it to be done?
I have a heat pump currently installed and I would like it to be the main source of heat. I thought that the gas boiler should only turn on when it is below the set value outside. The option I am writing about switches the stove into standby mode, so the pumps also do not work, which I like because the circulation pumps from the heat pump will work.
I would be grateful for having a look at this.

MichaelDvP, 2022-03-16:
I have also a RC35, setting summertemp to "off"  is logged as:
`Thermostat(0x10) -> All(0x00), RC35Set(0x47), data: 09 (offset 22)`
So you can set in ems-esp the temperature to 9°C to switch permanently off.
I think an extra option is not needed.

majdzik84, 2022-03-16:
Cool. Thank you for this information. I will test it in the evening.

MichaelDvP, 2022-03-16:
I've tested this morning, outdoortemp was 4°C, setting summertemp to 9 via ems-esp switches to summermode. But it takes a minute to show up, so be patient.

majdzik84, 2022-03-17:
Works great. Thank you.

closed 2022-03-17 by majdzik84