emsesp/EMS-ESP32 #406: Boiler LOGAMAX PLUS GB172-14 & Controller BC25 & RC310 & - system pressure ?

zibous, 2022-03-20:
**Question**
Is there a way to get the **system pressure**? 

**Device information**
```json
{
  "System": {
    "version": "3.4.0b8",
    "uptime": "002+08:13:45.736",
    "uptime (seconds)": 202425,
    "freemem": 149,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0,
    "analog sensors": 0
  },
  "Network": {
    "connection": "WiFi",
	 ....
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 204131,
    "bus reads (tx)": 46963,
    "bus writes (tx)": 5,
    "bus incomplete telegrams": 11,
    "bus reads failed": 291,
    "bus writes failed": 1,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 64502,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0,
    "API calls": 0,
    "API fails": 0,
    "syslog started": true,
    "syslog level": "debug",
    "syslog ip": "10.1.1.96",
    "syslog queue": 1
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 66,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "74.02",
      "entities": 40,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 "
    },
    {
      "type": "Controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}	  
```

MichaelDvP, 2022-03-20:
Only a few boilers have a pressure sensor, afaik only hig power burners, checking system pressure to ensure that pump runs before starting burner (prevent overheating). 
You can check in telegram 18
```
Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 43 02 BB 64 64 09 01 15 01 01 CF 7D 00 80 00 02 0A FF 2D 48 00 00 FF 00 10 83 00
                                                                                                          ^^
```
The marked FF means there is no sensor. My boiler electronics does not have a connector for a pressure sensor.

But you can add an analog sensor to emsesp, name it pressure and scale to bar. Analog input is max. 1500mV, so you need a resistor network or use a sensor with higher pressure range. I use a cheap 5V supply, 0,5-4,5V output sensor with 1,2MPa (12bar) direct on input, giving a (not very exact) measurement range of 3.3 bar@1500mV. But for checking for leakage or refilling it's good enough. The sensor [is like this](https://www.ebay.de/itm/194075389836?hash=item2d2fcb6f8c:g:iswAAOSwDRFgiq9V), but was twoyears ago less 10 €.

zibous, 2022-03-20:
@MichaelDvP 
Thanks but how can i check the telegram 18 ?
When i use telnet and log=all i can'nt see `Boiler(0x08) -> All(0x00)`??

```
001+04:04:00.285 T 5: [telegram] Rx: 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00 9F
001+04:04:00.751 T 6: [telegram] Rx: 08 00 34 00 39 02 52 02 52 21 00 00 03 00 00 26 6A 00 04 42 00 80 00 85
001+04:04:00.956 T 7: [telegram] Rx: 08 00 18 00 23 01 64 64 17 09 01 25 40 80 00 02 52 80 00 00 68 FF 2D 48 00 C8 00 02 00 86
001+04:04:01.732 D 8: [emsesp] Fetching values for deviceID 0x08
001+04:04:01.732 D 9: [telegram] Tx read request to device 0x08 for type ID 0x14
001+04:04:01.732 D 10: [telegram] Tx read request to device 0x08 for type ID 0x16
001+04:04:01.732 D 11: [telegram] Tx read request to device 0x08 for type ID 0x33
001+04:04:01.732 D 12: [telegram] Tx read request to device 0x08 for type ID 0x26
001+04:04:01.820 D 13: [telegram] Sending read Tx [#126], telegram: 0B 88 14 00 1B
001+04:04:01.820 T 14: [telegram] Rx: 0B 88 14 00 1B DF
001+04:04:01.862 D 15: [emsesp] Last Tx read successful
001+04:04:01.862 T 16: [telegram] Rx: 08 0B 14 00 06 39 E0 98
001+04:04:02.184 D 17: [telegram] Sending read Tx [#127], telegram: 0B 88 16 00 1B
001+04:04:02.196 T 18: [telegram] Rx: 0B 88 16 00 1B D7
001+04:04:02.266 D 19: [emsesp] Last Tx read successful
001+04:04:02.266 T 20: [telegram] Rx: 08 0B 16 00 FF 3C 64 00 06 FA 0A 01 05 64 0A 02 00 00 00 00 00 00 00 00 00 00 00 00 28 00 2D AD
001+04:04:02.516 D 21: [telegram] Sending read Tx [#130], telegram: 0B 88 16 1B 20
001+04:04:02.516 T 22: [telegram] Rx: 0B 88 16 1B 20 DA
001+04:04:02.556 D 23: [emsesp] Last Tx read successful
001+04:04:02.557 T 24: [telegram] Rx: 08 0B 16 1B 00 B6
001+04:04:02.881 D 25: [telegram] Sending read Tx [#128], telegram: 0B 88 33 00 1B
001+04:04:02.881 T 26: [telegram] Rx: 0B 88 33 00 1B 43
001+04:04:02.949 D 27: [emsesp] Last Tx read successful
001+04:04:02.949 T 28: [telegram] Rx: 08 0B 33 00 08 FF 39 FB 00 28 00 02 46 D8 FF FF 00 6F
001+04:04:03.180 D 29: [telegram] Sending read Tx [#129], telegram: 0B 88 26 00 1B
001+04:04:03.180 D 30: [emsesp] Fetching values for deviceID 0x10
001+04:04:03.180 D 31: [telegram] Tx read request to device 0x10 for type ID 0x2A5
001+04:04:03.180 D 32: [telegram] Tx read request to device 0x10 for type ID 0x2B9
001+04:04:03.180 D 33: [telegram] Tx read request to device 0x10 for type ID 0x2AF
001+04:04:03.180 D 34: [telegram] Tx read request to device 0x10 for type ID 0x29B
001+04:04:03.180 D 35: [telegram] Tx read request to device 0x10 for type ID 0x2F5
001+04:04:03.180 D 36: [telegram] Tx read request to device 0x10 for type ID 0x31B
001+04:04:03.180 D 37: [telegram] Tx read request to device 0x10 for type ID 0x23A
001+04:04:03.180 D 38: [telegram] Tx read request to device 0x10 for type ID 0x240
001+04:04:03.252 T 39: [telegram] Rx: 0B 88 26 00 1B 17
001+04:04:03.305 D 40: [emsesp] Last Tx read successful
001+04:04:03.305 T 41: [telegram] Rx: 08 0B 26 00 01 05 00 15 00 1E 52 64 23
001+04:04:03.667 D 42: [mqtt] Publishing topic ems-heizung/heartbeat (#32335, retain=0, retry=1, size=281, pid=1)
001+04:04:03.671 D 43: [telegram] Sending read Tx [#131], telegram: 0B 90 FF 00 19 01 A5
001+04:04:03.692 T 44: [telegram] Rx: 0B 90 FF 00 19 01 A5 FD
001+04:04:03.809 D 45: [emsesp] Last Tx read successful
001+04:04:03.811 T 46: [telegram] Rx: 10 0B FF 00 01 A5 80 00 01 30 23 00 30 28 01 E7 03 03 01 01 E7 02 33 00 00 11 01 03 FF FF 00 04
001+04:04:04.107 D 47: [telegram] Sending read Tx [#139], telegram: 0B 90 FF 19 20 01 A5
001+04:04:04.127 T 48: [telegram] Rx: 0B 90 FF 19 20 01 A5 D1
001+04:04:04.202 D 49: [emsesp] Last Tx read successful
001+04:04:04.202 T 50: [telegram] Rx: 10 0B FF 19 01 A5 06 04 00 00 00 00 FF 64 3C 00 3C 01 FF 01 02 BF
001+04:04:04.472 D 51: [telegram] Sending read Tx [#132], telegram: 0B 90 FF 00 19 01 B9
001+04:04:04.491 T 52: [telegram] Rx: 0B 90 FF 00 19 01 B9 E1
001+04:04:04.596 D 53: [emsesp] Last Tx read successful
001+04:04:04.596 T 54: [telegram] Rx: 10 0B FF 00 01 B9 FF 2E 30 26 28 03 00 FF FF 00 30 01 E1 20 01 0F 05 00 00 01 22 B9
001+04:04:04.903 D 55: [telegram] Sending read Tx [#133], telegram: 0B 90 FF 00 19 01 AF
001+04:04:04.919 T 56: [telegram] Rx: 0B 90 FF 00 19 01 AF F7
001+04:04:04.991 D 57: [emsesp] Last Tx read successful
001+04:04:04.991 T 58: [telegram] Rx: 10 0B FF 00 01 AF 00 28 00 00 2D 2D 0F 01 19 1E 00 02
001+04:04:05.235 D 59: [telegram] Sending read Tx [#134], telegram: 0B 90 FF 00 19 01 9B
001+04:04:05.235 T 60: [telegram] Rx: 0B 90 FF 00 19 01 9B C3
001+04:04:05.314 D 61: [emsesp] Last Tx read successful
001+04:04:05.314 T 62: [telegram] Rx: 10 0B FF 00 01 9B 01 01 00 00 00 02 00 30 3C DA
001+04:04:05.635 D 63: [telegram] Sending read Tx [#135], telegram: 0B 90 FF 00 19 01 F5
001+04:04:05.729 T 64: [telegram] Rx: 0B 90 FF 00 19 01 F5 AD
001+04:04:06.130 D 65: [telegram] Sending read Tx [#140], telegram: 0B 90 FF 00 19 01 F5
001+04:04:06.130 T 66: [telegram] Rx: 0B 90 FF 00 19 01 F5 AD
001+04:04:06.253 D 67: [emsesp] Last Tx read successful
001+04:04:06.253 T 68: [telegram] Rx: 10 0B FF 00 01 F5 01 00 04 03 00 FF 0C 02 00 08 04 00 00 00 00 00 00 00 00 00 01 7F
001+04:04:06.627 D 69: [telegram] Sending read Tx [#136], telegram: 0B 90 FF 00 19 02 1B
001+04:04:06.644 T 70: [telegram] Rx: 0B 90 FF 00 19 02 1B 45
001+04:04:06.684 D 71: [emsesp] Last Tx read successful
001+04:04:06.684 T 72: [telegram] Rx: 10 0B FF 00 02 1B 3C 2D FD
001+04:04:06.958 D 73: [telegram] Sending read Tx [#137], telegram: 0B 90 FF 00 19 01 3A
001+04:04:06.958 T 74: [telegram] Rx: 0B 90 FF 00 19 01 3A 62
001+04:04:07.008 D 75: [emsesp] Last Tx read successful
001+04:04:07.008 T 76: [telegram] Rx: 10 0B FF 00 01 3A 00 75 41
001+04:04:07.257 D 77: [telegram] Sending read Tx [#138], telegram: 0B 90 FF 00 19 01 40
```

MichaelDvP, 2022-03-20:
You have raw-mode activated, Telegram 18 is here:
```
001+04:04:00.956 T 7: [telegram] Rx: 08 00 18 00 23 01 64 64 17 09 01 25 40 80 00 02 52 80 00 00 68 FF 2D 48 00 C8 00 02 00 86
                                                                                                    ^^
```
also with FF in offset 17, no pressure sensor.

zibous, 2022-03-20:
@MichaelDvP 
Thanks. 👍

closed 2023-02-19 by proddy