emsesp/EMS-ESP32 #407: EMS-ESP Version v3.4.0b8: Please add manufacturer to EMS-ESP Thermostat

zibous, 2022-03-22:
**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

Homeassistant: The **manufacturer** is empty (missing) for the EMS-ESP Thermostat

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

[image]

```
unique_id: thermostat_datetime
command_topic: ems-heizung/thermostat/datetime
state_topic: ems-heizung/thermostat_data
name: Thermostat date/time
value_template: '{{value_json.datetime}}'
device:
  identifiers:
    - ems-esp-thermostat
  name: EMS-ESP Thermostat
  manufacturer: ''  <----
  model: RC300/RC310/Moduline 3000/1010H/CW400/Sense II
  via_device: ems-esp
platform: mqtt
```

MichaelDvP, 2022-03-22:
The brand is coded in telegram 0x02 offset 9, but not all devcies reports it. Please check in terminal `read 10 2` or `read 10 2 9`. 
Known codes are:
```
        BOSCH,        // 1
        JUNKERS,      // 2
        BUDERUS,      // 3
        NEFIT,        // 4
        SIEGER,       // 5
        WORCESTER,    // 11
        IVT           // 13
```
If your thermostat have a unknown positive number in offset 9, tell us the number and the brand and we can add it.

zibous, 2022-03-22:
@MichaelDvP 
Sorry but i do not get the  telegram 0x02 offset 9...

**telnet**
   ```
      su 
      log all

    003+02:22:06.309 D 339: [emsesp] Fetching values for deviceID 0x10
    003+02:22:06.309 D 340: [telegram] Tx read request to device 0x10 for type ID 0x2A5
    003+02:22:06.309 D 341: [telegram] Tx read request to device 0x10 for type ID 0x2B9
    003+02:22:06.309 D 342: [telegram] Tx read request to device 0x10 for type ID 0x2AF
    003+02:22:06.309 D 343: [telegram] Tx read request to device 0x10 for type ID 0x29B
    003+02:22:06.309 D 344: [telegram] Tx read request to device 0x10 for type ID 0x2F5
    003+02:22:06.309 D 345: [telegram] Tx read request to device 0x10 for type ID 0x31B
    003+02:22:06.309 D 346: [telegram] Tx read request to device 0x10 for type ID 0x23A
    003+02:22:06.309 D 347: [telegram] Tx read request to device 0x10 for type ID 0x240
    003+02:22:06.392 T 348: [telegram] Incomplete Rx: 0B 88 26 00 1B 17 08 0B 26 00 01 05 00 15 00 1E 52 64
    003+02:22:06.663 T 349: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 24 64 00
    003+02:22:06.663 D 350: [emsesp] Process setting 0x1A for device 0x08
    003+02:22:06.705 T 351: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x0267), data: 00 00
    003+02:22:06.963 T 352: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x02A5), data: 80 00 01 30 24 00 30 28 02 4D 03 03 01 02 4D 01 CD 00 00 11 01 03 FF FF 00
    003+02:22:07.178 T 353: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
    003+02:22:07.252 D 354: [mqtt] Publishing topic ems-heizung/heartbeat (#85296, retain=0, retry=1, size=287, pid=1)
    003+02:22:07.506 T 355: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 39 01 EA 01 EA 01 00 00 03 00 00 26 9D 00 04 45 00 80 00
    003+02:22:07.735 T 356: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 24 01 2D 64 00 01 01 20 40 80 00 01 EA 80 00 00 00 FF 30 59 00 CC 00 00 00
    003+02:22:08.000 D 357: [telegram] Sending read Tx [#207], telegram: 0B 88 26 00 1B
    003+02:22:08.080 T 358: [emsesp] Me(0x0B) <- Boiler(0x08), UBASettingsWW(0x26), data: 1B
    003+02:22:08.122 D 359: [emsesp] Last Tx read successful
    003+02:22:08.127 T 360: [emsesp] Boiler(0x08) -> Me(0x0B), UBASettingsWW(0x26), data: 01 05 00 15 00 1E 52 64
    003+02:22:08.277 T 361: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x02A5), data: 08 04 00 00 00 00 FF 64 3C 00 3C 01 FF 01 02 (offset 25)
    003+02:22:08.468 T 362: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x031D), data: 00 00 0A 07
    003+02:22:08.763 D 363: [telegram] Sending read Tx [#199], telegram: 0B 90 FF 00 19 01 A5
    003+02:22:08.763 T 364: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Monitor(0x02A5), data: 19
    003+02:22:08.894 D 365: [emsesp] Last Tx read successful
    003+02:22:08.894 T 366: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x02A5), data: 80 00 01 30 24 00 30 28 02 4D 03 03 01 02 4D 01 CD 00 00 11 01 03 FF FF 00
    003+02:22:09.295 D 367: [telegram] Sending read Tx [#208], telegram: 0B 90 FF 19 20 01 A5
    003+02:22:09.595 T 368: [telegram] Incomplete Rx: 0B 90 FF 19 20 01 A5 FF
    003+02:22:09.855 D 369: [telegram] Sending read Tx [#209], telegram: 0B 90 FF 19 20 01 A5
    003+02:22:09.875 T 370: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Monitor(0x02A5), data: 20 (offset 25)
    003+02:22:09.951 D 371: [emsesp] Last Tx read successful
    003+02:22:09.951 T 372: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x02A5), data: 08 04 00 00 00 00 FF 64 3C 00 3C 01 FF 01 02 (offset 25)
    003+02:22:10.095 D 373: [mqtt] Publishing topic ems-heizung/boiler_data (#85297, retain=0, retry=1, size=747, pid=1)
    003+02:22:10.196 D 374: [mqtt] Publishing topic ems-heizung/boiler_data_ww (#85298, retain=0, retry=1, size=512, pid=1)
    003+02:22:10.220 D 375: [telegram] Sending read Tx [#200], telegram: 0B 90 FF 00 19 01 B9
    003+02:22:10.240 T 376: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Set(0x02B9), data: 19
```

**Syslog**
```
       /var/log# tail -f syslog | grep "0x02"
       --> no output
```

I have RC310 Thermostat (`RC310 Regelgerät weiss Buderus`):

```
{
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "74.02",
      "entities": 40,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 "
    },
```

MichaelDvP, 2022-03-22:
Again: You have to check in terminal with the command `read 10 2` or `read 10 2 9`!

zibous, 2022-03-22:
@MichaelDvP 
ok. --> Result:

```
ems-heizung:$ read 10 2
003+02:38:18.192 I 1151: [emsesp] Thermostat(0x10) -> Me(0x0B), Version(0x02), data: 9E 4A 02 00 FF 00 00 00 00
ems-heizung:$ read 10 2 9
003+02:38:37.503 I 1152: [emsesp] Thermostat(0x10) -> Me(0x0B), Version(0x02), data: <empty> (offset 9)
```

MichaelDvP, 2022-03-22:
So there is no brand stored in the thermostat (manufacturer is always Bosch). Sorry,

proddy, 2022-03-26:
nothing we can do here, sorry.

closed 2022-03-26 by proddy