emsesp/EMS-ESP32 #408: NTP does not work

KeesBleijenberg, 2022-03-22:
In the web interface of the gateway (version v3.4.0b8) I set the NTP server to time.google.com and click on save. The UTC-time does not change and stays jan 11 1970 ( zero?) even after a day or so, Another ntp-server like ntp.time.nl  makes no difference. What am I doing wrong?

The system has a Bosch Fr100 thermostat. This thermostat has also a time This time is correct, but I can't see it in the web interface.
How does the NTP time works with the thermostat time. 
Is the thermostat time ignored?

MichaelDvP, 2022-03-22:
Some thoughts:
Check your network router, is ems-esp allowed to contact the external server? 
Often the router also acts as ntp-server for internal devices, try this. 
The NTP status in emsesp shows only if the service is started, not if gets a answer,that is maybe confusing.
If you disable NTP, you can set the clock of emsesp manually.
The thermostat clock should show in dashboard thermostat, it is read in all thermostats. But it is not very exact and after powercut it is not set and need manual setting (or viea emsesp-ntp). We have a option to set the thermostat clock from NTP, but there is no option to sync the emsesp to the thermostat, NTP is the better option to sync and there is no need to have two sync sources.

MichaelDvP, 2022-03-22:
Maybe a good dea to combine this with #386. 
When receiving the thermostat time check 
- ems-esp has valid time (ntp or manual) and thermostat differs (+-15 sec) and is writeable  -> set thermostat clock to emsesp-clock
- ems-esp has no valid time, thermostat clock is set ->set ems-esp clock to thermostat cloc

KeesBleijenberg, 2022-03-23:
> Some thoughts:
> Check your network router, is ems-esp allowed to contact the external server?
> Often the router also acts as ntp-server for internal devices, try this.
> The NTP status in emsesp shows only if the service is started, not if gets a answer,that is maybe confusing.

This morning the UTC time was 18:42:39 jan 12 1970. After a reboot the time was 18:52:05 and now the UTC-time and datea are correct. I don't understand what has happened, but it works.

> If you disable NTP, you can set the clock of emsesp manually.

I tried this:
- disabled NTP
- In the dashboard I double click the thermostat edit date/time and changed the time (a few minutes later) and I click on send => Write command failed

<snip>

MichaelDvP, 2022-03-23:
If you disable NTP, you have a time setting button in NetworkTime-> Status to set the ESP clock. 
The thermostat clock in dashboard needs for setting also day of week and daylightsaving as single char, this is shown in the input filed and [documented here](https://emsesp.github.io/docs/#/Command?id=thermostats-thermostat).

MichaelDvP, 2022-03-30:
@KeesBleijenberg Have you checked the latest version? There are some improvements and you can check the the sync status in web/mqtt/api. The status now shows if there is realy a time received from ntp server.
Please check and close the issue if ok.

KeesBleijenberg, 2022-03-31:
> @KeesBleijenberg Have you checked the latest version? There are some improvements and you can check the the sync status in web/mqtt/api. The status now shows if there is realy a time received from ntp server.
> Please check and close the issue if ok.

This is what I've done:
Upgraded to version EMS-ESP Version v3.4.0b10 (verified in the system tab)
I click on Network Time and opened tab NTP Settings.
There I check enable NTP
Server: time.google.com 
Time zone: Europe/Amsterdam 
When I go to tab NTP-status I see Status: Inactive.
The local time is 15:34:00 which is incorrect (it is 15:23). I think this is the time on the thermostat. 
Is NTP inactive because there is a thermostat time? If that is the case maybe "using thermostat time' is better then 'inactive'.

MichaelDvP, 2022-03-31:
> Is NTP inactive because there is a thermostat time?

No, ntp-status shows the status of received ntp times now, if the server does not answer and no time received within the last two hours the status goes to inactive. Try a different ntp-server. If you push the save-button in settings the ntp is restarted and trys immediatly to contact the server, the status should go to active if the server replys.

The esp32 contains a real time clock with good accuracy, the thermostats often have only a rc oszillator and option to correct the clock. If there is no real reference like ntp and both clocks have a valid, but maybe wrong, time, an automatic sync makes no sense, and manual sync will log a warning. Syncing esp clock to thermostat clock only makes sense if the esp has no valid time setting.

MichaelDvP, 2022-03-31:
Have you blocked emsesp from contacting the internet in your router? 
If your router is configured as ntp server, then try to use this. 
Is the dns server set in emsesp? Try ip address for ntp server: `216.239.35.0` (time.google.com).

proddy, 2022-04-01:
@MichaelDvP eltoro on [Discord](https://discord.com/channels/816637840644505620/816638587508490250/959343020573065246)  is getting the log flooded with these errors after the latest build:

````
000+08:46:16.769 I 22: [thermostat] thermostat time correction from ntp
000+08:46:17.618 I 23: [thermostat] thermostat time correction from ntp
000+08:46:19.006 I 24: [thermostat] thermostat time correction from ntp
000+08:46:19.905 I 25: [thermostat] thermostat time correction from ntp
000+08:46:21.280 I 26: [thermostat] thermostat time correction from ntp
000+08:46:21.819 I 27: [thermostat] thermostat time correction from ntp
000+08:46:22.569 I 28: [thermostat] thermostat time correction from ntp
000+08:46:23.222 I 29: [thermostat] thermostat time correction from ntp
000+08:46:23.769 I 30: [thermostat] thermostat time correction from ntp
000+08:46:24.294 I 31: [thermostat] thermostat time correction from ntp
000+08:46:25.144 I 32: [thermostat] thermostat time correction from ntp
000+08:46:25.943 I 33: [thermostat] thermostat time correction from ntp
000+08:46:27.293 I 34: [thermostat] thermostat time correction from ntp
000+08:46:28.081 I 35: [thermostat] thermostat time correction from ntp
000+08:46:28.868 I 36: [thermostat] thermostat time correction from ntp
000+08:46:29.419 I 37: [thermostat] thermostat time correction from ntp
````

His thermostat is  "Rego 2000/3000", "0x10", "product id": 172

KeesBleijenberg, 2022-04-01:
> Have you blocked emsesp from contacting the internet in your router? If your router is configured as ntp server, then try to use this. Is the dns server set in emsesp? Try ip address for ntp server: `216.239.35.0` (time.google.com).

The router does not block emsp in any way and is not configured as a ntp server.
I think the problem was my settings for the dns. I had as first DNS server 8.8.8.8 and a second DNS server 192.168.2.255. It should have been 192.168.2.254.  (*.*.*.255 is broadcast)
First I tried 216.239.35.0 as NTP server, but I got random network errors.: a red bar saying Network error. In the log I could not find these errors. These network errors sometimes popped up after a reboot of the gateway. 
After I filled in the correct DNS and a reboot everything works fine. So it's okay now. 

I find the error messages not very helpful. When the NTP service cannot find the server the message NTP = inactive  does not tell you where to look for errors. But maybe improving the error messages is difficult.

Thanks for all your help!

proddy, 2022-04-01:
glad it works. BTW on my router I use 8.8.8.8 (google) and 1.1.1.1 (cloudflare) as DNS servers, nothing internal. 

Unfortunately, the NTP calls in espressif don't return any status since it's an asynchronous UDP call.

Can you close this issue?

MichaelDvP, 2022-04-01:
> @MichaelDvP eltoro on [Discord](https://discord.com/channels/816637840644505620/816638587508490250/959343020573065246) is getting the log flooded with these errors after the latest build:
> 
> ```
> 000+08:46:16.769 I 22: [thermostat] thermostat time correction from ntp
> ```
> 
> His thermostat is "Rego 2000/3000", "0x10", "product id": 172

We need to check system info and a debug log. Is it the latest version? 
Seems the datetime command can not be executed. I remember the Rego 2000 is a bit different from RC300, like in #356, where it does not broadcast the monitor telegram. Maybe for this thermostat the datetime is not writable, or in adifferent way. Another possible cause: tx-writes are disabled, this should be checked before automatic clock calibration. I'll add the check.

MichaelDvP, 2022-04-02:
For the rego clock we should check is there are tx_errors (not writable) or the clock does not provide seconds (in a log all). Since the clock is only broadcasted every minute, we can also check if the minute setting is  wrong.

vmonkey, 2022-04-03:
I have the same issue in beta 11 as @proddy - that NTP time shows inactive (I tried, e.g., time.google.com) when url is used. It works with the time.google.com IP address though. The second issue- if the NTP status is inactive but enabled, the web interface shows the "Set time" button, but setting the time via the button results in Error 400.

MichaelDvP, 2022-04-04:
Should we extend the status to `disabled(gray)/inactive(red)/active(green)`?
Allow manual time setting (button) only if ntp disabled or also if inactive?

vmonkey, 2022-04-04:
1) ad my issue 1- I have "Use fixed IP address" enabled in the Network settings and the DNS IP fields were left empty (I believed it would use the router default). After putting a DNS server there, NTP time.google.com works. 

2) ad button - from my perspective, the button makes sense there, or it should at least point the user to how to set the correct time. Maybe, there could be a popup window saying something similar to
"Time set successfully. Because NTP is enabled, it may overwrite this time when available. Would you like to keep NTP enabled?" This would make the most sense if NTP is turned on by default (I do not remember whether it is true), a default (presumably incorrect) time zone is set, and the NTP server is temporarily unavailable.

The second, simpler option would be to just list the status inactive, remove the button, but suggest the user turn off NTP to set the time manually.

MichaelDvP, 2022-05-02:
I think it's already merged. Actual the button for manual time setting is active if NTP is not active (but maybe enabled).
Status also shows Disabled/Inactive/Active.

proddy, 2022-05-02:
yes, think it's merged. I see the button. I'll close this and we can re-open if needed later

commit 786a94b4 (2022-03-29): try to fix #408 and #412
commit d269029b (2022-03-29): Merge pull request #424 from MichaelDvP/dev

try to fix #408 and #412
closed 2022-05-02 by proddy