emsesp/EMS-ESP32 #410: EMS-ESP Version v3.4.0b8: Error with Zigbee2MQTT

zibous, 2022-03-26:
**Bug description**

I have a problem with **Zigbee2MQTT** and I don't know whether it is due to the EMS-ESP32, Zigbee2MQTT or Homeassistant:

**debug message Zigbee2MQTT:**
```
Mar 26 03:03:03 **** node[814]: 
 #033[34mZigbee2MQTT:debug#033[39m 2022-03-26 03:03:03: 
 Received MQTT message on 'homeassistant/sensor/ems-heizung/boiler_lastcode/config' 
 with data '{"~":"ems-heizung","uniq_id":"boiler_lastcode",
 "stat_t":"~/boiler_data","name":"Boiler last error code",
 "val_tpl":"{{value_json.lastcode}}","dev":{"ids":["ems-esp-boiler"]}}'
```

**Steps to reproduce**
```bash
cat /var/log/syslog | grep "Received MQTT message on 'homeassistant/sensor/ems-heizung/"
```

**Expected behavior**
If this message occurs, then the processing of aioesphomeapi is aborted.

proddy, 2022-03-26:
not sure I understand the error. EMS-ESP publishes that MQTT message to the broker and it crashes ESPhome?

zibous, 2022-03-26:
@proddy 

Mea culpa, 
It's on the Zigbee2Mqtt. **All HA discovery items messages** are always displayed there in the **Zigbee2Mqtt debug log** !
It must be a coincidence that this has to do with my error in the `aioesphomeapi`.

Thanks for the help and the excellent work for the EMS-ESP32 application.