emsesp/EMS-ESP32 #418: rename entitiy descriptions with total

proddy, 2022-03-28:
From Harvey:

by the way (minor): the word "total" is easier to recognize at a fixed position, eg. always in front. Sometimes it is at the end like in "auxiliary electrical heater energy consumption total".
so the (long) description is inconsistent to "total energy supplied cooling" which is nice in my opinion.
`< total | nothing >< energy supplied | energy consumption >< heating | dhw | cooling | auxilary electrical heater | pool | ... >`
Everything is ok, but a consistent order of keywords make thing even more beautiful (short names are what they are :-)

This is very minor, but possibly enhances future sorting of rows .... just an idee

Thk
Harvey

_Originally posted by @harvey637 in https://github.com/emsesp/EMS-ESP32/issues/411#issuecomment-1080392526_

harvey637, 2022-03-28:
Hi,
latest b10 looks very promising, sorting ok, getting all MY selected entities.
Thank You very much!

two very minor suggestions:
- put the heart as marker in the dashboard to the left side of the name, looks alinged and not depending on name length
- this entity seems to be related to warmwater (dhe) pls say so: "control starts (wwstarts2)" -> "dhw control starts2" not to be confused with "dhw starts (wwstarts), maybe this is also "dhw control starts (wwwstarts)" ?????

cu
Harvey

proddy, 2022-03-28:
putting the little red heart on the left of the name won't line up the entity names and will look off, no?

harvey637, 2022-03-29:
Hi,
I am happy as it is now, green heart is better than purple heart :-), understanding that using a different alternating background is too much work in CSS.
cu
Harvey

proddy, 2022-03-29:
the alternating background is easy, just one line to add. I just think it makes the table look messy because of the banded grey rows. I'll send a screenshot.

closed 2022-03-30 by proddy