emsesp/EMS-ESP32 #436: make shower timer settings configurable

lbbrhzn, 2022-04-02:
**Is your feature request related to a problem? Please describe.**
The settings for the shower timer are currently hardcoded. The max shower duration is fixed to 7 minutes.

**Describe the solution you'd like**
Make the max show duration configurable through settings. Change the unit to seconds instead of ms.

| Setting | Description|
| ------- | ------------|
| SHOWER_PAUSE_TIME                   | 15 seconds, max time if water is switched off & on during a shower |
| SHOWER_MIN_DURATION              | 2 minutes, before recognizing its a shower |
| SHOWER_OFFSET_TIME                  | 5 seconds grace time, to calibrate actual time under the shower |
| SHOWER_COLDSHOT_DURATION  | 10 seconds for cold water before turning back hot water |
| SHOWER_MAX_DURATION             |  7 minutes, before trigger a shot of cold water |

**Describe alternatives you've considered**
Forking the repo and changing the hardcoded constants.

proddy, 2022-04-02:
ok, wasn't sure if anyone actually used that feature! Do you just want the max duration (7 mins) to be configured?

lbbrhzn, 2022-04-02:
yes!

proddy, 2022-05-01:
added in b15

closed 2022-05-01 by proddy