emsesp/EMS-ESP32 #437: Junkers ISM 2 Module not recognized

knopfsocke, 2022-04-02:
**Is your feature request related to a problem? Please describe.**
EMS-ESP32 discoverd two unknow devices. One of them has to be the ISM 2 I use to control the components of my solar system together with the FW 200 controller.

The ISM 2 module is connected as follows:

R1 = solar circuit pump for 1st collector array (SP)
R6 = valve return increase (DWU1)
NTC1 = collector temperature sensor 1st field (T1)
NTC2 = Bottom storage tank temperature sensor on the heating water side (solar storage tank T2)
NTC3 = center heating water-side cylinder temperature sensor (solar cylinder T3)
NTC6 = heating network return temperature sensor (T4)

>   "Devices": [
    {
      "type": "Boiler",
      "name": "Junkers GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.02)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.14)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.08)",
      "handlers": "0x10C 0x11E"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.08)",
      "handlers": "0x10C 0x11E"
    },
    {
      "type": "Controller",
      "name": "Controller (DeviceID:0x09, ProductID:152, Version:02.08)"
    },
    {
      "type": "Unknown",
      "name": "Junkers unknown (DeviceID:0x30 ProductID:103, Version:23.04)"
    },
    {
      "type": "Unknown",
      "name": "Junkers unknown (DeviceID:0x41 ProductID:100, Version:20.08)"
    }
  ]
}

MichaelDvP, 2022-04-02:
Please use `log level all` in system->system log and wait more than a minute, then download the log and attach it here (drag&drop). 
If ISM2 is mostly compatible with ISM1 it's a simple add of product id 103 to database. But i think there are some extras in ISM2.
For Device-ID 0x41 i have no idea what it doing. Norbers list says powermodule for heatingcircuit hc10, but we have a 0x40 as rf temperature sensor, maybe something like that?

knopfsocke, 2022-04-02:
[log.txt](https://github.com/emsesp/EMS-ESP32/files/8402886/log.txt)

MichaelDvP, 2022-04-02:
Thanks, It s telegram 0x103 like ISM1 and in addition telegram 0x104. 0x104 has two temperatures (offset 0 and 2) and the valve state (offset 16/17).

The 0x41module is something with warm water, sending telegram 0x34 with setpoint (55°C) and measured temperature (54,9°C): `Unknown(0x41) -> All(0x00), UBAMonitorWW(0x34), data: 37 02 25 02 25 00 00 00 00` 
and sometimes sending commands in 0x1E to the boiler `Unknown(0x41) -> Boiler(0x08), ?(0x1E), data: 02 1D` 
The boiler broadcasts only that it has no warm water system: `Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 00 (offset 8)` 
@proddy Should we add a own class or register as generic type module? For a first test i think i'll add it to generic first and push to my github. I do not see a setting telegram for the warm water.

@knopfsocke Please check [here](https://github.com/MichaelDvP/EMS-ESP32/releases), after update you'll need to refresh page in browser. Please post [system info](http://ems-esp.local/api/system) and [http://ems-esp.local/api/solar](http://ems-esp.local/api/solar) and [http://ems-esp.local/api/generic](http://ems-esp.local/api/generic)

MichaelDvP, 2022-04-02:
Hmm, then better add the 0x41 to solar `ISM2-Dhw`? Can you make a log while changing the dhw-setpoint on the thermostat to fetch the command for this module?

knopfsocke, 2022-04-02:
Thanks, you're fast :+1: 

I tested with [Development Build v3.4.0b10](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest)

**Requested API Results**
[system.txt](https://github.com/emsesp/EMS-ESP32/files/8403154/system.txt)
[solar.txt](https://github.com/emsesp/EMS-ESP32/files/8403155/solar.txt)
[generic.txt](https://github.com/emsesp/EMS-ESP32/files/8403153/generic.txt)

**Screenshots with more visual results**
[image]
[image]

The solar values looking very good :1st_place_medal: 

**my observations**
In the Screenshot you can see that the temperature value is a little different from the current external/internal temperature (54,7°C) and the requested set temperature (55°C). I activated the system log again and changed in the FW200 the warm water program settings the target temperature from 55°C to 59°C, that changed the DHW set temperature in the ems-esp ui. The external/internal temperature goes very slowly down. At a certain point, the heater and pump switches on to charge the hot water storage tank, and then the temperature rises again. Evereything is hopefully in the log file: [log2.txt](https://github.com/emsesp/EMS-ESP32/files/8403221/log2.txt)

I found another little thing in the new ui settings, i will create an extra github issue for that.

proddy, 2022-04-02:
> @proddy Should we add a own class or register as generic type module? For a first test i think i'll add it to generic first and push to my github. I do not see a setting telegram for the warm water

@MichaelDvP and new class would be nice, but agree test under generic first

MichaelDvP, 2022-04-03:
I've added the missing ISM2 values. 
We have SM100 dhw in solar class, because the SM100 does not have a own product-id for the dhw, only uses a different device-id. So i moved the ISM-dhw also to solar class, consitent with the SM100 dhw. In the log i've found the Seltemp-setting in telegram 0x35 (like boilers) and added the setting command.
@knopfsocke please update and check again (same link as before).

knopfsocke, 2022-04-03:
Thank you @MichaelDvP , here are the new test results:

**API Results**
[system.txt](https://github.com/emsesp/EMS-ESP32/files/8404291/system.txt)
[solar.txt](https://github.com/emsesp/EMS-ESP32/files/8404292/solar.txt)
[generic.txt](https://github.com/emsesp/EMS-ESP32/files/8404290/generic.txt)

**Log Files**

1. I tried to change the temperature from 55°C to 57°C in ems-esp web UI. Doesn't work: [log-setDHWTemperature-UI.txt](https://github.com/emsesp/EMS-ESP32/files/8404294/log-setDHWTemperature-UI.txt)
2. I sucessfully changed the temperature again with FW200: [log-setDHWTemperature-FW200.txt](https://github.com/emsesp/EMS-ESP32/files/8404293/log-setDHWTemperature-FW200.txt)

**UI Screenshot**
[image]

knopfsocke, 2022-04-03:
**Update:** With the latest build from @MichaelDvP, the MQTT Solar content changed. For example I don't get values for collectorTemp, tankBottomTemp, solarPump, ...

MichaelDvP, 2022-04-03:
Oh, yes, the solar mqtt need a change to support multiple modules (like mixers). I'll update..
The change from webui i can not find in the log, there is nothing sent.

knopfsocke, 2022-04-03:
> The change from webui i can not find in the log, there is nothing sent.
Strange, I can try again. If you plan another update, i will wait for that.

knopfsocke, 2022-04-03:
I found the problem, maybe :-)

In the settings the value is marked as read-only, but the pencil symbol in the dashboard is still visible. I can enter a new value, but no error message appeared in the UI. I suggest to gray out the symbol in the dashboard if the attribute is marked as read-only.

[image]
[image]

proddy, 2022-04-03:
if an entity is forced to be read-only in the Customization screen then there will be no light blue Pencil icon next to the entity name in the Dashboard. If you're seeing something different then we have a bug somewhere

knopfsocke, 2022-04-03:
no bug, sorry. it was my fault, i missinterpreted the meaning of the icons.

MichaelDvP, 2022-04-03:
I think the icons in the advice should be colered as selected:
[image]

@knopfsocke Please try again to log and change settemp via ui. The new temperature will be overwritten by thermostat and will not be permanent, but in the log we should see a change. (btw. the mqtt should work now).

proddy, 2022-04-03:
> I think the icons in the advice should be colered as selected: [image]

yes, much better. you change it or shall I ?

knopfsocke, 2022-04-03:
**Test with Dev 3.4 b11**

API Files:

- [generic_b11.txt](https://github.com/emsesp/EMS-ESP32/files/8405137/generic_b11.txt)
- [solar_b11.txt](https://github.com/emsesp/EMS-ESP32/files/8405138/solar_b11.txt)
- [system_b11.txt](https://github.com/emsesp/EMS-ESP32/files/8405150/system_b11.txt)

Log File while setting dhw temperature via ems-esp UI from 55°C to 57°C:

- [log_b11.txt](https://github.com/emsesp/EMS-ESP32/files/8405140/log_b11.txt)

`2022-04-03 19:28:56.744 D 224: [command] Calling command 'solar/wwseltemp' (selected temperature) with value 57`
`2022-04-03 19:28:56.744 D 225: [emsesp] Write command successful`

Despite positive feedback in the UI and Log, the new value was not accepted:
[image]

knopfsocke, 2022-04-03:
[image]

MichaelDvP, 2022-04-04:
That's strange, after the `[emsesp] Write command successful` the setting on the bus. `Me(0x0B) -> Solar(0x41).. `  is not logged. Could you please try again with actual build and also send the settings (button on help page). 
I've tested with my SM10 and added the command (useless for my module, but i can see the log).

knopfsocke, 2022-04-04:
[emsesp_settings.txt](https://github.com/emsesp/EMS-ESP32/files/8411642/emsesp_settings.txt)
[log.txt](https://github.com/emsesp/EMS-ESP32/files/8411643/log.txt)

knopfsocke, 2022-04-04:
Another try and new insights: Changing the "maximum cylinder temperature" works fine!! It seams, that I cannot set "dhw selected temperature" only?

knopfsocke, 2022-04-09:
Hello @MichaelDvP , 

first of all I would like to thank you for your help. After one week of observation, the ISM2 (0x30/103) module for the solar system seems to provide all the relevant data. But since I still don't have the upper storage tank temperature and the tank charging pump status, I went down to the basement again and took a closer look at the IPM1 (0x41/100) module.

I opened the case and followed the cables. A pump and two sensors are connected:

- The pump (P1) for charging the water tank is connected to the IPM1. The lower area of the storage tank is charged via solar, the upper area via gas heating.
- The first sensor leads to the upper area of the buffer tank (SF1). Input of hot water, when P1 is activated.
- The second sensor goes to the low loss header/hydraulic switch (VF).

From the manual: To charge the buffer tank after the low loss header/hydraulic switch, you must have code 3 or higher. In my case code 3 is already occupied by another module and the installer has set the coding switch of the IPM1 to 10 (heating circuit 10).

I tried to paint a quick sketch of the setup with only the relevant IPM parts, if I should paint a better one please let me know.
[image]

Hope we can add this special settings to ems-esp32 together? I can test again and provide you the information you need, if that is ok for you. :-)

MichaelDvP, 2022-04-10:
Oh, the 0x41 is a IPM, so we need to set it to mixers, i thought only the ISM2 is missing.  
Need a bit of rework. i'll also check the telegrams from Norbert.

MichaelDvP, 2022-04-10:
Do you know what ID 0x10D (msgID_269) contains?  Norbert outputs only the hexdump, but read no value from [this telegram](https://github.com/norberts1/hometop_HT3/blob/fa62d4af84a3544d3c767be26f1b4fd2f075ce71/HT3/sw/lib/ht_discode.py#L2236).

knopfsocke, 2022-04-10:
Hmm not sure. For IPM heating circuit II, the source of message ID 0x10D, the following compontents are connected:

V_IN = 230V
P1 = Heating circuit pump for HC1 (0x20, not relevant)
P2 = Heating circuit pump for HC2
M2 = heating mixer position motor
MF2 = Mixed circuit flow temperature sensor
BUS IN and BUS OUT = between two modules

[image]

Does that help or any idea, what else i can check?

MichaelDvP, 2022-04-10:
Please do in telnet a `read 20 10D` to get the telegram, i've never seen it and don't know what and how many values it contains. ID 0x20 is mixing a heating circuit, i'm confused that a dhw telegram is send from this address.

I've changed the 0x41device to mixer and added the values to wwc10, please check in my beta.

knopfsocke, 2022-04-10:
### Telnet

```
000+00:04:49.318 I 17: [emsesp] Mixer(0x20) -> Me(0x0B), ?(0x010D), data: <empty>
ems-esp:$ read 41 1E
000+00:05:27.523 I 19: [emsesp] Mixer(0x41) -> Me(0x0B), HydrTemp(0x1E), data: <empty>
ems-esp:$ read 10 35
000+00:05:51.501 I 21: [emsesp] Thermostat(0x10) -> Me(0x0B), UBAFlags(0x35), data: <empty>
ems-esp:$ read 10 23
000+00:06:01.583 I 22: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x23), data: <empty>
ems-esp:$ read 21 10D
000+00:06:12.599 I 23: [emsesp] Mixer(0x21) -> Me(0x0B), ?(0x010D), data: <empty>
ems-esp:$ read 20 10D
000+00:06:16.358 I 24: [emsesp] Mixer(0x20) -> Me(0x0B), ?(0x010D), data: <empty>
ems-esp:$ read 20 10C
000+00:06:24.975 I 25: [emsesp] Mixer(0x20) -> Me(0x0B), IPMStatusMessage(0x010C), data: 01 01 00 00 00 25
```

### Log
[log.txt](https://github.com/emsesp/EMS-ESP32/files/8459052/log.txt)

### Mixer Device 0x41
The web interfaces shows the device as a mixer now. MQTT Messages are received. Transition from Solar to Mixer sucessfull.

knopfsocke, 2022-04-10:
Does this help?

```
ems-esp:$ show devices
These EMS devices are currently active:

Boiler: Junkers GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:06.02)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x34 0x2A 
 Fetched telegram type IDs: 0x14 0x16 0x33 0x26 
 Pending telegram type IDs: 0xBF 0xC2 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 
 Ignored telegram type IDs: 

Thermostat: Junkers FW200 (DeviceID:0x10, ProductID:106, Version:12.14)
 Received telegram type IDs: 0x06 
 Fetched telegram type IDs: 0x16F 0x170 0x165 0x166 
 Pending telegram type IDs: 0xA3 0xA2 0x12 0x171 0x172 0x167 0x168 0x123 
 Ignored telegram type IDs: 0x35 0x23 0xBE 0x1D3 

Mixer: Junkers IPM (DeviceID:0x20, ProductID:102, Version:20.08)
 Received telegram type IDs: 0x10C 
 Fetched telegram type IDs: 
 Pending telegram type IDs: 0x11E 
 Ignored telegram type IDs: 0xBE 

Mixer: Junkers IPM (DeviceID:0x21, ProductID:102, Version:20.08)
 Received telegram type IDs: 0x10C 
 Fetched telegram type IDs: 
 Pending telegram type IDs: 0x11E 
 Ignored telegram type IDs: 0xBE 

Mixer: Junkers IPM (DeviceID:0x41, ProductID:100, Version:20.08)
 Received telegram type IDs: 0x34 
 Fetched telegram type IDs: 
 Pending telegram type IDs: 0x1E 
 Ignored telegram type IDs: 0xBE 0x10D 

Solar: Junkers ISM2 (DeviceID:0x30, ProductID:103, Version:23.04)
 Received telegram type IDs: 0x104 
 Fetched telegram type IDs: 0x103 0x101 
 Pending telegram type IDs: 
 Ignored telegram type IDs: 0xBE 

Controller: Controller (DeviceID:0x09, ProductID:152, Version:02.08)
 Received telegram type IDs: 
 Fetched telegram type IDs: 
 Pending telegram type IDs: 0x06 
 Ignored telegram type IDs: 
```

norberts1, 2022-04-10:
@MichaelDvP 
> Do you know what ID 0x10D (msgID_269) contains?

That MsgID:269 contains one Byte with status-informations on (in german language):

 MsgID 269_0_0; Byte6; Bit0: 'Status für den Temperaturfühler (NTC) im Warmwasserspeicher am Schaltmodul'
 MsgID 269_0_1; Byte6; Bit1: 'Status für den Thermostat im Warmwasserspeicher am Schaltmodul'

Perhaps only 'closed' and 'open' or 1 and 0 values are possible as informations.
Keep in mind and as you already knows, my bytes-numbering starts with '0' at the first byte of the message.

@MichaelDvP 
Regarding #459 (MsgID: 187) if have some more information in my database. I'll update my document a.s.a.p. if requested.

MichaelDvP, 2022-04-10:
@norberts1 Thanks for the info, that's not really much information in this telegram. If the NTC is broken we will see it on the measurement, i'm not sure if it makes sense to read and show this value.

@knopfsocke most of the telegrams you requested are empty. 0x23 is send from thermostat to mixer to set the required flowtemp, why should we read this value from thermostat, it is alread shown in mixer message 0x10C and shown as flowsettemp.
What value do you miss? I dont want to add register every empty or redundand telegram, that will blow the esp memory.
Could you please post the output of `show` command on telnet?

knopfsocke, 2022-04-10:
```
ems-esp:$ show
EMS-ESP version 3.4.0b11

Boiler: Junkers GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:06.02) (53)
  heating active: 0
  tapwater active: 0
  selected flow temperature: 36.0 °C
  burner selected max power: 100 %
  heating pump modulation: 0 %
  outside temperature: 9.8 °C
  current flow temperature: 41.2 °C
  mixing switch temperature: 43.5 °C
  gas: 0
  gas stage 2: 0
  flame current: 0.0 uA
  heating pump: 0
  fan: 0
  ignition: 0
  oil preheating: 0
  heating activated: 1
  heating temperature: 70.0 °C
  burner pump max power: 100 %
  burner pump min power: 10 %
  pump delay: 3
  burner min period: 10
  burner min power: 0 %
  burner max power: 100 %
  hysteresis on temperature: -6.0 °C
  hysteresis off temperature: 6.0 °C
  set flow temperature: 33.0 °C
  burner current power: 0 %
  burner starts: 59903
  total burner operating time: 477 days 12 hours 10 minutes
  burner stage 2 operating time: 0 days 0 hours 0 minutes
  total heat operating time: 477 days 12 hours 10 minutes
  total UBA operating time: 2386 days 1 hours 4 minutes
  last error code: EA(229) 18.03.2022 11:18 (0 min)
  service code number: 204
  maintenance message: H00
  maintenance scheduled: 0
  time to next maintenance: 6000 hours
  next maintenance date: 01.01.2009
  dhw selected temperature: 60.0 °C
  dhw type: 0
  dhw comfort: 0
  dhw flow temperature offset: 40.0 °C
  dhw max power: 100 %
  dhw circulation pump available: 0
  dhw charging type: 1
  dhw hysteresis on temperature: -5.0 °C
  dhw hysteresis off temperature: 0.0 °C
  dhw disinfection temperature: 70.0 °C
  dhw circulation pump mode: 2
  dhw circulation active: 0
  dhw activated: 1
  dhw 3way valve active: 0
  dhw set pump power: 0 %

Thermostat: Junkers FW200 (DeviceID:0x10, ProductID:106, Version:12.14) (19)
  date/time: 10.04.2022 18:21
  hc1 selected room temperature: 21.0 °C
  hc1 current room temperature: 27.7 °C
  hc1 mode: 3
  hc1 mode type: 2
  hc1 heat temperature: 21.0 °C
  hc1 eco temperature: 15.0 °C
  hc1 nofrost temperature: 5.0 °C
  hc1 control device: 0
  hc1 program: 0
  hc2 selected room temperature: 21.0 °C
  hc2 current room temperature: 27.7 °C
  hc2 mode: 3
  hc2 mode type: 2
  hc2 heat temperature: 21.0 °C
  hc2 eco temperature: 15.0 °C
  hc2 nofrost temperature: 5.0 °C
  hc2 control device: 0
  hc2 program: 0

Mixer: Junkers IPM (DeviceID:0x20, ProductID:102, Version:20.08) (2)
  hc1 setpoint flow temperature: 36.0 °C
  hc1 pump status (PC1): 1

Mixer: Junkers IPM (DeviceID:0x21, ProductID:102, Version:20.08) (4)
  hc2 flow temperature (TC1): 27.3 °C
  hc2 mixing valve actuator (VC1): 76 %
  hc2 setpoint flow temperature: 28.0 °C
  hc2 pump status (PC1): 1

Mixer: Junkers IPM (DeviceID:0x41, ProductID:100, Version:20.08) (3)
  wwc10 selected temperature: 55.0 °C
  wwc10 current intern temperature: 53.5 °C
  wwc10 current extern temperature: 53.5 °C

Solar: Junkers ISM2 (DeviceID:0x30, ProductID:103, Version:23.04) (11)
  collector temperature (TS1): 44.7 °C
  cylinder bottom temperature (TS2): 45.2 °C
  pump (PS1): 0
  pump working time: 238 days 2 hours 0 minutes
  maximum cylinder temperature: 70.0 °C
  collector shutdown: 0
  cyl heated: 0
  cylinder middle temperature (TS3): 46.7 °C
  return temperature heat assistance (TS4): 29.6 °C
  heat assistance valve (M1): 1
  energy last hour: 504.0 Wh

Controller: Controller (DeviceID:0x09, ProductID:152, Version:02.08) (0)

ems-esp:$ 
```

Sorry, @MichaelDvP I didn't know that empty (useless) telegrams are sent on the bus. You added the missing values from my ISM2, so I'm happy right now.

MichaelDvP, 2022-04-10:
No problem, It was unclear if there are values from these telegrams are missing. In the latest build i test to read 0x23 in thermostat and show as targetflowtemp, but it's the same as the mixers setpoint flow temperature. 
(btw. the thermostat calculate the flowtemp and sends it in 0x23 to the mixer, the mixer publishes that value in 0x10C and sends the same value in 0x1A to the boiler, And the boiler publishe the same value in 0x18 or 0xE4, depending on type. You see we have a lot of redundand information. If you have more mixers, the boiler sets the highest value.

knopfsocke, 2022-04-11:
Thanks very much! This helps me to understand the system better and with the telnet command I have also learned something new and can take a better look myself in the future before asking questions.

I have one last question to this point: 
- I still can't find the pump status P1 (on/off) which is charging my water heater.
  - When the boiler flame goes out and hot water production is finished (boilder_data:heatingactive), pump P1 continues to run for approx. 3 minutes.
- Who controls the pump, the mixer or the thermostat? Are the mixers always just sensors/actuators and all the intelligence is in the thermostat?

Thanks in advance

MichaelDvP, 2022-04-11:
>  still can't find the IPM1 pump status P1 (on/off) which is charging my water storage cyclinder. 

I think this is a  bit in offset 5 of 0x34 telegram. Boilers have some options in this byte: bit0-ww preparation, bit1-onetimechanrge, bit3-charging, bit6-ww operation, bit7 -ww priority. I've set in my build the bit0 as pump, please check, if it does not work you have to log while pump is running.

> Who controls the pump, the mixer or the thermostat? Are the mixers always just sensors/actuators and all the intelligence is in the thermostat?

Only bosch/junkers knows, it can be different for different types of thermostats/mixers/boilers. But mostly it is the mixer that controls. There should be a setting telegram to change the control parameters of the mixer, like hystereses on/off, etc. Try with a `read 41 33` this is the settings telegram used in boilers, maybe it's the same.

knopfsocke, 2022-04-11:
> I think this is a bit in offset 5 of 0x34 telegram. Boilers have some options in this byte: bit0-ww preparation, bit1-onetimechanrge, bit3-charging, bit6-ww operation, bit7 -ww priority. I've set in my build the bit0 as pump, please check, if it does not work you have to log while pump is running.

It's not bit0 it's bit3-charging:
```
ems-esp:$ read 41 34
000+00:10:09.020 I 11: [emsesp] Mixer(0x41) -> Me(0x0B), UBAMonitorWW(0x34), data: 37 02 04 02 04 00 00 00 00
ems-esp:$ read 41 34
000+00:15:17.957 I 12: [emsesp] Mixer(0x41) -> Me(0x0B), UBAMonitorWW(0x34), data: 37 01 FE 01 FE 08 00 00 00
ems-esp:$ read 41 34
000+00:28:02.519 I 26: [emsesp] Mixer(0x41) -> Me(0x0B), UBAMonitorWW(0x34), data: 37 02 28 02 28 00 00 00 00
```

You're right, there is a settings telegram:
```
ems-esp:$ read 41 33
000+00:27:56.255 I 25: [emsesp] Mixer(0x41) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 46 FB FF 28 FF 07 46 00 FF 00
```

:+1:  Your ems-esp software tools are great! It's possible to analyze quickly and precisely if you know how to do it, and I know it now :-)

knopfsocke, 2022-04-12:
thanks. the parameters are working. i will check the pump signal, but it may take a while because the sun has loaded the memory too much. I will probably still have to take a bath for the test.

Another question of understanding. For me, the boiler **setflowtemp** toggles for 2-3 values between 0 and xx (e.g. toggle betwenn 0 and 31°C --> 0-0-0-31-31-0-0-0-31-31-31-31-0-0-0). This is certainly not a good value for a control parameter.

The **selFlowTemp** of the boiler and the **setFlowTemp* request of the mixer, on the other hand, have an even progression. I don't know why the boiler toggles between values that I can't derive from the mixer and thermostat requests. Does that make sense or do i simply have to ignore it? any idea?

knopfsocke, 2022-04-13:
> I've changed the pump bit and added the parameters, please check.

Pump is working. Thanks!

MichaelDvP, 2022-04-14:
The setflowtemp is a value the boiler gets as command 0x1A, your log shows, that mixer 0x21 requests a positive flowtemp, and mixer 0x41 sends the same telegram to boiler with 0. The boiler will pick the highest value to heat up, this is reported in Boiler-Monitor-telegram as selflowtemp.  You can set the setflowtemp to ignore.

knopfsocke, 2022-04-14:
thank you very much michel after prody adopted the changes in his dev branch, i think we can close this post now. I am very grateful to you both.

closed 2022-04-15 by MichaelDvP