emsesp/EMS-ESP32 #438: Ethernet Phy Settings for LAN8710 (e.g. Olimex ESP32-GATEWAY)

knopfsocke, 2022-04-02:
**Is your feature request related to a problem? Please describe.**

By adding with v3.4dev the Olimex ESP32-POE-ISO to board profiles plus settings to customize Ethernet modules https://github.com/emsesp/EMS-ESP32/issues/301 it is now possible to set the eth phy settings to custom values :+1: 

I have an Olimex ESP32-GATEWAY with LAN87**1**0. The problem in the dev build i see, is that I have to select LAN8720 or TLK110 before i can change the custom values for the LAN8710.

*Dev Build v3.4 0b10 Settings LAN DropDown*
[image]

**Describe the solution you'd like**
Add a custom PHY to the Dropdown to activate the custom phy settings.

**Describe alternatives you've considered**
Add the Olimex ESP32-GATEWAY https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware to the dropdown list with hard coded phy settings.

**Additional context**
Here the LAN8710 parameters i found and currently working for me:
> phy_addr   = 0;
power      = 5;
mdc        = 23;
mdio       = 18;
type       = ETH_PHY_LAN8720;
clock_mode = ETH_CLOCK_GPIO17_OUT;

*Dev Build v3.4 0b10 Settings LAN8710*
[image]

I tested with [Development Build v3.4.0b10](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest) Pre-release!

proddy, 2022-04-02:
I'm not sure I follow. What's wrong with selecting Custom, then choosing LAN8720 (which is the chipset of the Olimex ESP32-GATEWAY) and then choosing the correct power, address and clock mode?

knopfsocke, 2022-04-03:
I'm also not sure if maybe I'm wrong. There is no doubt about it, the devil is in the detail. 

All 4 Olimex ESP32 Ethernet Products have a LAN87**1**0 and **not** a LAN87**2**0:
| Product | PHY | Details |
| ------------- | ------------- | -------- |
| Olimex ESP32-GATEWAY  | LAN8710A-EZC(QFN32) | [ESP32-GATEWAY Rev_G Schematic](https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/HARDWARE/Hardware%20revision%20G/ESP32-GATEWAY_Rev_G.pdf) |
| Olimex ESP32-POE-ISO  | LAN8710A-EZC(QFN32) | [ESP32-POE-ISO Rev_H Schematic](https://github.com/OLIMEX/ESP32-POE-ISO/blob/master/HARDWARE/ESP32-PoE-ISO-Rev.H/ESP32-PoE-ISO_Rev_H.pdf) |
| Olimex ESP32-POE  | LAN8710A-EZC(QFN32) | ... |
| Olimex ESP32-EVB  | LAN8710A-EZC(QFN32) | ... |

A short look into the datasheet i see no major differences:
| | LAN8710A/Ai | LAN8720A/Ai |
| ----------- | ----------- | ------------- |
| **Pins** | 32 | 24 |
| **Protocol** | RMII, MII | RMII |
| **driver/receiver** | 2x2 | 4x4 |

[Datasheet LAN8710A/Ai](https://ww1.microchip.com/downloads/en/DeviceDoc/00002165B.pdf)
[Datasheet LAN8720A/Ai](https://ww1.microchip.com/downloads/en/DeviceDoc/00002164B.pdf)

proddy, 2022-04-03:
ah, I didn't catch the 8720 vs 8710.

but does it work with LAN8720 then?

Looking at Olimex's code examples I see they still use the 8720 drivers (https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/SOFTWARE/Newest-board-revisions/ETH_PHY_demo/main/ethernet_example_main.c)

knopfsocke, 2022-04-03:
:-) It's is only working, if i change the default settings for LAN8720 in ems-esp ui settings to the values in the LAN8710 column:
| | LAN8720 |  LAN8710 |
|--------|------|--------|
| **phy_addr** | 1 |**0**|
| **power** | 16 | **5**|
| **mdc** | 23 | 23|
| **mdio** | 18 | 18|
| **clock_mode** | ETH_CLOCK_GPIO0_IN | **ETH_CLOCK_GPIO17_OUT** |

knopfsocke, 2022-04-03:
[image]

knopfsocke, 2022-04-03:
For me, it looks like that the driver is compatible, but the connection between the ESP32 and the LAN chip is different between the 4 Olimex boards, so address, power and clock mode needs to be adapted.

proddy, 2022-04-03:
thats why I added the custom mode, to set the ethernet address, power and clock. So it works as intended, right?

bbqkees, 2022-04-04:
The thing is even with a compatible 8710 or 8720 Ethernet chips, there are a many possible pin configurations possible.
So with custom mode you can then select the right pins for your board.

knopfsocke, 2022-04-04:
> thats why I added the custom mode, to set the ethernet address, power and clock. So it works as intended, right?

Yes it works without any problem since a few days. Do you think, it makes sense to rename the dropdown into "LAN8710/LAN8720"? Nethertheless, I learned now, that both are compatible, so from my side, we can close this issue if this is also ok for you?

closed 2022-04-10 by proddy