emsesp/EMS-ESP32 #439: 3.4 b10 not reading all entities from Heat Pump anymore

mvjt, 2022-04-03:
**Bug description**
Later/Latest version is not consistent in generating entities. One day I have 95/40, another day 75/25 etc. Right now, 5 minutes after reboot, I´m missing a lot of entities. I've touched the HP display to try to make it force send, but this does not help. I'm pretty sure this worked much better in previous versions. I have 100% quality in the EMS connection.

**Steps to reproduce**
Don't know. In general, after reboot, different amount of entities can be detected. This morning, before reboot I had > 95 entities for the boiler and 40 for the thermostat. 5-10 min after reboot I only have 75/25.

**Expected behavior**
Read same amount of entities every time.

**Screenshots**

**Device information**
`{
  "System": {
    "version": "3.4.0b10",
    "uptime": "000+00:05:18.427",
    "uptime (seconds)": 318,
    "freemem": 115,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "8C:CE:4E:95:50:97",
    "IPv4 address": "10.10.10.121/255.255.255.0",
    "IPv4 gateway": "10.10.10.1",
    "IPv4 nameserver": "10.10.10.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 2555,
    "bus reads (tx)": 438,
    "bus writes (tx)": 409,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 842,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 6,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx",
      "device id": "0x08",
      "product id": 172,
      "version": "02.03",
      "entities": 75,
      "handlers received": "0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x048D",
      "handlers pending": "0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "Rego 2000/3000",
      "device id": "0x10",
      "product id": 172,
      "version": "02.03",
      "entities": 21,
      "handlers received": "0x06",
      "handlers fetched": "0x02A5 0x02B9 0x02A6 0x02BA",
      "handlers pending": "0xA3 0xA2 0x12 0x02AF 0x029B 0x02B0 0x029C 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0
    },
    {
      "type": "Controller",
      "name": "Rego 3000",
      "device id": "0x09",
      "product id": 240,
      "version": "38.03",
      "entities": 0
    }
  ]
}`

**Additional context**
**Boiler**
`{
  "heatingactive": "off",
  "tapwateractive": "off",
  "selflowtemp": 32,
  "selburnpow": 40,
  "heatingpump2mod": 67,
  "outdoortemp": 0.2,
  "curflowtemp": 30.6,
  "rettemp": 48.3,
  "syspress": 0,
  "burngas": "off",
  "flamecurr": 0,
  "heatingpump": "off",
  "fanwork": "off",
  "ignwork": "off",
  "curburnpow": 67,
  "servicecodenumber": 0,
  "uptimecontrol": 136764,
  "uptimecompheating": 124964,
  "uptimecompcooling": 0,
  "uptimecompww": 11799,
  "uptimecomppool": 0,
  "totalcompstarts": 969,
  "heatingstarts": 757,
  "coolingstarts": 0,
  "wwstarts2": 212,
  "poolstarts": 0,
  "nrgconstotal": 1525,
  "nrgconscomptotal": 1521,
  "nrgconscompheating": 1264,
  "nrgconscompww": 257,
  "nrgconscompcooling": 0,
  "nrgconscomppool": 0,
  "auxelecheatnrgconstotal": 4,
  "auxelecheatnrgconsheating": 0,
  "auxelecheatnrgconsww": 4,
  "auxelecheatnrgconspool": 0,
  "nrgsupptotal": 6913,
  "nrgsuppheating": 6083,
  "nrgsuppww": 830,
  "nrgsuppcooling": 0,
  "nrgsupppool": 0,
  "hppower": 3.2,
  "hpcompon": "on",
  "hpactivity": "hot water",
  "hpheatingon": "off",
  "hpcoolingon": "off",
  "hpwwon": "on",
  "hppoolon": "off",
  "hpbrinepumpspd": 53,
  "hpswitchvalve": "off",
  "hpcompspd": 67,
  "hpcircspd": 10,
  "hpbrinein": 6.9,
  "hpbrineout": 4.4,
  "hpsuctiongas": 8.9,
  "hphotgas": 72.5,
  "hptc0": 48.3,
  "hptc1": 52.2,
  "hptc3": 50.1,
  "hptr3": 48,
  "hptl2": 0,
  "hppl1": -2.2,
  "hpph1": 52.8,
  "wwsettemp": 58,
  "wwcirc": "on",
  "wwcurtemp": 50.5,
  "wwcurtemp2": 50,
  "wwonetime": "off",
  "wwdisinfecting": "off",
  "wwcharging": "on",
  "wwrecharging": "off",
  "wwtempok": "on",
  "ww3wayvalve": "on",
  "wwstarts": 0,
  "wwworkm": 0
}`

**Thermostat**
`{
  "datetime": "08:08:45 03.04.2022",
  "hc1": {
    "seltemp": 21,
    "mode": "manual",
    "modetype": "eco",
    "ecotemp": 15,
    "manualtemp": 21,
    "comforttemp": 21,
    "designtemp": 42,
    "minflowtemp": 22,
    "maxflowtemp": 42,
    "targetflowtemp": 0,
    "summermode": "off",
    "tempautotemp": -1,
    "fastheatup": 0
  },
  "hc2": {
    "seltemp": 21,
    "mode": "manual",
    "modetype": "eco",
    "ecotemp": 15,
    "manualtemp": 21,
    "comforttemp": 21,
    "targetflowtemp": 0,
    "summermode": "off",
    "tempautotemp": -1
  }
}`

mvjt, 2022-04-03:
Just rebooted again. Boiler entities went up to 75, when "Loading" was stuck when I refreshed, and all of a sudden the entitiy incrementation started all over again. And after that, I have 94/21 entities. So it is really inconsistent.

mvjt, 2022-04-03:
And now I have 94/35

`{
  "datetime": "09:03:56 03.04.2022",
  "dampedoutdoortemp": 8,
  "building": "light",
  "minexttemp": -20,
  "wwmode": "low",
  "wwcircmode": "off",
  "wwchargeduration": 60,
  "wwcharge": "off",
  "wwdisinfecting": "off",
  "wwdisinfectday": "so",
  "wwdisinfecttime": 120,
  "hc1": {
    "seltemp": 21,
    "mode": "manual",
    "modetype": "eco",
    "ecotemp": 15,
    "manualtemp": 21,
    "comforttemp": 21,
    "summertemp": 18,
    "targetflowtemp": 0,
    "summersetmode": "auto",
    "summermode": "off",
    "tempautotemp": -1
  },
  "hc2": {
    "seltemp": 21,
    "mode": "manual",
    "modetype": "eco",
    "ecotemp": 15,
    "manualtemp": 21,
    "comforttemp": 21,
    "summertemp": 18,
    "targetflowtemp": 0,
    "summersetmode": "auto",
    "summermode": "off",
    "tempautotemp": -1
  }
}`

proddy, 2022-04-03:
are you getting the information pasted above from the API calls http://ems-esp.local/api/thermostat and http://ems-esp.local/api/boiler ? The first is the boiler, the second is the thermostat.

MichaelDvP, 2022-04-03:
Uptime 318s, most of the handlers pending. People are always so impatient. 
> 2022-04-03 08:51:21.814 E 10: [emsesp] Last Tx write rejected by host

For this we need a detailed log.

mvjt, 2022-04-03:
> are you getting the information pasted above from the API calls http://ems-esp.local/api/thermostat and http://ems-esp.local/api/boiler ? The first is the boiler, the second is the thermostat.

Yes. In original post I highlighted in bold before the payload. The second comment was the thermostat. Both coming from the API calls.

Its been three hours now and I still only have 35/40 from the thermostat.

What type of detailed log do you need?

MichaelDvP, 2022-04-03:
In system info you have Uptime 318 seconds,  bus reads (tx)": 438, "bus writes (tx)": 409. Something is sending more than 1 command and more than one read per second to the bus. The ems-bus normally handles ~3 messages per second, normal communication seems to be blocked by the ems-esp activity.  Check with log all what's happening on the bus.

mvjt, 2022-04-03:
From the log

`2.554 N 349: [emsesp] Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 16 04 0C 03 01 2B 06 03
000+00:08:42.752 N 350: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 1B
000+00:08:42.794 N 351: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 0C 03 01 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:08:42.794 I 352: [thermostat] thermostat time correction from ntp
000+00:08:43.092 N 353: [emsesp] Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 16 04 0C 03 01 2C 06 03
000+00:08:43.377 N 354: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 1B
000+00:08:43.419 N 355: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 0C 03 01 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:08:43.419 I 356: [thermostat] thermostat time correction from ntp
000+00:08:43.717 N 357: [emsesp] Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 16 04 0C 03 01 2C 06 03
000+00:08:44.402 N 358: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 1B
000+00:08:44.444 N 359: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 0C 03 01 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:08:44.444 I 360: [thermostat] thermostat time correction from ntp
000+00:08:44.654 N 361: [emsesp] Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 16 04 0C 03 01 2D 06 03
000+00:08:44.852 N 362: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 1B
000+00:08:44.894 N 363: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 0C 03 01 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:08:44.894 I 364: [thermostat] thermostat time correction from ntp
000+00:08:45.192 N 365: [emsesp] Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 16 04 0C 03 01 2E 06 03
000+00:08:45.402 N 366: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 1B
000+00:08:45.444 N 367: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 0C 03 01 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:08:45.444 I 368: [thermostat] thermostat time correction from ntp
`

proddy, 2022-04-03:
if you disable NTP do you get back all the entities?

mvjt, 2022-04-03:
> if you disable NTP do you get back all the entities?

Previously I disabled both TX and NTP at the same time. Now I re-enabled TX only but kept NTP off, and I get the same flooded log again including It seems I need to disable TX (enable read-only) to not get NTP flooding...

proddy, 2022-04-03:
yes I think so too, I'm confident Michael will know what is going on here

MichaelDvP, 2022-04-03:
The problem is, that this thermostat clock always returns seconds zero and give a difference to esp-clock. We need to check this before setting the time. 
Check NTP status, if it show status inactive, the thermostat is not synct, i tried it here.

mvjt, 2022-04-03:
Seems not the case. See screenshot below with both console and NTP is presented as inactive (inPrivate window)
[image]

mvjt, 2022-04-03:
Seems not true in my case. NTP is disabled but it still tries to sync time

[image]

mvjt, 2022-04-03:
Could there be two issues?:
1. This thermostat does not return what you expect and
2. No check if NTP is enabled or not here?
https://github.com/emsesp/EMS-ESP32/blob/29110e96e53ff8ebb006b44a74aaa19d2bb2dd04/src/devices/thermostat.cpp#L1183

proddy, 2022-04-03:
#2 you're looking at the wrong branch, the ntp check is there: https://github.com/emsesp/EMS-ESP32/blob/0bd122b643f0fdc1e1e32e132d3bee4ad0c2db36/src/devices/thermostat.cpp#L1248

mvjt, 2022-04-03:
It is clear the GW is trying to sync time despite NTP being disabled (also reported disable by api/settings). So there is a discrepancy somewhere. Is the code you linked in the GW dev FW as of yesterday?

mvjt, 2022-04-03:
Seem my version was from Friday evening, and yesterday there was changes around NTP...

proddy, 2022-04-03:
also, just to rule out memory issues (see #318) since you're on Ethernet make sure the WiFi SSID is empty (in settings) and also the Access Point is set to ''when Wifi disconnected"

mvjt, 2022-04-03:
Not possible to clear WiFi SSID...But "When WiFi Disconnected" has always been set.

MichaelDvP, 2022-04-03:
Are the ` [emsesp] Last Tx write rejected by host`  also releated to the time setting? 
Please send a log where i can see the telegrams that are rejected. 
Maybe the rego thermostat ismuch more different from RC300 and needs an extra flag to make time non writeable etc..

mvjt, 2022-04-03:
@MichaelDvP yes they were related. At least I don´t get them when not using NTP. I will try to get logs...

mvjt, 2022-04-03:
One interesting observation is that the date/time reported to EMS is 17:28 right now with correct date. So it is 20 minutes behind. But the clock on the HP display is correct...

MichaelDvP, 2022-04-03:
You can try with [this build](https://github.com/MichaelDvP/EMS-ESP32/releases), i've fixed a bug in time checking for IVT and disabled the automatic clock correction for IVT. 
Please enable ntp and tx. Then try to log all and set from dashboard the datetime to "ntp"
[image]
and to time input (with day-of-week and daylightsaving)
[image]
log a minute longer, the clock normally is send out by thermostat every minute, but your IVT seems to stop broadcasting some telegrams if display sleeps, so we may need to fetch the clock active to get the thermostat time.

mvjt, 2022-04-03:
Thanks @MichaelDvP. I think I've done what you asked for. I changed value to "ntp" and later on a second time to a static time/date with your second syntax and waited a minutes (ish). All recorded with full logging.

Do you want the full trace (massive) or just a specific snippet?

Snippet#1
`
2022-04-03 18:58:49.348 DEBUG 1761: [emsesp] No telegram type handler found for ID 0x294 (src 0x10)
2022-04-03 18:58:49.467 DEBUG 1762: [telegram] Sending read Tx [#194], telegram: 0B 90 06 00 1B
2022-04-03 18:58:49.483 TRACE 1763: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 1B
2022-04-03 18:58:49.524 DEBUG 1764: [emsesp] Last Tx read successful
2022-04-03 18:58:49.524 TRACE 1765: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 12 03 39 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2022-04-03 18:58:49.553 DEBUG 1766: [mqtt] Publishing topic ems-esp/thermostat_data (#615, retain=0, retry=1, size=762, pid=1)
2022-04-03 18:58:49.792 DEBUG 1767: [telegram] Sending read Tx [#195], telegram: 0B 90 06 1B 20
2022-04-03 18:58:49.808 TRACE 1768: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 20 (offset 27)
2022-04-03 18:58:49.849 DEBUG 1769: [emsesp] Last Tx read successful
2022-04-03 18:58:49.849 TRACE 1770: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 12 03 39 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
2022-04-03 18:58:50.028 TRACE 1771: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x0293), data: 01 (offset 3)
2022-04-03 18:58:50.048 TRACE 1772: [emsesp] Thermostat(0x10) -> Gateway(0x48), ?(0x0293), data: <empty> (offset 3)
2022-04-03 18:58:50.048 DEBUG 1773: [emsesp] No telegram type handler found for ID 0x293 (src 0x10)
2022-04-03 18:58:50.068 TRACE 1774: [emsesp] Gateway(0x48) <- Thermostat(0x10), ?(0x0294), data: 01 (offset 3)
`
Snippet#2
`
2022-04-03 19:00:45.751 TRACE 2692: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07
2022-04-03 19:00:45.751 DEBUG 2693: [emsesp] No telegram type handler found for ID 0xF7 (src 0x08)
2022-04-03 19:00:46.031 TRACE 2694: [emsesp] Controller(0x09) -> All(0x00), RCTime(0x06), data: 16 04 12 03 3B 2A 06 01 09 FF 00
2022-04-03 19:00:46.031 DEBUG 2695: [emsesp] No telegram type handler found for ID 0x06 (src 0x09)
2022-04-03 19:00:46.714 TRACE 2696: [emsesp] Boiler(0x08) -> All(0x00), ?(0xF7), data: D1 00 00 07`

MichaelDvP, 2022-04-03:
Please the full log, but drag&drop the txt-file here as attachment.

mvjt, 2022-04-03:
[log-3.txt](https://github.com/emsesp/EMS-ESP32/files/8405111/log-3.txt)

MichaelDvP, 2022-04-03:
Thanks, the thermostat does not broadcast any telegram i can't find `Thermostat(0x10) -> All(0x00)`. So we have to set all telegrams to fetch. Also the thermostat ignores/changes offsets in tx-reads and fills the telgrams with zeros to requested lenght. 
Very strange. Need to think about this.
> I think I've done what you asked for. I changed value to "ntp" and later on a second time to a static time/date with your second syntax and waited a minutes (ish). All recorded with full logging.

Yes, thats what i want from you, thanks .But i can only see one change: 
```
2022-04-03 18:58:52.647 TRACE 1821: [emsesp] Me(0x0B) -> Thermostat(0x10), RCTime(0x06), data: 16 04 12 03 3A 30 06 03
2022-04-03 18:58:52.647 DEBUG 1822: [emsesp] Process setting 0x06 for device 0x10
2022-04-03 18:58:52.654 DEBUG 1823: [emsesp] Last Tx write successful
```
Seems that writing the time is possible. But the thermostat reply is with seconds set to zero and wrong dst.
```
2022-04-03 18:58:52.983 TRACE 1828: [emsesp] Me(0x0B) <- Thermostat(0x10), RCTime(0x06), data: 1B
2022-04-03 18:58:53.024 DEBUG 1829: [emsesp] Last Tx read successful
2022-04-03 18:58:53.024 TRACE 1830: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 96 04 12 03 3A 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

mvjt, 2022-04-03:
OK. Sounds good. Right now I only have 39 thermostat entities. I had 51 before this FW upgrade. Is it most recent code in general?

mvjt, 2022-04-03:
Yes we noticed in our other issue that it seems to only send for a short while after the display has been touched. When I wake up the display:

`
2022-04-03 20:37:46.711 DEBUG 20470: [emsesp] No telegram type handler found for ID 0xF7 (src 0x08)
2022-04-03 20:37:47.066 TRACE 20471: [emsesp] Controller(0x09) -> All(0x00), RCTime(0x06), data: 16 04 14 03 24 2B 06 01 09 FF 00
2022-04-03 20:37:47.066 DEBUG 20472: [emsesp] No telegram type handler found for ID 0x06 (src 0x09)
2022-04-03 20:37:47.965 TRACE 20473: [emsesp] Controller(0x09) <- Thermostat(0x10), ?(0xF9), data: 11 FF 01 B9 0A
2022-04-03 20:37:48.004 TRACE 20474: [emsesp] Thermostat(0x10) -> Controller(0x09), ?(0xF9), data: FF 01 B9 0A 17 00 00 00 0A 00 00 00 2A 00 00 00 3C 00 00 00 2A
2022-04-03 20:37:48.004 DEBUG 20475: [emsesp] No telegram type handler found for ID 0xF9 (src 0x10)
2022-04-03 20:37:48.036 TRACE 20476: [emsesp] Controller(0x09) <- Thermostat(0x10), RC300Summer(0x02AF), data: 0A (offset 4)
2022-04-03 20:37:48.057 TRACE 20477: [emsesp] Thermostat(0x10) -> Controller(0x09), RC300Summer(0x02AF), data: 37 2A 00 00 15 1E 00 00 00 15 (offset 4)
2022-04-03 20:37:48.078 TRACE 20478: [emsesp] Controller(0x09) <- Thermostat(0x10), RC300Curves(0x029B), data: 02 (offset 7)
2022-04-03 20:37:48.098 TRACE 20479: [emsesp] Thermostat(0x10) -> Controller(0x09), RC300Curves(0x029B), data: 2A 37 (offset 7)
`

MichaelDvP, 2022-04-04:
> [emsesp] Controller(0x09) -> All(0x00), RCTime(0x06), data: 16 04 14 03 24 2B 06 01 09 FF 00 

This clock has normal year data, seconds and DST set. Are there more telegrams broadcasted from the contoller? (containing: `Controller(0x09) -> All(0x00)`)
I think i would be helpfull to make full logs of some minutes, one after the display is touched, one later after the thermostats stops broadcasting..

mvjt, 2022-04-04:
Seems RCTime is broadcasted every minute by the controller even without touching the display. I have not touched it since yesterday and it was still sending every minute...The only other thing it is broadcasting is an error message

`Controller(0x09) -> All(0x00), ErrorMessage(0xBF), data: 09 F0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

proddy, 2022-04-09:
@MichaelDvP what do we need to do here?

MichaelDvP, 2022-04-09:
Without knowing settings/info/entities and  without complete logs we can do nothing. 
We don't know which telegrams are not broadcasted in thermostat sleep and which entities are not updated/missing. 
I think you can close this.

closed 2022-04-09 by proddy