emsesp/EMS-ESP32 #440: Cannot expand Boiler nor Thermostat entities on dashboard

mvjt, 2022-04-03:
**Bug description**
Rather frequently, with later versions, the UI Dashboard gets stuck. I cannot expand the boiler or dashboard entities although the number of entities are correct. /api/boiler and /api/thermostat shows the entities, but I cannot expand the dashboard. 

**Steps to reproduce**
Not sure, but restart of the gateway solves the problem. Refresh from clean cache does not solve the problem.

**Expected behavior**
Nodes to be expanded.

**Screenshots**
[image]

**Device information**
`{
  "System": {
    "version": "3.4.0b10",
    "uptime": "000+00:08:56.886",
    "uptime (seconds)": 536,
    "freemem": 113,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "8C:CE:4E:95:50:97",
    "IPv4 address": "10.10.10.121/255.255.255.0",
    "IPv4 gateway": "10.10.10.1",
    "IPv4 nameserver": "10.10.10.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 4264,
    "bus reads (tx)": 714,
    "bus writes (tx)": 668,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 1318,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 4,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx",
      "device id": "0x08",
      "product id": 172,
      "version": "02.03",
      "entities": 94,
      "handlers received": "0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "Rego 2000/3000",
      "device id": "0x10",
      "product id": 172,
      "version": "02.03",
      "entities": 21,
      "handlers received": "0x06",
      "handlers fetched": "0x02A5 0x02B9 0x02A6 0x02BA",
      "handlers pending": "0xA3 0xA2 0x12 0x02AF 0x029B 0x02B0 0x029C 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0
    },
    {
      "type": "Controller",
      "name": "Rego 3000",
      "device id": "0x09",
      "product id": 240,
      "version": "38.03",
      "entities": 0
    }
  ]
}`

**Additional context**
**Boiler**
`{
  "heatingactive": "off",
  "tapwateractive": "off",
  "selflowtemp": 29,
  "selburnpow": 39,
  "heatingpump2mod": 38,
  "outdoortemp": 6,
  "curflowtemp": 30.1,
  "rettemp": 28.6,
  "syspress": 0,
  "burngas": "off",
  "flamecurr": 0,
  "heatingpump": "off",
  "fanwork": "off",
  "ignwork": "off",
  "heatingactivated": "on",
  "heatingtemp": 80,
  "burnminperiod": 0,
  "burnminpower": 0,
  "burnmaxpower": 0,
  "boilhyston": 0,
  "boilhystoff": 0,
  "curburnpow": 38,
  "ubauptime": 186909,
  "lastcode": "A11(1051) 20.03.2022 12:46 - 20.03.2022 12:52",
  "servicecodenumber": 0,
  "uptimecontrol": 136785,
  "uptimecompheating": 124975,
  "uptimecompcooling": 0,
  "uptimecompww": 11810,
  "uptimecomppool": 0,
  "totalcompstarts": 970,
  "heatingstarts": 758,
  "coolingstarts": 0,
  "wwstarts2": 212,
  "poolstarts": 0,
  "nrgconstotal": 1526,
  "nrgconscomptotal": 1522,
  "nrgconscompheating": 1264,
  "nrgconscompww": 258,
  "nrgconscompcooling": 0,
  "nrgconscomppool": 0,
  "auxelecheatnrgconstotal": 4,
  "auxelecheatnrgconsheating": 0,
  "auxelecheatnrgconsww": 4,
  "auxelecheatnrgconspool": 0,
  "nrgsupptotal": 6914,
  "nrgsuppheating": 6083,
  "nrgsuppww": 831,
  "nrgsuppcooling": 0,
  "nrgsupppool": 0,
  "hppower": 2.2,
  "hpcompon": "on",
  "hpactivity": "heating",
  "hpheatingon": "on",
  "hpcoolingon": "off",
  "hpwwon": "off",
  "hppoolon": "off",
  "hpbrinepumpspd": 43,
  "hpswitchvalve": "off",
  "hpcompspd": 38,
  "hpcircspd": 8,
  "hpbrinein": 7.5,
  "hpbrineout": 3.5,
  "hpsuctiongas": 10.4,
  "hphotgas": 57.8,
  "hptc0": 28.6,
  "hptc1": 34.3,
  "hptc3": 33.1,
  "hptr3": 32.2,
  "hptl2": 0,
  "hppl1": 1.5,
  "hpph1": 33.8,
  "poolsettemp": 28,
  "wwsettemp": 58,
  "wwseltemp": 52,
  "wwseltemplow": 50,
  "wwseltempsingle": 65,
  "wwcircpump": "on",
  "wwhyston": 0,
  "wwhystoff": 0,
  "wwdisinfectiontemp": 65,
  "wwcircmode": "continuous",
  "wwcirc": "on",
  "wwcurtemp": 52.3,
  "wwcurtemp2": 51.9,
  "wwactivated": "on",
  "wwonetime": "off",
  "wwdisinfecting": "off",
  "wwcharging": "off",
  "wwrecharging": "off",
  "wwtempok": "off",
  "ww3wayvalve": "off",
  "wwstarts": 0,
  "wwworkm": 0
}`

mvjt, 2022-04-03:
Now 10 minuter later it started to work again without restart. Really weird

`{
  "label": "customizations",
  "Customizations": {
    "sensors": [],
    "analogs": [],
    "masked_entities": []
  }
}`

proddy, 2022-04-03:
We need to determine if it's a web rendering issue or the EMS-ESP core itself. Does this only happen to the boiler or also the thermostat?

The deviceData endpoint is encoded so you''ll never see the data in the browser dev tool. What's better is to look at the Headers and look at the the content-length and connection in the Response Header section. 

Can you post the JSON output of http://ems-esp.local/api/boiler so I can simulate it here.

MichaelDvP, 2022-04-03:
> bus writes (tx)": 668

That's to much, please disable mqtt->publish single values on change->publish to command topic. This looks like a mqtt-broker echo of each publish. This is a lot of load for mqtt and ems-bus. If this is not the cause, check your scripts sending commands to ems-esp.

> "freemem": 113

The eth needs ~40-50k, together with the large number of heatpump entities this can be critical. 
If possible, it is a good test to connect with wifi and select a board profle without ETH-Phy. If this works we have to think about how to split the `generate_values_web` to smaller parts. Also disable unused services like OTA, saving a few kb of heap. @proddy Should we add a mDNS enable switch to network settings, mDNS uses ~9k.

mvjt, 2022-04-03:
> That's to much, please disable mqtt->publish single values on change->publish to command topic. This looks like a mqtt-broker echo of each publish. This is a lot of load for mqtt and ems-bus. If [...]

Not sure if you're asking me but MQTT is basically default. "Publish single value topics on change" and "Publish command output to a 'response' topic" are and have always been disabled.

mvjt, 2022-04-03:
> We need to determine if it's a web rendering issue or the EMS-ESP core itself. Does this only happen to the boiler or also the thermostat?
>
> The deviceData endpoint is encoded so you''ll never see [...]
The JSON of the boiler was already attached to the original comment above. 

It happens to both. When it works, there is clearly a lot more data in the payload including human readable entity names. So it is clear the response seems to be empty.  See below how the payload looks when it works

[image]

mvjt, 2022-04-03:
Probably related to #439

mvjt, 2022-04-06:
No further issues since #439  was resolved.

closed 2022-04-06 by mvjt