emsesp/EMS-ESP32 #441: MQTT reconnecting

vmonkey, 2022-04-03:
**Bug description**
I am running Mosquito in a docker container on RPi4. On that Pi, I have scheduled regular backups, during which the Mosquito container is stopped. After re-starting Mosquito, ems-esp does not reconnect automatically and must be restarted to connect back.

**Expected behavior**
Reconnect

**Device information**
{ "System": { "version": "3.4.0b10", "uptime": "000+01:02:28.822", "uptime (seconds)": 3748, "freemem": 159, "reset reason": "Software reset CPU / Software reset CPU", "temperature sensors": 0 }, "Network": { "connection": "WiFi", "hostname": "ems-esp", "SSID": "hodnekrabic", "BSSID": "D8:07:B6:C8:96:B6", "RSSI": -52, "MAC": "7C:87:CE:49:A3:14", "IPv4 address": "192.168.68.113/255.255.255.0", "IPv4 gateway": "192.168.68.1", "IPv4 nameserver": "0.0.0.0" }, "Status": { "bus status": "connected", "bus protocol": "HT3", "bus telegrams received (rx)": 4132, "bus reads (tx)": 1101, "bus writes (tx)": 0, "bus incomplete telegrams": 0, "bus reads failed": 0, "bus writes failed": 0, "bus rx line quality": 100, "bus tx line quality": 100, "MQTT status": "connected", "MQTT publishes": 1960, "MQTT publish fails": 0, "temperature sensors": 0, "temperature sensor reads": 0, "temperature sensor fails": 0, "analog sensors": 0, "API calls": 0, "API fails": 0 }, "Devices": [ { "type": "Boiler", "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3", "device id": "0x08", "product id": 95, "version": "18.14", "entities": 64, "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34", "handlers fetched": "0x14 0x16 0x33", "handlers pending": "0xC2 0x26 0x2A" }, { "type": "Thermostat", "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II", "device id": "0x10", "product id": 158, "version": "18.05", "entities": 64, "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267", "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02F5 0x031B 0x023A 0x0240", "handlers pending": "0xA3 0x12 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x031E" }, { "type": "Mixer", "name": "MM100", "device id": "0x21", "product id": 160, "version": "24.05", "entities": 4, "handlers received": "0x02D8" }, { "type": "Controller", "name": "HT3", "device id": "0x09", "product id": 95, "version": "18.14", "entities": 0 } ] }

MichaelDvP, 2022-04-03:
If possible try to stop mosquitto before stopping the container. 
If the connection to mqtt broker is broken, ems-esp need 3 times of keep-alive-period to detect the disconnect (defaults to 3 minutes). Also reconnect can take this time if the connection is not closed normally.

vmonkey, 2022-04-03:
Thanks, I will try that. Regarding time - even 6 hours is not sufficient.

MichaelDvP, 2022-04-03:
Some other has reported this before, but i can not reproduce. I've done a lot of tests in #305 with different qos and different methods of connection cut, but the actual version of mqtt always reconnects. But this was with a rpi3b/mosquitto, but not in a docker. Maybe something how docker handles the port forwarding to mosquitto?

proddy, 2022-04-03:
I run mosquitto in a docker container and have spent many hours trying to reproduce this too in the past. But for the first time I  experienced the same issue where EMS-ESP wouldn't not reconnect. This was after I upgraded my Ubuntu server to 22.04/jammy so kinda makes sense. 

It's hard to reproduce and I wouldn't spend any more time trying to find the root cause in my opinion.

vmonkey, 2022-04-03:
@proddy It seems that the issue is reproducible on my  setup (at least it happened in all cases of RPi backing up). I will test that also manually. I would be happy to help the project. Would recording logs be helpful for finding the cause? Is there something else I can do?

proddy, 2022-04-03:
can you build the firmware from source? if so I can add some extra debug statements in when compiling using the EMSESP_DEBUG build flag so you can see when the MQTT connection breaks and how often it tries to reconnect.

vmonkey, 2022-04-03:
@proddy I believe I should be able to build it. Thanks!

This is how I can reproduce the issue manually:
1) docker stop mosquito
2) wait 6 minutes
3) docker start mosquitto

6 minutes is the time the backing-up process takes.

MichaelDvP, 2022-04-03:
Tested without docker:
1. sudo service mosquitto stop -> ems-esp mqtt shows disconnected
2. wait 6 minutes
3. sudo service mosquitto start -> esp-esp connected

proddy, 2022-04-03:
same here. ran mosqutto on win11, killed it, restored after 15mins and it reconnected.

vmonkey, 2022-04-03:
Unfortunately, it does not reconnect to me. I tried again with the latest mosquito (previously it was ~ 2 months old version) and ems-esp beta 11, still the same behavior. I will compile with the EMSESP_DEBUG as suggested and post the output. I totally understand you have higher priority tasks, so in the worst case I will create a script that calls the ems-esp device to reboot/hit the save button in the MQTT part to make it automatically usable.

proddy, 2022-04-04:
to do a custom build, create a `pio_local.ini` file and add the line `my_build_flags = -DEMSESP_DEBUG -DEMSESP_USE_SERIAL -DCORE_DEBUG_LEVEL=3`

it's easier if you just find a spare ESP32 and connect it via the USB to your PC/laptop. You'll need a serial connection to see the logs and you're only testing the MQTT reconnect so don't need the EMS circuit

vmonkey, 2022-04-06:
Thanks! 

In the end I solved the issue by disabling "Set Clean Session" in the ems-esp config.

proddy, 2022-04-06:
I was actually thinking the same thing the other day and had to remind myself what a 'clean session' did. Came across this http://www.steves-internet-guide.com/mqtt-clean-sessions-example. Basically setting Clean Session to false will make a persistent connection.

@MichaelDvP shall we change the default back to FALSE? Also mention this in the Wiki under Troubleshooting/MQTT?

MichaelDvP, 2022-04-07:
> @MichaelDvP shall we change the default back to FALSE?

I haven't realized that the default was changed. Yes, i think the default should be FALSE.

proddy, 2022-04-07:
it's been like that for the last 1.5 years :-) https://github.com/emsesp/EMS-ESP32/commit/0623724d745476f24fcfb8aff3f8e96348dddd87

proddy, 2022-04-08:
default is false now

MichaelDvP, 2022-04-08:
But i still can not reproduce the reconnect issue, in both settings emsesp reconnects if mosquitto is stopped/started, Test with 6 minutes, 30 min, 3 hours. (Test with qos 0).

vmonkey, 2022-04-08:
To be fair - I was not able to use the custom build as I do not have a spare esp32 board and I did not want to mess up the ems-esp device. I am experienced with programming (mostly c++) but, unfortunately, I do not have any experience with hw building. Thus, I cannot provide the debug log of the failing cases; sorry for that :-/. Playing with boards and building devices is indeed something I would love to learn, but it will take me some time. If I come to any findings, I will post them here. 

Setting clean session to false fixed the issue for me; I came to that by surveying reconnect issues people had with MQTT, e.g., https://stackoverflow.com/questions/65289607/libmosquitto-does-not-properly-reconnect. I attach here the setting that I use for MQTT (which should be mostly default), the issue was reproducible with "set clean session" enabled.
[image]

MichaelDvP, 2022-04-08:
BTW: MQTT Discovery always uses the json publishes not the "single values on change". If you like to get the values on change you need to set the interval to 0=automatic and disable the "single values on change", then the json is published on change.

@proddy Should we automatically disable the "single values" option if MQTT Discovery is activated? And/or make sure that Publish-Interval 0 publishes with mqtt-discovery on. (Change [here:](https://github.com/emsesp/EMS-ESP32/blob/8479be5bd4ca13ad699b8ef1db3b6467a2e30ff1/src/emsesp.cpp#L946)  to `if (!Mqtt::publish_single() || Mqtt::ha_enabled()) {`

proddy, 2022-04-08:
yes good idea.

vmonkey, 2022-04-08:
@MichaelDvP thanks for the clarification!

closed 2022-04-08 by proddy