emsesp/EMS-ESP32 #444: 3.4b11 problems

harvey637, 2022-04-04:
Hi,
me again :-)

working parts look nice, e.g. the marking of "no api" in the dashboard enties, saving customisation of thermostat

NOT WORKING:
saving ANY customisation for "boiler" gives error 413

infos: not using MQTT

cu
Harvey

proddy, 2022-04-04:
ok, its probably due to a recent change I made where I send all the entities back to the server, and you have 100 of them. I'll need to redesign a few things

harvey637, 2022-04-04:
Hi,
thank you for looking into this!

You are right, I have 100 entites only for boiler (around 60 of them are for warmwater, which I dont use and want to hide).
cu
Harvey

MichaelDvP, 2022-04-04:
This is again a to small buffer. But proddys last change was only to send always all entities, before it has worked with a few exclusions, but will give the same error if you exclude all. Proddy metioned that he plans to pack the buffer, which is some work to do. As an intermediate solution you can check with a larger buffer. I've made [a test](https://github.com/MichaelDvP/EMS-ESP32/releases) with extra debug message to check. If you use this and set a log-debug before sending the exclusions you will see a message like `[emsesp] Masked entities json size: 1307`. (This size is for my boiler with 67 entities). Yours will be more than 2048, the old limit for http:413.

harvey637, 2022-04-04:
Hi Michael,
tested your special version - WORKING!

This is the log entry: ` [emsesp] Masked entities json size: 2153`

In the first run I removed (hided from api and dashboard)  around 57 entities, making life for iobroker/ems-esp easier,

cu
Harvey

Hope this helps.

MichaelDvP, 2022-04-04:
@proddy: I've counted the worst case possible entities, to calculated buffer/packed size:
Heatpump: 152
RC35 with 4 hc: 173
RC300 with 4 hc: 119 (i've expected more)
Solar 72 with ww

proddy, 2022-04-04:
@MichaelDvP that's good to know. I'll make a change to the customization screen so only the changed entities are sent back instead of the complete list. If we hit a memory limit we can always add a warning and tell users to save in batches

proddy, 2022-04-05:
fixed in b11

closed 2022-04-05 by proddy