emsesp/EMS-ESP32 #448: CRF200S thermostat

zakeajl, 2022-04-04:
Hi, I'm a newbie in the home assistant world.
I managed to connect the gateway without breaking anything (yet) to the heater.
I have a CRF200S thermostat or that's what the EMS panel says. I'd like to know if there is any way to control the heater with this thermostat and the gateway.
I've tried to change some parameters in the heater and it seems it responds when the heating is off in the thermostat.
Anyway, I'd be glad if someone could tell me if there is anything elese to try or if I should just buy another thermostat.
Thank's in advance.

MichaelDvP, 2022-04-05:
Take a look in #138: This thermostat does not reply to any request and do not send out own values (only clock). 
Sorry, but it's not possible to show/edit any values for this thermostat. If you find a way to get any value out of this thermostat, we are happy to implement it.

bbqkees, 2022-04-06:
Do you actually have that thermostat? I thought it was another one.

closed 2023-07-03 by proddy