emsesp/EMS-ESP32 #450: 3.4 b11 - very small observations / clarifications

harvey637, 2022-04-05:
Hi,
saving customization works great - no problems with saving in chunks!

some very small "beautifying" of things:
thermostat:
- "date/time" -> "time/date" (as this is the order of infos: "10:13:31 05.04.2022", or change the value to the more often used term date/time)
- "hc1/summermode" = "on|off" (index is shown, value is more intuitive)
boiler:
- some entities are shown as index, where a value list "On|Off" is known to the api and send correctly, (HP heating, HP Compressor, ...)
- some entities have (short and long) names, but no value (e.g. mixing switch temperature (switchtemp)): could they be marked "no_api + hidden" by default?

please notice that this is NOT complains, just "customer" oberservations in MY real life installation. I want to help to improve even small things, hopefully also only small code changes :-)

Keep on your good work please!
cu
Harvey

harvey637, 2022-04-05:
Hi,
1: mine looks like this: "date/time 16:42:07 05.04.2022", changing one or the other maks it right :-)
2: summer_mode is "0", but e.g. control_mode is "optimized" (also see date/time ... )
[image]
in the api I get the correct valuelist:
```
{
  "name": "summersetmode",
  "fullname": "hc1 set summer mode",
  "circuit": "hc1",
  "value": 1,
  "type": "enum",
  "enum": [
    "summer",
    "auto",
    "winter"
  ],
  "readable": true,
  "writeable": true,
  "visible": true
} 
```
perfect, "auto" is shown in dashboard (see screenshot) , index is given to api (as requested)

for "summermode" in iobroker/ems-esp I get:
```
  "_id": "ems-esp.0.thermostat.hc1.summermode",
  "type": "state",
  "common": {
    "id": "thermostat.hc1.summermode",
    "name": "ems: hc1 summer mode",
    "type": "number",
    "read": true,
    "write": false,
    "role": "value",
    "states": {
      "0": "Off",
      "1": "On"
    },
```

which makes me think the info (On|Off) is somewhere (found in the "states: 0: Off 1: On" part).

4: ok, perfect. My Idea was, that if a NAME is found on the bus but NO value than never a value will be found. If entities are still not all found there is also no NAME. (I have some NAMES for some sensors, that might not be in my bosch 7400, but they are reported (found) by name without value (in contrast to some non_existing sensors with value=0).
5: thank you
6: after changing one entity ONLY this entity is in the customization (only printed last part hiere:)
```
          "03wwdisinfectday"
        ]
      },
      {
        "product_id": 172,
        "device_id": 8,
        "entities": [
          "08heatingtemp"
        ]
      }
    ]
```

but ALL the customizations for the other devices .... strange 
cu
Harvey

MichaelDvP, 2022-04-05:
Please check your setting of bool format:
[image]
The `summersetmode` command is an enum with winter/auto/summer, `summermode` status  is in RC300 bool (on/off), but i see for other thermostats we have also enum winter/summer. Maybe better to set the same scheme for all thermostats.
The entity list seems to be another full buffer, the last list is shown empty if the buffer is noot large enough to hold all entitiies.

harvey637, 2022-04-05:
Hi Michael,
as requested by iobroker/ems-esp plugin I set Boolean Format to 0|1 and Enum Format to Index. I thought that this is ONLY for API, as I see everything else in dashboard as enum-values like On|Off or auto|summer|winter or auto|manual. In the Api I see the requesed index of enums or 0|1 for boolean values.

Only strange with set_summer_mode ????

```
important settings in EMS-ESP:
API V2: MQTT Settings have to be boolean format 1/0 !
API V3: Formatting Options for Boolean Format has to be 1/0 and for Enum Format Number
The Enable API write commands settings within ems-esp has to be enabled
Bypass Access Token authorization on API calls has to be set or the token has to be entered.
```

MichaelDvP, 2022-04-05:
Oh, i thought that thomas has changed it to allow also other settings. I have the ioBroker-adapter installed for testing (but using ioBroker-mqtt with publish on change), and it seems to work with my settings. 

There are a few enums with only  2 options, that's because other thermostats have more options in the same files, prog_1/prog_1 is for Junkers prog_a..prog_f, see [here](https://github.com/emsesp/EMS-ESP32/blob/3d072f73b73f63ac260bd5f78129a72b2a84fe8b/src/locale_EN.h#L385-L388).

harvey637, 2022-04-05:
I will check other format settings with iobroker/ems-esp too.

But some things are sorted out: enums with two values versus booleans - thank you for helping. Do you even think on replacing them with "Off|On" ?

Leaves this date/time thingy (looks easy) and this funny hpheatingon=255 value, as this is boolean, but not 1|0 :-(

Buffer length can wait, as saving in chunks works. I will look further if ther might be an "restore saved settings/customizations". 
cu
Harvey

MichaelDvP, 2022-04-05:
IRC [look here](https://github.com/tp1de/ioBroker.ems-esp/blob/dba4b2c2d05265ae3c694bb562a8f2281684eb11/lib/ems.js#L425-L445) Thomas checks all bool states, and i think enum is also correctly indexed. Youre free to select your favorite bool format.

The hpheatingon=255 is unclear why not handled as bool. I suspect a `auto` compiler directive as cause. 

In my github i've changed thermostat time, summermode, wwstarts2 and maybe bool is now ok, please check.

proddy, 2022-04-05:
> The "api/system/customization" does only contain some of the customisation, and I have a lot of it in product_id=172 (boiler) 

This is a bug, I get the same results. The masked_entities is only saving the last batch to the file system. I'll fix it. Thanks for the testing and reporting Harvey!

harvey637, 2022-04-06:
Hi Micheal,
date/time looks good, summer/winter is great!
api/customisations contains everything (after new setting of it)

Some very very minor opinions:
- what about date in ISO as"2022-04-05", think about british and american users?
- I really dont know, but is winter/summer for the boiler something as "heating period|cooling period" - just thinking ...

- controlstarts2 looks like "dhw control starts2" from the shortname - better long name.
- what about faking dashboard readings of boleans into "true|false" - also just thinking.
- as I see some locale in the source, how do I set it? (do I want to set it? :-)
- hpheatingon still 255 (api is ok with 1|0)

Thank You
Harvey

proddy, 2022-04-06:
> as I see some locale in the source, how do I set it? (do I want to set it? :-)

hmm, renaming an entity? The logic is in the code, as we do this for the sensors. Maybe a new enhancement issue in GitHub for later.

harvey637, 2022-04-06:
Hi @ Michael,
I see a src/locale_DE.h in proddys dev branch :-)

MichaelDvP, 2022-04-07:
> * what about date in ISO as"2022-04-05", think about british and american users?

I don't think bosch heating is very common in US, and ISO is not common in some european countries (Netherlands, Belgium, Switzeland, see here: [https://en.wikipedia.org/wiki/Date_format_by_country](https://en.wikipedia.org/wiki/Date_format_by_country)
> * I really dont know, but is winter/summer for the boiler something as "heating period|cooling period" - just thinking ...

Yes, for gas/oil boilers summer is heating off, only dhw preparation. For heatpump with cooling you have to check the manual.
> * controlstarts2 (wwstarts2) looks like "dhw control starts2" from the shortname - better long name.

all ww.. have full name prefixed with "dhw".
> * what about faking dashboard readings of boleans into "true|false" - also just thinking (the ...on shortnames should expand to ... running?)

In an older version there was dashboard always "on/off", but was later changed to make the selected bool format visible and avoid confusion.
> * as I see some locale in the source, how do I set it? (do I want to set it? :-)

See #22, there is a plan for localization, a while ago i've started to translate the full names to german, but it's not complete. You can replace the include of local_EN.h by local_DE.h and recompile, but webpage, terminal, logs have no localization yet, looks odd.

harvey637, 2022-04-07:
Hi @ Michael,
thank your for your explanation.
- date, ok, I can live with it (although I prefer ISO for automatic sorting e.g. of images, no problem here)
- yes, my bosch has cooling, so I will wait for summer to see the summermode change (setsummermode is auto)
- sorry to say that wwstarts2 still is only "control starts" in proddys latest dev b11
- true|false versus 1|0: now I cant distinguish between eg. "0" as for zero hours running, and on|off

I changed the format to true|false, now I see true|false in the dashboard, but also in the api. So I had to change MY OWN scripts working with the api with the help of regex :-)

Maybe that the api should be (a little bit more) independant from dashboard gui? Or with different format option for dashboard and api?
By the way, the "index|value" selector works this way, changing does not change dashboard, but only api .....

cu
Harvey

proddy, 2022-04-07:
We added the 1|0 for booleans because of Domoticz. Home Assistant can be configured to use anything. I'm not sure what ioBroker uses.

The dashboard was never really supposed to be the final end-stop for looking at the data. It's assumed most people will use home automation front-ends which is why its just a simple table (not filtering, sorting, searching...).

Adding a Boolean Format for the Dashboard is easy to do, if we want it.

harvey637, 2022-04-07:
Hi @ Proddy,
thank you.
iobroker/ems-esp plugin uses a lot of stuff like true|false, 0|1 "true"|"false". It also knows about booleans and enumerations.
But on top I also have some homewritten scripts in simple tcl or shell, where I use regex to fetch the "true" state.

It was a little bit confusing for me, that the "format" option for booleans are for both dashboard and api, but the format for enumerations is only for api. I see this as changing the format of both does only alter representation of boolenas in the dashboard AND api, but enumerations are allways shown as values in dashboard (whatever format enum option), but the api representation is changed accordingly.

So I think, there should be an explanation that the boolean option is for both, enum option only for api.

OR EVEN BETTER:
format option for dashboard (maybe we never need dashboard enum format option), so only dashboard-boolean format option, 
AND both format options for api, to satisfy domotiz with 1|0 and possible others with different format, and the enums as index (for systems handling index of enums and values of enums) or value (mostly for dump display systems.

just my idea ..... 

and please fix this simple "wwstarts2" -> "dhw controls starts2" long name (as Michael has done it)

Thank you and
cu
Harvey

proddy, 2022-04-08:
> sorry to say that wwstarts2 still is only "control starts" in proddys latest dev b11

@MichaelDvP is this changed in your build or shall I add it?

MichaelDvP, 2022-04-08:
Partly in my build, i have to change the tag to ww, i'll make a pr. I was thinking about the clock format. The thermostat clock is read every minute, publish is mostly asynchron wth 10 seconds, showing the clock with seconds is always incorrect. What do you think about publishing only "hh:mm"?
(The "last_error" are already in format "dd.mm.yyyy hh:mm")

harvey637, 2022-04-08:
Moin Michael,
"hh:mm" sounds very good to me! "dhw control starts2" also.
thank you
Harvey

MichaelDvP, 2022-04-08:
Ok, i'll also add RCTime-Message to controller, this strange IVT in #439 publishes the thermstat time via controller if thermostat display is in sleep. (and show  controller in web if it has entities)

closed 2022-04-24 by proddy