emsesp/EMS-ESP32 #454: Tx request quality 85% - BBQKees gateway - Bosch 7000iw

Ironirc, 2022-04-06:
The gateway is the only connected device on the bus (no thermostat).
Already tried a second gateway from BBQKees, using different cables.
Always tried all different modes, so it's probably not a hardware problem (at least not at the gateway's side).

See discussion near: https://discord.com/channels/816637840644505620/816958087160266775/961213369443037196

proddy, 2022-04-09:
any other users out there with a 7000iw to compare too?

elRadix, 2022-04-09:
I have a Bosch GC7000iW with Gateway S32 running fine, no connection issues
[image]

Ironirc, 2022-04-09:
@elRadix  Thanks for your feedback. 
I'm using 3.4.0b8  => Just upgraded. (might this be relevant @proddy ?)
Which TX mode are you using?
Are there other devices on the bus?

elRadix, 2022-04-09:
additional info if that helps, i don't have a thermostat

```
{
  "System": {
    "version": "3.4.0b11",
    "uptime": "000+22:21:06.495",
    "uptime (seconds)": 80466,
    "network time": "connected",
    "freemem": 165,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "bosch",
    "SSID": xxxxt",
    "BSSID":xxx",
    "RSSI": -36,
    "MAC": "3C:61:05:03:AC:B0",
    "IPv4 address": "192.168.1xxxx/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.xx"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 45737,
    "bus reads (tx)": 6721,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 6,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 17748,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 8027,
    "API fails": 2,
    "syslog started": true,
    "syslog level": "error",
    "syslog ip": "192.168.1.235",
    "syslog queue": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 60,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}

```

Ironirc, 2022-04-09:
It might be too early for definitive conclusions, but it seems that the upgrade has solved my problem?
[image]

proddy, 2022-04-10:
I can't recall any changes in the Tx code between b8 and b11. I'd say if it works great!

proddy, 2022-04-14:
closing this, I'm guessing it all works fine now? :)