emsesp/EMS-ESP32 #459: EMS-ESP only seeing Junkers boiler but neither the JUNKERS SAS HYprid Heat Pump nor the FW200

not-so-important, 2022-04-08:
**Question**
I yesterday received the EMS-ESP device from BBQKees.
EMS-ESP can see my Junkers boiler correctly, but none of the other devices are showing. 

My setup:
E32 Gateway connected to the service jack of the in 2013/2014 built Junkers Cerapur ZSB-14.
Junkers SupraECO SAS Hybrid HSC Heat Pump is installed and in the BUS as well.
FW200 Thermostat is installed in the living room controlling all parameters of the gas heating and the heat pump.
All devices are ok for controlling as of the Production date (e.g. FW200 is FD306)

A regular scan for devices didn't change anything. 
A deep scan (via telnet) showed some for me still weired output (see attached screenshots) and an unknown Thermostat.

Any help would be very much appreciated, for me in order to see and control with EMS-ESP via the FW200 and see/control the heat pump as well. 

The main use case will be:
I especially want to control the usage distribution between the gas heating and the heat pump via the gas/power price relation once my photovoltaik with batteries is installed and connected to my iobroker and MQTT based home automation.

Would it possibly make a difference to not use the service jack but the direct bus connect at the boiler parallel to the FW200?
I also updated FW to Version 3.4.0.0b11 without any change already ...

**Screenshots**

Result of the deep scan:

[image]

Unknown traffic on the bus:

[image]

Result of "read 17 2":

[image]

**Device information**

{
  "System": {
    "version": "3.4.0b11",
    "uptime": "000+14:28:46.699",
    "uptime (seconds)": 52126,
    "network time": "disconnected",
    "freemem": 109,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "XX:XX:XX:XX:XX:XX",
    "IPv4 address": "192.168.1.200/255.255.255.0",
    "IPv4 gateway": "192.168.1.254",
    "IPv4 nameserver": "192.168.1.254"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 298165,
    "bus reads (tx)": 5591,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 282,
    "bus reads failed": 31,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 2605,
    "API fails": 1
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x08",
      "product id": 123,
      "version": "06.02",
      "entities": 66,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "generic thermostat",
      "device id": "0x17",
      "product id": 0,
      "version": "00.00",
      "entities": 0,
      "handlers pending": "0xA3 0x06 0xA2 0x12 0xB1 0xB0"
    },
    {
      "type": "Controller",
      "name": "Controller",
      "device id": "0x09",
      "product id": 152,
      "version": "02.08",
      "entities": 0
    },
    {
      "type": "Connect",
      "name": "cascade",
      "device id": "0x0C",
      "product id": 0,
      "version": "00.00",
      "entities": 0
    }
  ]
}

MichaelDvP, 2022-04-08:
Seems you have a cascade module with heatpump and boiler connected on one side and thermostat on the other. Right? You have to connect ems-esp to the side of the thermostat.

not-so-important, 2022-04-08:
The heat pump itself has a controller. The FW200 is connected to that controller as well as the boiler bus. So it seems to be that the Hybrid controller is the center of action, indeed. I'll try to reconnect to the controller directly and will let you know the result.

MichaelDvP, 2022-04-08:
Then the cascade is integrated in heatpump. 
EMS have a single master, which is always the boiler, and clients like thermostat, ems-esp, etc. The cascade makes seperated ems-buses, one for each boiler/heatpump and a combined client bus for thermostat and ems-esp, etc. In this bus the cascade is master and provides a combined boiler information on boiler adress (device-id 0x08) and a few parameters per heating-source (operating hours, etc), on device-ids 0x70, 0x71, ....
On the bus from boiler-service-jack you see only the communication between cascade and this boiler. You have to connect to the bus where thermostat is connected.

not-so-important, 2022-04-08:
Reconnected to the Heat Pump controller results in a different, but nevertheless unsatisfying result.
I can see the FW200 with only few parameters (everything around the Heat Pump controlling is missing).
I can see the boiler, but without values at all plus two unknown devices.

[image]

Here's the changed config output:

{
  "System": {
    "version": "3.4.0b11",
    "uptime": "000+00:05:10.853",
    "uptime (seconds)": 310,
    "network time": "disconnected",
    "freemem": 133,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "8C:CE:4E:95:FD:A7",
    "IPv4 address": "192.168.1.200/255.255.255.0",
    "IPv4 gateway": "192.168.1.254",
    "IPv4 nameserver": "192.168.1.254"
  },
  "Status": {
    "bus status": "connected, tx issues - try a different tx-mode",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 482,
    "bus reads (tx)": 64,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 2,
    "bus reads failed": 11,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 86,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 12,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x70",
      "product id": 123,
      "version": "06.02",
      "entities": 0,
      "handlers pending": "0x06DC 0xE4"
    },
    {
      "type": "Thermostat",
      "name": "FW200",
      "device id": "0x10",
      "product id": 0,
      "version": "00.00",
      "entities": 10,
      "handlers received": "0x06 0x0166",
      "handlers fetched": "0x016F 0x0165",
      "handlers pending": "0xA3 0xA2 0x12 0x0170 0x0171 0x0172 0x0167 0x0168 0x0123"
    },
    {
      "type": "Unknown",
      "name": "unknown",
      "device id": "0x08",
      "product id": 168,
      "version": "01.05",
      "entities": 0
    },
    {
      "type": "Unknown",
      "name": "unknown",
      "device id": "0x09",
      "product id": 168,
      "version": "01.05",
      "entities": 0
    }
  ]
}

Here the result from another deep scan in this config:

[image]

And again ... lots of unknown traffic:

[image]

not-so-important, 2022-04-08:
Wow, that was quick ... much better!

I can see all the gas related boiler parameters and the temperature related FW200 settings now, which is great!
I unfortunately still do not see any of the heatpump related settings like gas/power price ratio, hybrid control method: economical, ecological ... etc. etc.
The Heatpump controller automatically steers whether to use the heat pump or the gas boiler to heat, depending on these specific settings. This all can be controlled directly from within the FW200 by changing the respective parameters.

Any chance to get these parameters showing up as well. I am happy to support this where and however possible.
Would you need any further information? Can I log anything that would help?

The new device setup after the FW change:

{
  "System": {
    "version": "3.4.0b11",
    "uptime": "000+00:16:58.437",
    "uptime (seconds)": 1018,
    "network time": "disconnected",
    "freemem": 125,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "XXX",
    "IPv4 address": "192.168.1.200/255.255.255.0",
    "IPv4 gateway": "192.168.1.254",
    "IPv4 nameserver": "192.168.1.254"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 1214,
    "bus reads (tx)": 168,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 4,
    "bus reads failed": 6,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 97,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 64,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Hybrid Heatpump",
      "device id": "0x08",
      "product id": 168,
      "version": "01.05",
      "entities": 65,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x0494 0x0495 0x048D 0x048F 0x048A"
    },
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x70",
      "product id": 123,
      "version": "06.02",
      "entities": 0,
      "handlers pending": "0x06DC 0xE4"
    },
    {
      "type": "Thermostat",
      "name": "FW200",
      "device id": "0x10",
      "product id": 106,
      "version": "12.14",
      "entities": 10,
      "handlers received": "0x06 0x0166",
      "handlers fetched": "0x016F 0x0165",
      "handlers pending": "0xA3 0xA2 0x12 0x0170 0x0171 0x0172 0x0167 0x0168 0x0123"
    },
    {
      "type": "Controller",
      "name": "Hybrid Heatpump",
      "device id": "0x09",
      "product id": 168,
      "version": "01.05",
      "entities": 0,
      "handlers pending": "0x06"
    }
  ]
}

Output of show values:

[image]

[image]

Much less unknown traffic on the bus:

[image]

MichaelDvP, 2022-04-08:
> This all can be controlled directly from within the FW200 by changing the respective parameters.
> Any chance to get these parameters showing up as well. I am happy to support this where and however possible.

If you can change these parameters on FW200, there is a message to the boiler/pump/controller containing the parameter. You have to log the messages and change the parameter on the thermostat. After this change there will be a message `Thermostat(0x10) -> <dest> <telegram> <data> (offset z)`, we need to know dest, telegram, offset and the meaning of data-value to add it to ems-esp.  In terminal you can use `watch 10` to see only the telgrams from/to thermostat. 
After a change often the destination broadcasts a the same telegram to All(0x00) with all data.

The unknown you already logged:
0x23 is first byte requested temperature after header (0x2B =43°C), second byte power (0x64 = 100%).
0xBC looks like it contains some measured temperatures in offset 4 and 6, (0x016C = 36,4°C ), you have to check what temperatures are reported in thermostat with this values.
Telegrams 0x29 and 0x08 i have no idea, the values are not changing.

If you need help with the logs, please attach as txt file, it's easier to search and edit. And add the description what (name) was changed in what order and to what value.

not-so-important, 2022-04-08:
O.K. ... will do so and attach the results as txt ... watch this space ... 
Anyhow, great support here ... thank you!

not-so-important, 2022-04-08:
Here is a block of a "watch 10" log where I changed the gas/power cost ratio factor from 3,7 to 3,6 and back to 3,7 in the FW200.
I am not yet fully able to read the log properly ... need to get used to the hex coding of the EMS/HT3 bus.

Are you able to locate where the setting was changed in the log?

000+01:36:05.987 N 36: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+01:36:06.006 N 37: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02
000+01:36:11.283 N 38: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 11 08 3A 0B 05 00
000+01:36:18.345 N 39: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+01:36:18.360 N 40: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+01:36:19.202 N 41: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00
000+01:36:54.648 N 42: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 32 (offset 1)
000+01:36:54.668 N 43: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2B
000+01:36:54.693 N 44: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+01:36:54.730 N 45: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2B 64 00
000+01:36:54.774 N 46: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
000+01:37:01.298 N 47: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:37:02.907 N 48: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+01:37:02.984 N 49: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 1E 41 41 07 01 00 03 FF 21 06 01 04 1C 26 2A 02 01 00
000+01:37:07.933 N 50: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+01:37:07.953 N 51: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02
000+01:37:11.390 N 52: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 11 08 3B 0B 05 00
000+01:37:19.359 N 53: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+01:37:19.375 N 54: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+01:37:19.816 N 55: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00
000+01:38:01.658 N 56: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:38:01.704 N 57: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00
000+01:38:01.909 N 58: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+01:38:01.979 N 59: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 1E 41 41 07 01 00 03 FF 21 06 01 04 1C 26 2A 02 01 00
000+01:38:07.967 N 60: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+01:38:07.985 N 61: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02
000+01:38:11.048 N 62: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 12 08 00 0B 05 00
000+01:38:19.924 N 63: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+01:38:19.958 N 64: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+01:38:20.397 N 65: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00
000+01:38:34.754 N 66: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 24 0B 1A 00 14 03 (offset 14)
000+01:38:43.296 N 67: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 25 0B 1A 00 14 03 (offset 14)
000+01:38:44.144 N 68: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xBE), data: 10 00 00 00 00
000+01:38:44.553 N 69: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x01D3), data: 00 00 00
000+01:38:56.448 N 70: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 32 (offset 1)
000+01:38:56.468 N 71: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2B
000+01:38:56.492 N 72: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+01:38:56.529 N 73: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2B 64 00
000+01:38:56.575 N 74: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
000+01:39:01.969 N 75: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:39:02.015 N 76: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 00 D0 F3 34 00
000+01:39:02.220 N 77: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19

MichaelDvP, 2022-04-08:
Looks like offset 14 in Telegram 0xBB, a bit uncommon that the thermostat send 6 bytes, maybe there are some associated settings. ratio 3.7 = 0x25 with factor 10, ratio 3.6 = 0x24: 
```
000+01:38:34.754 N 66: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 24 0B 1A 00 14 03 (offset 14)
000+01:38:43.296 N 67: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 25 0B 1A 00 14 03 (offset 14)
```
Can you check in terminal `read 8 BB` to see the complete telegram.  It seems there are more information stored.

MichaelDvP, 2022-04-08:
> Let me do some more parameter changes to the hybrid system and see if I find some more interesting bits to add to EMS-ESP ... > I begin to understand the telegram coding ...

👍

not-so-important, 2022-04-08:
Here's what happens when I switch the FW200 settings as follows ... Just a first try to find out if this reverse engineering may work ... there are several further parameter possibilities with e.g. CO2 optimized setup etc.

1) Change the gas/power price ratio to 3,7
2) Switch the cost optimization mode towards the fixed temperature mode (default value after switchin is 6 degree)
3) Changed the switchover temperature to 10 degrees
4) Switched the switchover temperature to 15 degrees
5) switched back to the cost optimization mode
6) Change the gas/power price ration to 3,7 again (default is 3,3 after switching modes)

##########

ems-esp:$ watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x10
000+01:24:02.719 N 1: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:24:02.764 N 2: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00
000+01:24:02.970 N 3: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+01:24:03.044 N 4: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 1E 41 41 07 01 00 03 FF 21 06 01 04 1C 26 2A 02 01 00
000+01:24:13.993 N 5: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 00 09 0C 0E 06 00
000+01:24:44.823 N 6: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+01:24:44.846 N 7: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02
000+01:24:48.480 N 8: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 25 0B 1A 00 14 03 (offset 14)
000+01:24:56.054 W 9: [telegram] Incomplete Rx: FF 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 03 00
000+01:24:57.083 N 10: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+01:24:57.095 N 11: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+01:24:57.531 N 12: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00
000+01:25:02.154 N 13: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:25:02.933 N 14: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:25:03.205 N 15: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 03 06 25 0B 1A 00 14 03 (offset 12)
000+01:25:03.410 N 16: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:25:03.457 N 17: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00
000+01:25:04.063 N 18: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+01:25:04.132 N 19: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 1E 41 41 07 01 00 03 FF 21 06 01 04 1C 26 2A 02 01 00
000+01:25:08.616 N 20: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2C
000+01:25:08.641 N 21: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2C
000+01:25:13.830 N 22: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 00 09 0D 0D 06 00
000+01:25:16.239 N 23: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 0A 25 0B 1A 00 14 03 (offset 13)
000+01:25:26.464 N 24: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 0F 25 0B 1A 00 14 03 (offset 13)
000+01:25:34.220 N 25: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 01 00 0F (offset 1)
000+01:25:34.264 N 26: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2C
000+01:25:34.289 N 27: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+01:25:34.318 N 28: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2C 64 00
000+01:25:34.362 N 29: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
000+01:25:45.617 N 30: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+01:25:45.635 N 31: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02
000+01:25:48.494 N 32: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 02 0F 21 0B 1A 00 14 03 (offset 12)
000+01:25:55.322 N 33: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 25 0B 1A 00 14 03 (offset 14)
000+01:25:57.975 N 34: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+01:25:57.986 N 35: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+01:25:58.627 N 36: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 D1 F3 34 00
000+01:26:04.352 N 37: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+01:26:04.401 N 38: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 00 D1 F3 34 00
000+01:26:04.830 N 39: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+01:26:04.904 N 40: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 1E 41 41 07 01 00 03 FF 21 06 01 04 1C 26 2A 02 01 00
000+01:26:13.766 N 41: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 00 09 0E 0D 06 00
000+01:26:46.459 N 42: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+01:26:46.500 N 43: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02
ems-esp:$ watch off

##########

This looks like the first value switch at 1)
[... cut ...]
000+01:25:34.318 N 28: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2C 64 00

I assume this is changing back to cost optimization under 5) with the standard value of 3,3
000+01:25:48.494 N 32: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 02 0F 21 0B 1A 00 14 03 (offset 12)
0x21 = 33 / 10 = 3,3

And here finally the switching back to the factor of 3,7 under 6)
000+01:25:55.322 N 33: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 25 0B 1A 00 14 03 (offset 14)
0x25 = 37 / 10 = 3,7

Makes sense?!

not-so-important, 2022-04-09:
Next log with comments inline:
After this I think we got the settings for "Cost Optimized" and "Outside Temperatue Switchover" Modes (see above).
This one gives the settings for the CO2-Cost-Mix Mode.
Missing (next log) will be the settings for the fourth and final mode, which is pure CO2-Optimized.

What I did:

1) Switched to CO2-Cost-Mix setting
2) Changed the fossile energy factor, default value 1,1 to 1,2 
3) Changed the electric power energy factor, default value 2,6 to 2,7 
4) electric power factor back to 2,6
5) fossile factor back to 1,1
6) Setting back to Cost Optimized mode, factor default 3,3
7) Set the cost factor back to 3,7
8) Changed the Wait time for support from the boiler, default 20 min to 19min and back to 20 min
9) Changed the Temp difference for boiler support, default 3 Kelvin to 4 Kelvin and back to 3 Kelvin

ems-esp:$ watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x10
000+13:37:17.566 N 0: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 32 (offset 1)
000+13:37:17.610 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2B
000+13:37:17.635 N 2: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+13:37:17.663 N 3: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2B 64 00
000+13:37:17.684 N 4: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
000+13:37:19.457 N 5: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 0C 09 19 15 06 00
000+13:37:29.821 N 6: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+13:37:29.838 N 7: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02
000+13:37:42.179 N 8: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+13:37:42.188 N 9: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+13:37:42.831 N 10: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CE F3 34 00
000+13:38:04.313 E 11: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 88 16 00 1B
000+13:38:07.197 W 12: [telegram] Incomplete Rx: FF 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 03 00
000+13:38:07.946 N 13: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+13:38:09.102 N 14: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+13:38:09.171 N 15: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 1E 41 41 07 01 00 03 FF 21 06 01 04 1C 26 2A 02 01 00

**#####
000+13:38:11.636 N 16: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 04 0F 21 0B 1A 00 14 03 (offset 12)
Mode change done at point 1) towards CO2-Cost-Mix
Default cost ratio is 3,3 as reflected by 0x21
#####**

000+13:38:19.245 N 17: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 0C 09 1A 15 06 00
000+13:38:27.394 N 18: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2A
000+13:38:27.419 N 19: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2A
000+13:38:30.581 N 20: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+13:38:30.604 N 21: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: A8 01 05 00 00 00 00 00 00 02

**#####
000+13:38:30.820 N 22: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 0C 1A 00 14 03 (offset 15)
Change under 2) fossile factor to 1,2 reflected at 0x0C = 12 / 10 = 1,2
#####**

**#####
000+13:38:41.639 N 23: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 1B 00 14 03 (offset 16)
Change under 3) electric factor to 2,7 reflected at 0x1B = 27 / 10 = 2,7
#####**

000+13:38:42.490 N 24: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+13:38:42.500 N 25: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+13:38:43.142 N 26: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 CF F3 34 00

**#####
000+13:38:57.471 N 27: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 1A 00 14 03 (offset 16)
Second change under 4) electric factor back to 2,6 reflected at 0x1A = 26 / 10 = 2,6
#####**

**#####
000+13:39:01.969 N 28: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 0B 1A 00 14 03 (offset 15)
[... 33 lines cut ...]
000+13:40:03.269 N 49: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+13:40:03.345 N 50: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 1E 41 41 07 01 00 03 FF 21 06 01 04 1C 26 2A 02 01 00

**#####
000+13:40:13.397 N 51: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 13 03 (offset 18)
000+13:40:17.238 N 52: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 14 03 (offset 18)
Changes under 8) Time change reflected in 0x13 = 19min and back to 0x14 = 20min
#####**

000+13:40:19.467 N 53: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 0C 09 1C 14 06 00

**#####
000+13:40:25.025 N 54: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 04 (offset 19)
000+13:40:28.589 N 55: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xBB), data: 03 (offset 19)
Changes under 9) Temp difference setting change reflected in 0x04 = 4 Kelvin and back to 0x03 = 3 Kelvin
#####**

[... cut ...]
000+13:40:39.991 N 58: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 29
000+13:40:40.023 N 59: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 29
000+13:40:44.085 N 60: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+13:40:44.100 N 61: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 41
000+13:40:44.541 N 62: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 00 D2 F3 34 00
ems-esp:$ watch off

MichaelDvP, 2022-04-10:
Wow, good analysis. 
The 0xFF in offset 11 is probably an "activated" setting. 
Please suggest short (api/mqtt)  and full names (web) for these parameters and check if the values are always positive or  allows negative setting (the temp difference?). If you know (or determine on thermostat) the min/max limits for each value that will be perfect.

Thiis is what i set for testing:
```
// hybrid heatpump
MAKE_PSTR_LIST(enum_gasPriceMode, F("co2"), F("cost"), F("temperature"), F("mix"))
MAKE_PSTR_LIST(gasPriceMode, F("gaspricemode"), F("gas price mode"))
MAKE_PSTR_LIST(switchOverTemp, F("switchovertemp"), F("switch over temperature"))
MAKE_PSTR_LIST(gasPriceRatio, F("gaspriceratio"), F("gas price ratio"))
MAKE_PSTR_LIST(fossileFactor, F("fossilefactor"), F("fossile factor"))
MAKE_PSTR_LIST(electricFactor, F("electricfactor"), F("electric factor"))
MAKE_PSTR_LIST(waitBoiler, F("waitboiler"), F("wait boiler"))
MAKE_PSTR_LIST(tempDiffBoiler, F("tempdiffboiler"), F("temperature difference boiler"))
```

The info shows, that this pump mainly sends HT3 telegrams, not the ems+ heatpump values (all pending):
```
"handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
"handlers fetched": "0x14 0x16 0x33",
"handlers pending": "0xBF 0xC2 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x0494 0x0495 0x048D 0x048F 0x048A"
```
i've changed the setting to reduce pending handlers and empty entities in custoization, check if now something is missing.
[Software is here.](https://github.com/MichaelDvP/EMS-ESP32/releases)

not-so-important, 2022-04-10:
Thanks! I'll give the new build a try later on today ...

The Junkers system should be all HT3 based as of a "friend" from Junkers ... 
BTW ... just in case ... it is correctly called " SUPRAECO SAS Hybrid HSC"

Here some further Infos on the parameter names from the Junkers Manual ... 
Suggestions (MQTT/Junkers Naming for API) in brackets:

Bit 13  (hybrid_strategy / Hybrid-Control-Strategy)
0x01 (co2 / CO2-Optimized)
0x02 (cost / Cost-Optimized)
0x03 (outside_temp / Outside-Temp-Switched)
0x04 (mix / CO2-Cost-Mix)

Bit 14 (switchover_temp / Outside-Switchover-Temp) ... goes from -20 degree to +20 degree
Bit 15 (energy_cost_ratio / Energy-Cost-Ratio) ... goes from 0,0 to 19,9 ... only positive numbers
Bit 16 (fossile_factor / Fossile-Energy-Factor) ... goes from 0,0 to 5,0 ... only positive numbers
Bit 17 (electric_factor / Electrical-Energy-Factor) ... goes from 0,0 to 5,0 ... only positive numbers

Bit 19 (delay_boiler_support / Delay-Boiler-Support) ... goes from 5min to 120min ... only positive numbers
Bit 20 (tempdiff_boiler_support / Tempdiff-Boiler-Support) ... goes from 1K to 99k ... only positive numbers

More to come on the further "still unknown" communication on the bus as soon as I have a bit more time again.

not-so-important, 2022-04-10:
Another finding fith the new version:

Lots of values on the FW200 side are gone now ...

[image]

MichaelDvP, 2022-04-10:
> Found one thing ... I changed the gas price ratio to 3,6 and the HP seems to take that value as the read 8 BB below shows ...
> But when physically checking the FW200, it keeps the old facor of 3,7 ... hmmm.

Check with `watch BB` and change the value via emsesp. Maybe the thermostat overwrites the value. Next is to  check if we can write BB to the thermsotat. First check with `read 10 BB` if the thermostat have a copy of this telegram. 
The try with `call system send "0B 10 BB 0E 24"` to change the value in offset 14 (0E) to 3.6 (24). 

The thermostat has not changed, i think it takes a minute to fetch all values.

not-so-important, 2022-04-10:
The thermostat has a copy:

ems-esp:$ read 10 BB
000+01:44:17.962 I 1: [emsesp] Thermostat(0x10) -> Me(0x0B), HybridHp(0xBB), data: 00 00 00 00 00 00 00 00 00 00 00 00 02 0F 25 0B 1A 00 14 03

and yes, I can set it:

ems-esp:# call system send "0B 10 BB 0E 24"
000+01:46:54.028 I 5: [emsesp] Thermostat(0x10) -> Me(0x0B), HybridHp(0xBB), data: 00 00 00 00 00 00 00 00 00 00 00 00 02 0F 24 0B 1A 00 14 03

and the FW200 after that shows the right value of 3,6!

MichaelDvP, 2022-04-10:
Ok, we have to move reading/writing of these values to thermostat.

not-so-important, 2022-04-10:
Yes .. just checked ... after a single change to "08 10 BB 0E 26", the Thermostat then instantly syncs the value down to the HP ...

ems-esp:# read 10 bb
000+02:38:27.363 I 35: [emsesp] Thermostat(0x10) -> Me(0x0B), HybridHp(0xBB), data: 00 00 00 00 00 00 00 00 00 00 00 00 02 0F 26 0B 1A 00 14 03
ems-esp:# read 8 bb
000+02:38:32.867 I 36: [emsesp] Boiler(0x08) -> Me(0x0B), HybridHp(0xBB), data: 00 00 00 00 00 00 00 00 00 00 00 FF 02 0F 26 0B 1A 00 14 03

MichaelDvP, 2022-04-10:
Moved the hybrid settings to thermostat, please check.

norberts1, 2022-04-10:
@MichaelDvP that document regarding MsgID 187 (BB)hex is updated now.
https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf
I'll hope it helps a bit.

not-so-important, 2022-04-10:
Works nicely now with the new version ... I can change the values and they instantly sync from the FW200 Thermostate to the HP ... Great job ... Thanks so far ...

I'll look into the further parameters as time allows ...

Is it ok to just keep this thread open to complete the setup?

closed 2023-02-19 by proddy