emsesp/EMS-ESP32 #461: How does the Timer processing of a thermostat clock program works?

kwertie01, 2022-04-08:
I'm working on adding the switch points of the Moduline 400 (RC30) thermostat Clock Program. This information is using the 0x3F telegram which is almost the same as the RC35 thermostat. The difference is that the Moduline 400 is using 4 temperature states (T1-T4) and the RC35 only 2.
Now I'm looking at the process_RC35Timer function and it looks like it only gets the first switch point and not all switch points of the 
program (max 42 switch points) 

[image]

**Question:**
Is this by intention or do I just don't understand how it should work? (probably the last 😂 )

I see btw that setting the switch point does support setting all of the 42 switch points one by one.

The reason why I ask it is because I don't want to immediately drastically change the timer function when there is a specific reason to only get the first switch point. 
Getting the data from telegram 0x3F is already complex enough because it is a Long telegram which you receive 4 times with a different offset and length to get it completely. 

Example of the output of my Moduline 400 with the 19 switch points I like to see

`10 0B 3F 00 03 36 03 66 01 81 23 36 23 66 21 81 43 57 43 66 41 81 63 36 63 66 61 81 83 2D 83 66 81 84 A3 3F A1 84 C3 3F C1 84 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 01 00 00 01 01 00 01 01 00 01 01 00  01  01  00 00`

```
offset	value	day	T	Time
00	03 36 	Mo	3	09:00    (this switch point is shown)
02	03 66	Mo	3	17:00
04	01 81   Mo	1	21:30
06	23 36 	Tu	3	09:00
08	23 66 	Tu	3	17:00
0A	21 81 	Tu	1	21:30
0C	43 57 	We	3	14:30
0E	43 66 	We	3	17:00
10	41 81 	We	1	21:30
12	63 36 	Th	3	09:00
14	63 66 	Th	3	17:00
16	61 81 	Th	1	21:30
18	83 2D 	Fr	3	07:30
1A	83 66 	Fr	3	17:00
1C	81 84 	Fr	1	22:00
1E	A3 3F 	Sa	3	10:30
20	A1 84 	Sa	1	22:00
22	C3 3F 	Su	3	10:30
24	C1 84	Su	1	22:00
```
Thanks

MichaelDvP, 2022-04-08:
The reason is not blow up the json in mqtt and for webpage. The logic is:
Writing only switch index publishes the entry of the new index. 
Writing index and data changes the index. data "not_set" clears a switchpoint.
This also shown in the "Format:" helptext in webpage: 
[image]
To change a complete program by hand will take some time, I think this is something to do by script.

MichaelDvP, 2022-04-08:
Yes RC35 is only on/off, but you can enter the temperature state as single number 0-3, for RC35 and RC30_N this is limited to 0-1. See [here](https://github.com/emsesp/EMS-ESP32/blob/555b1ad996c2ea5ad7651761ea3f1a2b9f6a6ecd/src/devices/thermostat.cpp#L2356) and [here](https://github.com/emsesp/EMS-ESP32/blob/dev/src/devices/thermostat.cpp#L2371-L2374).

kwertie01, 2022-04-08:
Thanks Michael for your answers! Really appreciated.
Writing only the index is indeed working. I tried that already before but forgot to set the length to 2. (my bad...)
Blowing up the json in mqtt could be a risk to be aware of. I will start implementing the Moduline 400 using the same method which is currently implemented for the RC35 and using a number instead of the on/of. 
Probably is is better to use 1-4 instead of 0-3 because it is then the same as the official naming of the 4 temperature levels in the Thermostat
T1=night, T2=day low, T3=day mid and T4=day high.
About reading/changing the whole program, I will think about it a little longer. You are right that doing it by hand is difficult. but I must say, changing it on the thermostat itself is also not that easy.  Would be nice to modify the program by sitting behind the PC using a web browser for instance. Just thinking out loud....

MichaelDvP, 2022-04-08:
In your example there is only T1-T3, is the low nibble of first byte 1-4 or 0-3? Maybe use single number 0-3 or alternative T1-T4?
Adding a nice UI as spreadsheet for the whole program is imho part of individual homeautomation system, ems-esp should only give access to all values.

proddy, 2022-04-09:
> Would be nice to modify the program by sitting behind the PC using a web browser for instance. Just thinking out loud....

Could you not create a script calling the API via Curl to set each individual switch point? See https://emsesp.github.io/docs/#/Command?id=examples.

A nice enhancement for later is a command (API or Console) to fetch and display all the times for a program.

And while thinking about this, at some point it would be good to be able to call commands from the WebUI (like watch, publish, fetch, reset..)

closed 2023-02-19 by proddy