emsesp/EMS-ESP32 #462: how to fetch only value of entity from api, not detailed info?

harvey637, 2022-04-08:
Hi, how do I get only the value of an entity via api?

the documented simple fetch does result in detailed info about entity, not only value ;.(

```
/usr/bin/wget --no-check-certificate -q http://ems-esp/api/thermostat/hc1/seltemp -O -
{
  "name": "seltemp",
  "fullname": "hc1 selected room temperature",
  "circuit": "hc1",
  "value": 20,
  "type": "number",
  "min": 5,
  "max": 29,
  "uom": "°C",
  "readable": true,
  "writeable": true,
  "visible": true
}
```

I hope to get only "20" 

cu
Harvey

proddy, 2022-04-09:
you can't, but we can consider adding a "/value" after the entity to bring only the value back.

harvey637, 2022-04-09:
Hi Proddy,
excellent idea, thank you
looking forward to test :-)
cu
Harvey

closed 2022-05-02 by proddy