emsesp/EMS-ESP32 #465: Config conflict: wwdisinfect

sciurius, 2022-04-10:
**Bug description**

EMS-ESP version 3.4.1b11

`homeassistant/binary_sensor/ems-esp/boiler_wwdisinfect/config` specifies `"val_tpl": "{{value_json.wwdisinfect}}"` but the actual JSON data seems to be `wwdisinfecting`. As a result, `HomeAssisant` complains:

    'dict object' has no attribute 'wwdisinfect' when rendering '{{value_json.wwdisinfect}}'

`homeassistant/binary_sensor/ems-esp/boiler_wwheat/config` specfies `"val_tpl": "{{value_json.wwheat}}"` but the actual JSON data doesn't have a `wwheat` attribute. HomeAssistant complains:

    'dict object' has no attribute 'wwheat' when rendering '{{value_json.wwheat}}'

MichaelDvP, 2022-04-11:
Go to "MQTT Settings" and disable "MQTT Discovery" setting (do NOT disable mqtt), click "Save", wait 5 minutes, reenable "MQTT Discovery"and click "Save".

sciurius, 2022-04-11:
IOW, these are leftovers from the previous version and retained by the MQTT broker?

proddy, 2022-04-11:
yes. The changes we made back in Dec 2021 (https://github.com/emsesp/EMS-ESP32/issues/211) and also mentioned in the Changelog. And there is also a knowledge article on the wiki on [this](https://emsesp.github.io/docs/#/Troubleshooting?id=ha-is-showing-errors-like-quotreceived-message-on-illegal-discovery-topicquot-or-quottemplate-variable-warning-39dict-object39-has-no-attributequot).

sciurius, 2022-04-11:
Then I gently apologise for not having taken the appropriate action at the time I read the release notes.
Thanks for the good work!

closed 2023-02-19 by proddy