emsesp/EMS-ESP32 #468: Is it possible to activate "Immediate DHW" on FW100?

pdbell, 2022-04-13:
Hi,

I currently have a FW100 connected a Bosch Boiler (details below) and I'm trying to work out if its possible to activate the "Immediate DHW" on the FW100 from the Bus, as if the advance button was pressed (button 4 in the diagram)

[image]

I am able to write setpoints on the FW100 for the heating mode, but so far I haven't found a parameter or telegram to help for DHW.  I am capable and happy to help with debugging data if it may be possible, but I thought I'd ask before digging too much further.

```
{
  "System": {
    "version": "3.4.0b8",
    "uptime": "000+22:10:02.466",
    "uptime (seconds)": 79802,
    "freemem": 146,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "SSID": "Landlyst",
    "BSSID": "7A:45:58:18:84:85",
    "RSSI": -68,
    "MAC": "48:55:19:BC:EC:30",
    "IPv4 address": "192.168.2.127/255.255.255.0",
    "IPv4 gateway": "192.168.2.1",
    "IPv4 nameserver": "192.168.2.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 55140,
    "bus reads (tx)": 6726,
    "bus writes (tx)": 14,
    "bus incomplete telegrams": 2,
    "bus reads failed": 6,
    "bus writes failed": 1,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 25537,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 0,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "18.06",
      "entities": 63,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xC2 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "FW100",
      "device id": "0x10",
      "product id": 105,
      "version": "13.05",
      "entities": 10,
      "handlers received": "0x06",
      "handlers fetched": "0x016F 0x0165",
      "handlers pending": "0xA3 0xA2 0x12 0x0170 0x0171 0x0172 0x0166 0x0167 0x0168 0x0123"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "18.06",
      "entities": 0
    }
  ]
}
```

Thanks in advance for the advice/help.
-Paul

MichaelDvP, 2022-04-13:
Please check this:
- start `log all` from web, in terminal you can also use `watch 10` command, to see the telegrams from thermostat.
- press the button on thermostat
- stop logging (telnet: watch off), download log and post it here (drag&drop txt-file)
- we check the log to find the telegram/value from the thermostat
- you check if the value is writable from the bus with a telnet command (we'll tell you the command)
- we add it to ems-esp.

(NorbertS have listed some ww telegrams, but i'm notsure if this is status or setting. I think a good candidat to look for is `thermostat(0x10) -> All(0x00) ?(0x31D) data: 01`)

pdbell, 2022-04-13:
Hi @MichaelDvP ,

Thanks for the help, here is the telnet stuff:

```

┌──────────────────────────────────────┐
│ EMS-ESP version 3.4.0b8              │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└──────────────────────────────────────┘

ems-esp:$ watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x10
001+05:01:41.098 N 0: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02),                                                                    data: 0A
001+05:01:41.117 N 1: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02),                                                                    data: 5F 12 06 00 00 00 00 00 00 00
001+05:01:53.315 N 2: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
001+05:01:53.317 N 3: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
001+05:01:53.519 N 4: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 AA 01 22 F3 34 00
001+05:01:54.331 N 5: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 13 0D 0C 00 03 00
001+05:02:02.473 N 6: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
001+05:02:02.516 N 7: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 AA 01 22 F3 34 00
001+05:02:02.628 N 8: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
001+05:02:02.702 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 23 4B 50 03 02 0A 03 FF 23 06 01 04 14 1E 22 01 02 00
001+05:02:29.850 N 10: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
001+05:02:29.878 N 11: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2E
001+05:02:29.911 N 12: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
001+05:02:29.940 N 13: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2E 64 00
001+05:02:29.976 N 14: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
001+05:02:42.128 N 15: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
001+05:02:42.154 N 16: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 06 00 00 00 00 00 00 00
001+05:02:54.239 N 17: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 13 0D 0D 00 03 00
001+05:02:54.567 N 18: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
001+05:02:54.582 N 19: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
001+05:02:54.778 N 20: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 AA 01 22 F3 34 00
001+05:03:02.463 N 21: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
001+05:03:02.522 N 22: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 AA 01 22 F3 34 00
001+05:03:02.699 N 23: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
001+05:03:02.773 N 24: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 23 4B 50 03 02 0A 03 FF 23 06 01 04 14 1E 22 01 02 00
001+05:03:43.166 N 25: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
001+05:03:43.192 N 26: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 06 00 00 00 00 00 00 00
001+05:03:54.220 N 27: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 13 0D 0E 00 03 00
001+05:03:55.269 N 28: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
001+05:03:55.271 N 29: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
001+05:03:55.440 N 30: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 AA 01 22 F3 34 00
001+05:04:01.379 N 31: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 37 (offset 1)
001+05:04:01.408 N 32: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x01D3), data: 08 00 00
001+05:04:02.428 N 33: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
001+05:04:02.470 N 34: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 AA 01 22 F3 34 00
001+05:04:02.583 N 35: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
001+05:04:02.655 N 36: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 23 4B 50 03 02 0A 03 FF 23 06 01 04 14 1E 22 01 02 00
001+05:04:31.803 N 37: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 37 (offset 1)
001+05:04:31.823 N 38: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2E
001+05:04:31.856 N 39: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
001+05:04:31.893 N 40: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2E 64 00
001+05:04:31.930 N 41: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
001+05:04:32.078 N 42: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
001+05:04:37.035 N 43: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2F
001+05:04:37.068 N 44: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2F
001+05:04:44.016 N 45: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
[... cut ...]
ems-esp:$ watch off
Watching telegrams is off
ems-esp:$
```
And attached is the log, I pushed the button at/close to 1913
[log (2).txt](https://github.com/emsesp/EMS-ESP32/files/8483379/log.2.txt)

I notice that ID 0x01D3 seems to reflect if I have pushed the button (the data is `08 00 00` when the button has been pressed and the mode is active, and `00 00 00` when its inactive), but I'm not sure exactly if/how I could write it (I have tried sending the telegram `0B 10 FF 00 00 D3 08 00 00` with no effect).

MichaelDvP, 2022-04-14:
Looks like the 0x01D3 is only a status information and the command is only send to the boiler:
`Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 37 (offset 1)`
With 55°C seltemp (before in N10 it was set to 15°C). 

Please check if `call send "0b 08 35 01 11 00 37"` works or is overwritten by thermostat.
What is your setting for boiler seltemp and settemp?  If i thiink right, it was wwseltemp 55, wwsettemp 15. 
If it works we can add a bool command wwcharge and send seltemp in telegram 0x35.

pdbell, 2022-04-14:
wwseltemp seems to be always 70 degrees, seems to correlate to the dial on the boiler:

```
{
  "wwsettemp": 55,
  "wwseltemp": 70,
  "wwtype": "buffer",
  "wwcomfort": "hot",
  "wwflowtempoffset": 30,
  "wwcircpump": "off",
  "wwchargetype": "3-way valve",
  "wwhyston": 0,
  "wwhystoff": 0,
  "wwdisinfectiontemp": 75,
  "wwcircmode": "2x3min",
  "wwcirc": "off",
  "wwcurtemp": 57.2,
  "wwcurtemp2": 57.2,
  "wwcurflow": 0,
  "wwactivated": "on",
  "wwonetime": "off",
  "wwdisinfecting": "off",
  "wwcharging": "on",
  "wwrecharging": "on",
  "wwtempok": "on",
  "wwactive": "off",
  "ww3wayvalve": "on",
  "wwsetpumppower": 0,
  "wwstarts": 9832,
  "wwworkm": 270323
}
```

Tried the command you mentioned, seems the thermostat might be setting it back (I split out the command for convenience):

```
ems-esp:# call system send "0B 08 35 01 11 00 37"
000+14:07:13.339 I 506: [emsesp] Boiler(0x08) -> Me(0x0B), UBAFlags(0x35), data: <empty>
000+14:07:13.730 N 507: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+14:07:13.746 N 508: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
000+14:07:13.909 N 509: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
000+14:08:01.605 N 510: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+14:08:01.651 N 511: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
000+14:08:01.854 N 512: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+14:08:01.927 N 513: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 23 4B 50 03 02 0A 03 FF 23 06 01 03 10 1A 1E 01 02 00
000+14:08:02.546 N 514: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+14:08:02.569 N 515: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 06 00 00 00 00 00 00 00
000+14:08:04.865 N 516: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 0C 0E 0C 00 04 00
000+14:08:14.698 N 517: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+14:08:14.717 N 518: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
000+14:08:14.943 N 519: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
000+14:08:51.236 N 520: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
000+14:08:51.264 N 521: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2B
000+14:08:51.297 N 522: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+14:08:51.334 N 523: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2B 64 00
000+14:08:51.354 N 524: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
000+14:09:01.484 N 525: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+14:09:01.530 N 526: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
000+14:09:01.702 N 527: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+14:09:01.773 N 528: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 23 4B 50 03 02 0A 03 FF 23 06 01 03 10 1A 1E 01 02 00
000+14:09:03.572 N 529: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+14:09:03.602 N 530: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 06 00 00 00 00 00 00 00
000+14:09:04.809 N 531: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 0C 0E 0D 00 04 00
000+14:09:15.733 N 532: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+14:09:15.749 N 533: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
000+14:09:15.945 N 534: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
000+14:10:01.581 N 535: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), data: 19
000+14:10:01.622 N 536: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
000+14:10:01.799 N 537: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), data: 19
000+14:10:01.873 N 538: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 01 23 4B 50 03 02 0A 03 FF 23 06 01 03 10 1A 1E 01 02 00
000+14:10:04.541 N 539: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+14:10:04.571 N 540: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 06 00 00 00 00 00 00 00
000+14:10:04.877 N 541: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 04 0C 0E 0E 00 04 00
000+14:10:16.611 N 542: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+14:10:16.625 N 543: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
000+14:10:16.790 N 544: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
---
000+14:10:53.239 N 545: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
---
000+14:10:53.267 N 546: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 2B
000+14:10:53.299 N 547: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+14:10:53.329 N 548: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 2B 64 00
[... cut ...]
000+14:11:05.517 N 555: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+14:11:05.541 N 556: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 06 00 00 00 00 00 00 00
000+14:11:18.416 N 557: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 02
000+14:11:18.436 N 558: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 58
000+14:11:18.628 N 559: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 01 00 96 01 30 F3 34 00
```

MichaelDvP, 2022-04-14:
Have you tried if wwonetime works? We havn't found a rule yet on which boilers it works, but it works only if current ww-temperature is lower than wwseltemp-5(hysteresis).

pdbell, 2022-04-14:
Heres an excerpt, from what i can see though it has no effect:

```
ems-esp:# call boiler wwonetime on
000+02:43:58.890 N 161: [emsesp] Me(0x0B) -> Boiler(0x08), UBAFlags(0x35), data: 23
000+02:44:01.012 N 162: [emsesp] Me(0x0B) <- Boiler(0x08), UBAMonitorWW(0x34), data: 1B
000+02:44:01.062 N 163: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 0F 00 EE 00 EE A0 00 04 03 00 04 1F FA 00 26 69 00
000+02:44:01.198 N 164: [emsesp] Me(0x0B) <- Boiler(0x08), UBATotalUptime(0x14), data: 1B
000+02:44:01.223 N 165: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 2B 20 AE
000+02:44:01.354 N 166: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 1B
000+02:44:01.389 N 167: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 58 64 00 00 F6 03 01 03 64 0A 04
000+02:44:01.547 N 168: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 1B
000+02:44:01.576 N 169: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 46 00 00 46 00 02 4B 00 FF FF
```

I've also tried that from the web interface, it seems to do the same thing, with no effect

MichaelDvP, 2022-04-14:
Last idea: try `call system send "0B 08 35 00 23 00 00 3C"`
activating onetime, disable program and set temperature to 60°C.

pdbell, 2022-04-15:
Hi @MichaelDvP,

First of all, thanks for all the ideas, I'm still getting my head around the EMS bus datagrams.

I tried this, and it worked briefly, but again the Thermostat overrode it:

```

ems-esp:# watch 35
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x35
ems-esp:# call system send "0B 08 35 00 23 00 00 3C"
000+21:16:48.237 N 1: [emsesp] Me(0x0B) -> Boiler(0x08), UBAFlags(0x35), data: 23 00 00 3C
000+21:16:48.394 N 2: [emsesp] Me(0x0B) <- Boiler(0x08), UBAFlags(0x35), data: 1B
000+21:16:48.409 I 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAFlags(0x35), data: <empty>
000+21:16:53.696 N 4: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
000+21:18:54.635 N 5: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
000+21:20:56.631 N 6: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
000+21:22:58.448 N 7: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
000+21:25:00.379 N 8: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 0F (offset 1)
```

I wonder if I can configure the FW100 to not schedule or manage the DHW, as it appears to be the boiler doing everything, except for the scheduling.

pdbell, 2022-04-15:
Eureka!

I think I'm now getting to the bottom of how the FW100 works in conjunction with the boiler...

The boiler expects setpoint temperatures to be written periodically, otherwise it will revert to the selected temperature.  This is why the FW100 is constantly overwriting, it needs to maintain the desired setpoint in the boiler to honor the schedule set in the device.

Notes:
- DHW schedule with no entries -> FW100 maintains last setpoint with periodic commands
- FW100 disable DHW -> EMS-ESP can control setpoint with 0x35 with telegrams, but failing to write at a similar interval will revert the value
- FW100 refresh interval: 2 minutes
- Boiler setpoint timeout interval: 4 minutes
```
ems-esp:# call system send "0b 08 35 01 11 00 28"
001+02:03:26.217 I 75: [emsesp] Boiler(0x08) -> Me(0x0B), UBAFlags(0x35), data: <empty>
ems-esp:# watch 34
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x34
001+02:03:48.241 N 76: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 65 02 65 B9 00 04 03 00 04 20 58 00 26 6E 00
001+02:03:58.184 N 77: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 66 02 66 B9 00 04 03 00 04 20 58 00 26 6E 00
001+02:04:08.350 N 78: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 66 02 66 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:04:18.163 N 79: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 66 02 66 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:04:28.049 N 80: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 66 02 66 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:04:38.079 N 81: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 67 02 67 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:04:48.084 N 82: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 67 02 67 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:04:58.027 N 83: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 67 02 67 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:05:08.194 N 84: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 68 02 68 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:05:18.224 N 85: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 68 02 68 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:05:27.949 N 86: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 68 02 68 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:05:37.954 N 87: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 68 02 68 A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:05:47.928 N 88: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6A 02 6A A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:05:57.902 N 89: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6A 02 6A A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:06:08.036 N 90: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6A 02 6A A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:06:17.849 N 91: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6A 02 6A A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:06:27.823 N 92: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6B 02 6B A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:06:37.797 N 93: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6B 02 6B A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:06:47.771 N 94: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6B 02 6B A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:06:57.657 N 95: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6B 02 6B A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:07:07.880 N 96: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6B 02 6B A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:07:17.692 N 97: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6C 02 6C A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:07:27.666 N 98: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 6C 02 6C A1 00 04 03 00 04 20 58 00 26 6E 00
001+02:07:37.391 N 99: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 46 02 6B 02 6B 89 00 00 03 00 04 20 58 00 26 6F 00
001+02:07:47.614 N 100: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 46 02 6C 02 6C 89 00 00 03 00 04 20 58 00 26 6F 00
```

I'm going to dig around a bit more (there is an option to configure the DHW to a level based on whether the mode is comfort or eco) but it seems if we would like to support setting setpoint on a boiler like this via datagram 0x35 we'd need a mechanism to keep writing it...

pdbell, 2022-04-20:
Ok, just a wrap up for anyone else who has a FW100... it seems the easiest way to get some level of DHW control is to assign different temperatures to the ECO/HEAT modes, and then configure the DHW to follow the heating modes (i.e. don't schedule the DHW separately).  As you can also control the setpoint for normal heating in all of the modes remotely as well, this means you can essentially use the FROST/ECO/HEAT modes to be used for setting DHW behavior.

This is "good enough" for what I'm doing, as I just want a convenient way of boosting my hot water and changing the schedule without going down into the basement.  I don't need full setpoint control of the hotwater, a few levels is fine, so I'll close this.  Thanks to @MichaelDvP  for the sparring and support.

closed 2023-02-19 by proddy