emsesp/EMS-ESP32 #470: Updating table component in the WebUI to support sorting & filtering

proddy, 2022-04-14:
Replacing MUI React Table component with https://github.com/table-library/react-table-library

https://github.com/proddy/EMS-ESP32/tree/newtable

harvey637, 2022-04-14:
Hi,
this sounds exciting!

By the way, will this also work with customization? That is asking for a "all selected" line to activate/deactivate the "favourite, read_only, hide_from_api, hide_from_dashboard" to work only on the selected entities. Mostly for peaple like me, who always have to deactivate around 60 DHW-entities, making 120+ clicks :-) And sad to say, that I have some rare reboots on rtc-watchdog, where all cusomization is lost, dont worry, I am used to live with that.

just thinking, not demanding - thank you
cu
Harvey

proddy, 2022-04-14:
Hi Harvey, yes, I had you mind when developing this. I'll add filters to the table so you can quickly group the customizations. I'll be sharing early versions so you can play around and give feedback.

proddy, 2022-04-17:
I made the first build. As usual, I got a bit carried away and touched a few other things. The latest customization code is not compatible with earlier 3.4 releases so any changes to the Analog sensors will be lost. 

The table columns are resizeable and you can sort on most of the columns. I've also added filters and the "set all" type buttons as requested by @harvey637 . Please take a look and let me know what you think.

Build is here:

[EMS-ESP-3_4_0b12-ESP32.bin](https://github.com/proddy/EMS-ESP32/releases/download/latest/EMS-ESP-3_4_0b12-ESP32.bin)

harvey637, 2022-04-18:
Hi,
first impressions with my personel ideas:
very nice, with some fine tuning to go .....
- the position to resize a column is not easy visable (a little more highlighting the selected columns header)
- initial column size abriviates values (eg. those that are enums like "optimized" for thermostat) in customizations, 
- filtering favourites with no favourites?
- filtering for content like "dhw" in the entity name? (makes is easy to dasable everything with warm water ...)
- sorting "up|down|default"
- favourites not visibly marked (no star) and not handled by sort, (they are somewhere due to sorting, only on|of for favourites

So my very short observations:
- sorting should separate favourites and non_favourites, and sort them individuelly
- filter by content (part of the entity lng name, like "all with dhw")
- unmark_all does not unmark read_only, which makes sense and is ok, only a bit irritating at first sight, maybe it should be greyed out too?

happy easter monday
Harvey

proddy, 2022-04-18:
Thanks for the feedback. Comments below:

> * the position to resize a column is not easy visable (a little more highlighting the selected columns header)

yes, I couldn't find any easy way but I'll keep trying to show a vertical bar. The resize is a bit of a hidden feature.  I don't expect users will spend a lot of time on the table. The idea with customizations is to set it once and leave it.

> * initial column size abriviates values (eg. those that are enums like "optimized" for thermostat) in customizations,

Yes, seen this and couldn't find a way around it. I'm still trying different things in the CSS code.

> * filtering favourites with no favourites?

it will show an empty table. This is normal, no?

> * filtering for content like "dhw" in the entity name? (makes is easy to dasable everything with warm water ...)

Ok, I'll add custom filtering. Thought you might want this!

> * sorting "up|down|default"

I added default sort on the Name column, but can change this back to whatever comes back from the server as it was before (which is the order of the entities are coded and pretty meaningless). Let me know what makes sense in your opinion.

> * favourites not visibly marked (no star) and not handled by sort, (they are somewhere due to sorting, only on|of for favourites

I've added that star back.

> * sorting should separate favourites and non_favourites, and sort them individually

not sure what you mean. Do you have an example? Which screen are you in?

> * filter by content (part of the entity lng name, like "all with dhw")

I will add.

> * unmark_all does not unmark read_only, which makes sense and is ok, only a bit irritating at first sight, maybe it should be greyed out too?

The button says 'Set Disabled' or 'Set Enabled' which is only referring to 'MQTT/API' and 'Hide from Dashboard' masks. Maybe a better description is needed? The web site needs to be responsive (i.e. also show nice on iPhone 7 screen) which is why I'm keeping the text short where possible.

proddy, 2022-04-18:
oh just saw your updates.

proddy, 2022-04-18:
Thanks for the feedback. Comments below:

> * the position to resize a column is not easy visable (a little more highlighting the selected columns header)

yes, I couldn't find any easy way but I'll keep trying to show a vertical bar. The resize is a bit of a hidden feature.  I don't expect users will spend a lot of time on the table. The idea with customizations is to set it once and leave it.

> * initial column size abriviates values (eg. those that are enums like "optimized" for thermostat) in customizations,

Yes, I've seen this too and couldn't find a way around it. I'm still trying different things in the CSS code.

> * filtering favourites with no favourites?

it will show an empty table. This is normal, no?

> * filtering for content like "dhw" in the entity name? (makes is easy to dasable everything with warm water ...)

Ok, I'll add custom filtering. Thought you might want this!

> * sorting "up|down|default"

I added default sort on the Name column, but can change this back to whatever comes back from the server as it was before (which is the order of the entities are coded and pretty meaningless). Let me know what makes sense in your opinion.

> * sorting should separate favourites and non_favourites, and sort them individually and mark favourites with asterisk

Favourites are shown with a Star now. But not sure what you mean here? Can you explain with an example?

> * no sorting (original order) should be possible as third option

Ok, I'll add this back. I had it in before.

> * filter by content (part of the entity long name, like "all with dhw")

Ok. will add custom filtering.

> * sorting is not my personel priority, as you said, the dashboard is not the primary display but something like iobroker, mqtt.

Sorting is handy in the Dashboard for grouping (everything that starts with boiler or dwh). But agree, it's more a gimmik

> * so sorting is "only" to make marking (hiding) of many entities easy (like all dhw), but this only works with content filter

agree

> * my idea of "sorting" is more a grouping, like "all consumed energy", "heating|cooling|compressor|activity", "all cooling stuff", favourites help a lot and may be enough or me. Putting entities into categories is overkill, do it in frontend software (iobroker, ...)

agree. I was thinking of adding groups and may do so later. The backend code supports it. I call it tag, like 'boiler' or 'dwh'.

> * unmark_all does not unmark read_only, which makes sense and is ok, only a bit irritating at first sight, maybe it should be greyed out too?

The button says 'Set Disabled' or 'Set Enabled' which is only referring to 'MQTT/API' and 'Hide from Dashboard' masks. Maybe a better description is needed? The web site needs to be responsive (i.e. also show nice on iPhone 7 screen) which is why I'm keeping the text short where possible.

> * count the number of hidden entities in dashboard overview (not yet done, but helps when overlooking/missing hidden entities)

Good point. I show the total number of entities but not the displayed one. I'll add.

> * with no entites at all (my gateway) shows a white up|down sorting button on the right side

I've seen this too. It's a nasty side affect and not sure how to remove it. I'll keep trying.

> * right column for changing value (pencil symbol) nice, but favourites (asterisk) on the left?

I show the asterix now in the name

harvey637, 2022-04-18:
Hi @ Proddy,
thank you with your feedfeedback :-)
Really, this is all very good stuff, so minor and personel oddities are completly ok with me. 

One experience is a bit strange: in the last 1~2 weeks with different firmwares I had some software resets, the last one say "rtc clock" something, dont remember exactly. After a reset some application settings were correctly reused like gpio-settings , format options ... but not the customization. This needs to be done manually after reset. Nothing big, just 100+ clicks to get ems-esp dashboard "nice", also I have 60+ unused and mostly nonsense values as variables in iobroker. So after customizing ems-esp I have to reload the iobroker plugin to get rid of unused entities. No big deal, just observing.
Strange is that some settings (the application settings) are retained, others (the customization) is lost. ...
Happy easter and blue sky to you - go outside, dont code!
Harvey

proddy, 2022-04-19:
> One experience is a bit strange: in the last 1~2 weeks with different firmwares I had some software resets, the last one say "rtc clock" something, dont remember exactly. After a reset some [...]
Strange is that some settings (the application settings) are retained, others (the customization) is lost. ...

@harvey637 this is not good. I've very alergic to EMS-ESP crashing. If you're able to reproduce this please let me know so I can try and reproduce it.

harvey637, 2022-04-19:
Hi, fast answer - thank and testing. I know this is work in progress so just feedback:
- in a perfect world the "filter" (content filtering - works!) the "all" mark/unmark should only alter the filtered entities (maybe name them "filtered ...")
- I have some optical glitches with the errorcode like partitially overlapping several lines (only in customization, may be of parenthesis in errorcode)
- if hidden from dashboard the read_write flag should be greyed out just like the favourites asterisk is (dont see -> cant write :-)
- favourites are not sorted on top of non_favourites
- column width good
- column width slider visible
-  selection of type in dashboard does not work if mouse over icon, only name like "boiler ...", but line is marked
- sorting of values is of no use (to me)?
- offer sorting only with > 2 entities
- offer favourites_only only with >= 1 favourite

ciao
Harvey

proddy, 2022-04-19:
Keep the feedback coming!

- there's a bug in the enable/disable buttons in that it removes the entities, so need to fix that quickly
- the optical glitches like the multiline I'm seeing too and will look into proving the layout
- read/write flag should be greyed out if its hidden from MQTT/API - good spot!
- sorting favorites I still don't understand. If the favourites are at the top it'll break the sorting logic. Suggest to leave it.
- sorting of values is indeed useless. I'll remove it.
- sorting but counting entities and favourites its a lot of extra code and not worth the effort (or the remaining RAM)
- selecting icon in Dashboard is a good spot too, I'll fix it
-

harvey637, 2022-04-19:
hi,
for the sorting and favourites an example (read left to right as top to down):
we have favourites f1, f2, f6, f7 and non_favourites 3, 4, 8, 9
with sorting number we get (down): f1 - f2 - 3 - 4 - f6 - f7 - 8 - 9
with sorting favourites and then non_favourites we get (down): f1 - f2 - f6 - f7 - 3 - 4 - 8 - 9
sort it other direction(up):
9 - 8 - f7 - f6 - 4 - 3 - f2 - f1
or "in two blocks":
f7 - f6 - f2 - f1 - 9 - 8 - 4 - 3

so  (as it now) favourites are only marked with an asterisk, not with the upper positions on the list.
What I "expect" is:
the favourites block always on top (down f1 - f2 - f6 - f7 or up: f7 -f6 -f2 -f1) 
and non_favourites block always below (down: 3 - 4 -8 - 9 or up: 9 - 8 - 4 - 3)
So sorting is in two blocks, each block sorted as requested, but all favourites always on top of all non_favourites.
hope this clarifies my idea.
from the programing I guess something like the first character of a favourite is always below letter "a" makes sort naturally ...
ciao
Harvey

proddy, 2022-04-20:
but if you sort by name, seeing the entity names in this order:  f1 - f2 - 3 - 4 - f6 - f7 - 8 - 9 would confuse people.

To do what you asked it would make sense to add another column to the Dashboard for Favourites so that column is sortable, so favourites come to the top of the list and are sorted alphabetically. But it makes it messy and we're overcomplicating it, for the small handful of people that will ever use it.

proddy, 2022-04-20:
> Hi, woke up again :-)
> a picture show more than 1000 words, this is the order NOW:
> [image: sort1]
> <https://user-images.githubusercontent.com/10156426/164177231-7444fda8-49ec-45a7-913e-e68875b281e4.png>
> You see all the NAMES are sorted, but my favourites are somewhere down, in
> boiler I even have to scroll down to find them.
>
> The old unsorted behaviour was to have favourites on top, the "rest" below.

I know what you mean, I just find it counter intuitive as Its not sorted by
name anymore. The show favorites button is there to give focus on favorites
and that data can be sorted.

I’ll see what I can do. I have another 6hrs on this flight…

harvey637, 2022-04-20:
good landing then :-)

dont try to make it overcomplicated, I tried with some more favourites and used the favourites_only button. This is also a nice view and already there!

ciao
Harvey

proddy, 2022-04-21:
did a new build, still some formatting things to fix but we're close

MichaelDvP, 2022-04-21:
Nice work. The filtering in customization is very good. 
The icon logic needs to check some more cases. Click favorite and then hide form web does not disable the favorite.
And there is a c&p issue [here](https://github.com/proddy/EMS-ESP32/blob/726b4fcc0b4e5dbed78705654fe3b2108688c703/interface/src/project/DashboardData.tsx#L258-L259), this shoud be
```
        NAME: (array) => array.sort((a, b) => a.n.localeCompare(b.n)),
        TEMPERATURE: (array) => array.sort((a, b) => a.t - b.t)
```
I personally like the native, thematically sorting, where selected roomtemp and current roomtemp are together and min/max values for burner/pumps. Alphabetical sorting seem useless to me, but others may prefere it.
To have the pencil column with sorting option would give parameters first or measured values first, i think this could be usefulll. If the favorite-star goes to a own column, this could also be a sort, but favorite only checkbox is also ok.

proddy, 2022-04-22:
Thanks Michael. I'll see i can make the sort 3 way with a no sorting option to keep the original sequence.

proddy, 2022-04-24:
I fixed the two issues Michael mentioned. Still looking into how to do a 3-way cycle between sorting instead of a toggle

proddy, 2022-04-24:
ok fixed. I'm going to merge this into Dev now

MichaelDvP, 2022-04-30:
Some of the tables needs bit more formatting. The header of dashboard status should have more width, padding, or a space like `&nbsp;FAIL` to make it better readable.
[image]
I also prefere numbers right aligned, 
[image]

proddy, 2022-04-30:
yes, I spent a few hours looking at this this week but its really hard to get the CSS working nicely so the columns align up. I'll play around with it further, and right-align numerical columns and also cut off cells with ellipses (...) instead of increasing the row height which I personally don't like.

proddy, 2022-05-01:
Made some more changes to the table formatting in b14

harvey637, 2022-05-07:
hi, sorry i am on bad internet, so feedback after mid May, cu harvey--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

   
  think it's all ok now.

proddy, 2022-05-20:
re-opening for one small annoance. When the window (browser) is resized the Table doesn't automatically resize and align the columns.

closed 2022-05-23 by proddy