emsesp/EMS-ESP32 #474: Auto restart if network or MQTT offline too long

jefft4, 2022-04-23:
I've noticed a couple of times that the ESP can lose the network connection - for example because the router was rebooted or an access point was offline - and fails to re-establish the MQTT connection when the network service is restored.  Most recently, my EMS-ESP32 was useless for 6 hours because MQTT had not reconnected, although I could access the web UI from a browser just fine.  The only way I could resolve this was to restart the ESP.

I know this is an issue sometimes with ESPs - I've seen it with ESP8266 and ESPEasy, where I use rules that say 'if network inaccessible for 10 minutes, reboot the ESP'.

It would be helpful if EMS-ESP could detect that either network or MQTT has been disconnected for too long - say, 15 minutes - and restart the system.

proddy, 2022-04-23:
We've had a few reported issues of this but it's always hard to reproduce, although I think it's something we should look into anf fix the root cause rather than implement a workaround. 

If you turn off Clean Session and use the latest build are you able to reproduce it?

Note this is also being discussed in the MqttClient library https://github.com/marvinroger/async-mqtt-client/issues/281

jefft4, 2022-04-25:
It was triggered by taking my whole network offline to change the UPS cabling, so it's not something I'm keen to try to replicate exactly ;-) but I'll play around later and see if I can cause it by bouncing the ESP's wi-fi connection from the AP, blocking its route to the MQTT server at the router temporarily and things like that.

proddy, 2022-04-25:
That would be helpful. We all tried this with https://github.com/emsesp/EMS-ESP32/issues/441

proddy, 2022-05-01:
added a fix in b14

closed 2022-05-05 by proddy