emsesp/EMS-ESP32 #477: Changing datetime on thermostat gives callback errors

tp1de, 2022-04-25:
Using API V3 call to change thermostat.datetime gives the following errors within log:

_2022-04-25 12:50:20.105 W 112: [thermostat] Set date: invalid data, wrong length
2022-04-25 12:50:20.105 E 113: [emsesp] Call failed with error: callback function failed (Error)_

API http return: _undefined callback function failed http://ems-esp/api/thermostat/datetime_

MichaelDvP, 2022-04-25:
Check [https://emsesp.github.io/docs/#/Command?id=thermostats-thermostat](https://emsesp.github.io/docs/#/Command?id=thermostats-thermostat) and compare the length of your input. We have this question often, so there is an extra warning log of "wrong length".

tp1de, 2022-04-25:
@MichaelDvP  you are right Michael, I tested  .... but it is still confusing for end-users:

Web-API output for read (get)  is: 25.04.2022 15:36
For change you need to post: 25.04.2022-15:36:00-0-1 

Don't you think that this might be confusing and should be changed? Either:
(1) http post: adding "-" between date and time and adding seconds and day of week and dst automatically   or
(2) Displaying the format similar to write by http get as well

closed 2023-02-19 by proddy