emsesp/EMS-ESP32 #490: ECO button status on Worcester Bosch Greenstar Highflow 550CDi?

Stevehans, 2022-05-04:
**Question**
Is there any way of reading the status of the eco button on my Worcester Bosch Greenstar Highflow 550CDi, I've exported the dashboard and nothing changes when I toggle the status?

**Device information**
{
  "System": {
    "version": "3.4.0b16",
    "uptime": "001+21:59:04.223",
    "uptime (seconds)": 165544,
    "network time": "connected",
    "freemem": 169,
    "reset reason": "Software reset CPU / Software reset CPU",
    "temperature sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "SSID": "r0s3m0unt",
    "BSSID": "C8:0E:14:17:A3:D5",
    "RSSI": -72,
    "MAC": "48:55:19:BE:39:84",
    "IPv4 address": "192.168.1.151/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 56603,
    "bus reads (tx)": 8293,
    "bus writes (tx)": 2,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 36549,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "API calls": 0,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "25.12",
      "entities": 48,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "25.12",
      "entities": 0,
      "handlers pending": "0x06"
    }
  ]
}

proddy, 2022-05-04:
open up the Terminal and type `read 8 33` and send back the telegram that shows

Stevehans, 2022-05-04:
ems-esp:$ read 8 33
002+01:47:56.322 I 0: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 36 00 00 00 00 02 46 00 FF FF
ems-esp:$

proddy, 2022-05-04:
the comfort (hot, eco, intelligent) is the 10th byte which has a value of x00 which means Hot. Try changing the eco button on the boiler and see if it changes.

Stevehans, 2022-05-04:
It's currently set to eco but always displays as hot in the dashboard.

Stevehans, 2022-05-04:
This changes as I toggle the button:-
```
002+01:49:19.685 N 21: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 13 80 00 80 00 00 01 00 00 D5 DB 00 39 B2 00
002+01:49:29.450 N 29: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 14 80 00 80 00 00 01 00 00 D5 DB 00 39 B2 00
002+01:49:39.434 N 31: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 13 80 00 80 00 00 01 00 00 D5 DB 00 39 B2 00

```

Stevehans, 2022-05-04:
0x13 = eco on 
0x14 = eco off

MichaelDvP, 2022-05-05:
No, this is the actual dhw temperature, changing between 0x114 and 0x113 (27,6°C and 27,5°C). 
Use `watch on` and push the boiler-button, then check which telegram is broadcasted on the button action.

MichaelDvP, 2022-05-05:
According to [manual, page 8](https://www.worcester-bosch.co.uk/support/literature/download/6720818076) and [NorbertS1, page 17](https://raw.githubusercontent.com/norberts1/hometop_HT3/master/HT3/docu/HT_EMS_Bus_messages.pdf) it could be last byte (offset 11 / byte 15)  in telegram 0x33, 0xFF - ECO (dhw priority), 0x00 - off.
@Stevehans Please check.

Stevehans, 2022-05-05:
Just checked, it's not offset 11, that stays at 0xFF regardless of ECO on or off

Just running a watch to see whether I can see anything toggling

Stevehans, 2022-05-05:
Nothing seems to change from what I can see, so not visible in the usual broadcast packets

Stevehans, 2022-05-06:
Doesn't look as though WB report all the various parameters, for instance, I also can't see the status of the dhw on/off from the programmer when then cycles.

PhillyGilly, 2022-05-13:
I'm not in a position to run tests atm, but what is the relationship between "boiler dhw comfort" and "boiler hysteresis dhw on"? I thought ECO mode just set the hysterisis to -10C. So can you monitor that instead?

Stevehans, 2022-05-18:
"boiler dwh comfort" is always set to "hot" regardless of the status of the ECO button, likewise wwhyston is always 0.

MichaelDvP, 2022-05-18:
> Nothing seems to change from what I can see, so not visible in the usual broadcast packets

I don't think it's a broadcast telegram, i expect a message from controller to boiler when pushing the button (`Controller(0x09 ->Boiler(0x08)`)

Stevehans, 2022-05-18:
Nothing sent on a watch when I toggled it, if it's there then I may need to keep reading some of the type values and see what changes there (hence the other enhancement for the ability to have a debug poll of various types)

glitter-ball, 2022-06-09:
On my Greenstar 24i System ErP, 'dhw comfort' changes from 'eco' to 'hot' when the setting is toggled. The 'dhw hysteresis on temperature' stays at -5.0 deg C throughout. When toggling on the boiler, the change doesn't appear immediately in the dashboard - you either have to wait for an update or refresh the page, apparently.

Stevehans, 2022-06-10:
I'm fairly certain that my 550 desn't do that, I'll check as soon as I get back next week and update here.

chaosmaster, 2022-07-28:
It seems I am having the same issue on Junkers Cerastar Comfort with CR100 Thermostat.
I can't see any messages from the controller.
This is the log when starting with eco enabled, then disabling it, then reenabling it:

```
000+04:31:14.091 N 242: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+04:31:14.176 N 243: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x02A5), data: 01 0B 21 22 00 00 22 22 05 A0 02 03 01 00 52 03 41 00 00 11 01 02
000+04:31:14.401 N 244: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x02A5), data: 0A 6A 00 00 07 00 00 00 00 00 00 4B 00 3C 01 FF 00 00 (offset 22) 
000+04:31:14.722 N 245: [emsesp] Thermostat(0x18) -> All(0x00), RC300WWmode2(0x031D), data: 00 00 05 07
000+04:31:18.724 N 246: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 53 00 00 00 20 64 C0 01 55 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+04:31:18.964 N 247: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 32 01 54 80 00 81 00 00 01 00 00 85 6E 01 41 5B 00
000+04:31:19.723 N 248: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 53 00 00 00 20 44 C0 01 54 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
ems-esp:$ 
ems-esp:$ 
ems-esp:$ 
000+04:31:28.754 N 249: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 53 00 00 00 20 44 C0 01 54 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+04:31:28.994 N 250: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 32 01 54 80 00 81 00 00 01 00 00 85 6E 01 41 5B 00
000+04:31:38.709 N 251: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 52 00 00 00 20 44 C0 01 53 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+04:31:38.950 N 252: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 32 01 53 80 00 81 00 00 01 00 00 85 6E 01 41 5B 00
000+04:31:47.735 N 253: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x02A5), data: 00 51 03 42 (offset 13) 
000+04:31:48.702 N 254: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 50 00 00 00 20 44 C0 01 53 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+04:31:48.942 N 255: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 32 01 53 80 00 81 00 00 01 00 00 85 6E 01 41 5B 00
```

Here's the output of `show devices`

```
These EMS devices are currently active:

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:35.04)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A 
 Fetched telegram type IDs: 0x14 0x16 0x33 
 Pending telegram type IDs: 0xBF 0xC2 0x26 
 Ignored telegram type IDs: 

Thermostat: RC200/CW100 (DeviceID:0x18, ProductID:157, Version:41.08)
 Received telegram type IDs: 0x06 0xA2 0x31D 
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x2F5 0x23A 
 Pending telegram type IDs: 0xA3 0x12 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x31B 0x31E 0x267 0x240 
 Ignored telegram type IDs: 0xBF 

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:35.04)
```

Stevehans, 2022-10-11:
I will do that this evening.

closed 2023-02-19 by proddy