emsesp/EMS-ESP32 #491: (Cosmetic) Arrow shown in watch is in the wrong direction?

Stevehans, 2022-05-04:
The arrow in the watch is shown in the wrong direction for transmissions from the EMS controller to the boiler, I believe that the request is from the controller to the boiler for uptime.

001+22:03:03.020 N 8: [emsesp] Me(0x0B) <- Boiler(0x08), UBATotalUptime(0x14), data: 1B
001+22:03:03.045 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 18 F1 33

MichaelDvP, 2022-05-04:
> 001+22:03:03.020 N 8: [emsesp] Me(0x0B) <- Boiler(0x08), UBATotalUptime(0x14), data: 1B

ems-esp requests telegram 0x14 from boiler with length of max. 0x1B, the arrow shows that it's a request (destination | 0x80)

> 001+22:03:03.045 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 18 F1 33

Boler sends data to ems-esp, telegram 0x14

proddy, 2022-05-04:
maybe it should always be <from> -> <to>, so only one arrow. Anything sent to "me" will be a Read anyway

MichaelDvP, 2022-05-05:
No, thermostat and km200 sends out a lot of requests and we need to identify it in pretty_telegram.
It's the same misunderstanding as wit tx-reads and tx-writes. 
We can use texts instead of arrows and change `data` to `length` for requests. 
`Me(0x0B) read Boiler(0x08), UBATotalUptime(0x14), length: 1B` 
` Boiler(0x08) write Me(0x0B), UBATotalUptime(0x14), data: 18 F1 33`
Is this more clear?

proddy, 2022-05-15:
let's just leave it.

closed 2023-02-19 by proddy