emsesp/EMS-ESP32 #496: RC30: hc2 design temperature missing

stobies, 2022-05-12:
**Bug description**

I have just freshly installed v3.4.0b17 on my bbqkees device and overall it is working great.

One thing I have noticed missing is the design temperatur for hc2. Is is shown for hc1 and I know for sure that I can set the design temperatur for hc2 in the RC30 also.

**Steps to reproduce**

Attach EMS-ESP via the maintenance port to heating system, inspect found values for RC30 thermostat.

**Expected behavior**

I would expect hc2 design temperature to be among the shown values for the RC30.

**Device information**

{
  "System Status": {
    "version": "3.4.0b17",
    "uptime": "000+17:02:39.309",
    "freemem": 117,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 70399,
    "bus reads (tx)": 20493,
    "bus writes (tx)": 5,
    "bus incomplete telegrams": 4,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 5,
    "tx gpio": 17,
    "dallas gpio": 4,
    "pbutton gpio": 33,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 6,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "BK13/BK15/Smartline/GB1x2",
      "device id": "0x08",
      "product id": 64,
      "version": "02.07",
      "entities": 69,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC30",
      "device id": "0x10",
      "product id": 67,
      "version": "02.08",
      "entities": 72,
      "handlers received": "0xA3 0x06 0xA2 0x12 0x38 0x39",
      "handlers fetched": "0x3E 0x3D 0x3F 0x48 0x47 0x49 0xA5 0x37",
      "handlers pending": "0x42 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60",
      "handlers ignored": "0xAC"
    },
    {
      "type": "Mixer",
      "name": "MM10",
      "device id": "0x21",
      "product id": 69,
      "version": "02.00",
      "entities": 6,
      "handlers received": "0xAB",
      "handlers fetched": "0xAA",
      "handlers pending": "0xAC"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.01",
      "entities": 0,
      "handlers pending": "0x06"
    }
  ]
}

MichaelDvP, 2022-05-12:
If you have hc1with radiators and hc2 with floorheating, i think i know the reason: RC35 stores the designtemp for floorheating in a different position (offset 36), and i think RC30 always use offset 17. Please read or fetch from log the telegram 0x47 (both parts) and post here.

MichaelDvP, 2022-05-12:
Yes, there is 30°C in offset 17 and telegram is only 31 bytes long. I've made a fix and PR, you can [test here](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest),

stobies, 2022-05-19:
Validated - thanks a lot!

proddy, 2022-05-20:
merged.

commit 1863e57f (2022-05-14): fix #496, RC30 design temp
commit ba315f21 (2022-05-14): Merge pull request #497 from MichaelDvP/dev

fix #496, RC30 design temp
commit 33e58ec4 (2022-05-20): RC30 designtemp write, 2.fix #496
closed 2022-05-20 by proddy