emsesp/EMS-ESP32 #499: What is the difference between "dhw turn on/off" and "dhw activated"?

MichelBarneveld, 2022-05-13:
What is the difference between "dhw turn on/off" and "dhw activated"?
Both are preventing warm water when turned off. 

And which of these 2 should I use to send short bursts of cold water by toggling it off and on?

MichaelDvP, 2022-05-14:
"dhw activated" is the standard boiler function. Writing does not always work, some boilers have a hardware switch on the controller and "dhw activated" is only readable.
"dhw turn off" is a special function proddy uses for shower coldshot in dhw flow systems. It sets the boiler to testmode, activates 3-way-valve and pump, but switches the burner off.  "dhw turn on" cancels testmode and sets boiler to normal operation again.

Try out which setting fits better your demands.

MichelBarneveld, 2022-05-15:
Thank you @MichaelDvP . Although both options are working for my Nefit Trendline, I'll use the turn on/off setting.

closed 2023-02-19 by proddy